arendst/Tasmota #12131: H804 Sonoff Basic Module LED output flickering (ver 9.4.0)

blackdiamondsoftware, 2021-05-18:
### PROBLEM DESCRIPTION
With release 9.4.0 the H804 PWB LED dimming module (sonnoff Basic) has started occasionally pulsing the RGB channels to full intensity every few seconds when the intensity is set lower to 100%

both via MQTT or local web control

Sonoff Basic (H804)
Pre-compiled binary 9.4.0
Flashed OTA

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): H804
- [ x] Tasmota binary firmware version number used: 9.4
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: OTA
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
stat/IslandRGB/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:12:58.471 MQT: stat/IslandRGB/RESULT = {"Module":{"20":"H801"}}
19:12:58.684 MQT: stat/IslandRGB/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"288":"Led1"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"420":"PWM5"},"GPIO5":{"321":"Led_i2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"418":"PWM3"},"GPIO13":{"417":"PWM2"},"GPIO14":{"419":"PWM4"},"GPIO15":{"416":"PWM1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X ] Provide the output of this command: `Status 0`:
```lua

19:13:34.861 MQT: stat/IslandRGB/STATUS = {"Status":{"Module":20,"DeviceName":"IslandRGB","FriendlyName":["Island RGB"],"Topic":"IslandRGB","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:13:34.866 MQT: stat/IslandRGB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release-9.3.1/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:21:30","StartupUTC":"2021-05-18T17:52:04","Sleep":50,"CfgHolder":4617,"BootCount":57,"BCResetTime":"2021-01-28T06:05:18","SaveCount":519,"SaveAddress":"F7000"}}
19:13:34.872 MQT: stat/IslandRGB/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"405/699"}}
19:13:34.878 MQT: stat/IslandRGB/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["inFused",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","2805C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
19:13:34.890 MQT: stat/IslandRGB/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
19:13:34.896 MQT: stat/IslandRGB/STATUS5 = {"StatusNET":{"Hostname":"IslandRGB","IPAddress":"172.16.20.31","Gateway":"172.16.20.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.20.1","Mac":"A0:20:A6:1A:B1:11","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:13:34.900 MQT: stat/IslandRGB/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.20.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1AB111","MqttUser":"hass","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:13:34.906 MQT: stat/IslandRGB/STATUS7 = {"StatusTIM":{"UTC":"2021-05-18T18:13:34","Local":"2021-05-18T19:13:34","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:04","Sunset":"20:28"}}
19:13:34.912 MQT: stat/IslandRGB/STATUS10 = {"StatusSNS":{"Time":"2021-05-18T19:13:34"}}
19:13:34.921 MQT: stat/IslandRGB/STATUS11 = {"StatusSTS":{"Time":"2021-05-18T19:13:34","Uptime":"0T00:21:30","UptimeSec":1290,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":56,"Color":"35,143,20,0,0","HSBColor":"113,86,56","White":0,"CT":153,"Channel":[14,56,8,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:Flash 9.4 and set RGB output <100%

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
steady output at desired level

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
none

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-05-18:
Please, can you share the output of the console? Please, remember to set weblog to 4 before your tests in order to have more debug information. Thanks

blackdiamondsoftware, 2021-05-18:
I rolled back to the previous firmware to solve issue i will load 9.4 again and get that log

blackdiamondsoftware, 2021-05-18:
Console Output below, after taking a look at this it seems to be hanging and re-booting

00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F4, Count 536
00:00:00.055 QPC: Count 1
00:00:00.060 SRC: Restart
00:00:00.066 Project tasmota IslandRGB Version 9.4.0(tasmota)-2_7_4_9(2021-04-23T10:07:22)
00:00:00.179 WIF: Checking connection...
00:00:00.180 WIF: Attempting connection...
00:00:00.516 WIF: Connecting to AP1 inFused Channel 11 BSSId 02:EC:DA:3E:EC:0F in mode 11n as IslandRGB...
00:00:01.754 WIF: Checking connection...
00:00:01.755 WIF: Connected
00:00:02.008 HTP: Web server active on IslandRGB with IP address 172.16.20.31
00:00:02.577 RTC: UTC 2021-05-18T18:30:42, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
19:30:43.005 MQT: Attempting connection...
19:30:43.049 MQT: Connected
19:30:43.052 MQT: tele/IslandRGB/LWT = Online (retained)
19:30:43.054 MQT: cmnd/IslandRGB/POWER = 
19:30:43.055 MQT: Subscribe to cmnd/IslandRGB/#
19:30:43.058 MQT: Subscribe to cmnd/tasmotas/#
19:30:43.061 MQT: Subscribe to cmnd/DVES_1AB111_fb/#
19:30:43.063 MQT: Unsubscribe from homeassistant/status
19:30:43.068 MQT: tele/IslandRGB/INFO1 = {"Info1":{"Module":"H801","Version":"9.4.0(tasmota)","FallbackTopic":"cmnd/DVES_1AB111_fb/","GroupTopic":"cmnd/tasmotas/"}}
19:30:43.074 MQT: tele/IslandRGB/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"IslandRGB","IPAddress":"172.16.20.31"}}
19:30:43.086 MQT: tele/IslandRGB/INFO3 = {"Info3":{"RestartReason":"Hardware Watchdog"}}
19:30:43.094 MQT: stat/IslandRGB/RESULT = {"POWER":"ON"}
19:30:43.098 MQT: stat/IslandRGB/POWER = ON
19:30:44.087 SRC: MQTT
19:30:44.089 CMD: Group 0, Index 1, Command "STATE", Data ""
19:30:44.096 MQT: tele/IslandRGB/STATE = {"Time":"2021-05-18T19:30:44","Uptime":"0T00:00:06","UptimeSec":6,"Heap":29,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":59,"MqttCount":1,"POWER":"ON","Dimmer":74,"Color":"22,189,0,0,0","HSBColor":"113,100,74","White":0,"CT":153,"Channel":[9,74,0,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:30:44.131 MQT: stat/IslandRGB/RESULT = {"Time":"2021-05-18T19:30:44","Uptime":"0T00:00:06","UptimeSec":6,"Heap":29,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":59,"MqttCount":1,"POWER":"ON","Dimmer":74,"Color":"22,189,0,0,0","HSBColor":"113,100,74","White":0,"CT":153,"Channel":[9,74,0,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:30:46.477 QPC: Reset
19:30:46.872 HTP: Console
19:30:47.418 MQT: tele/IslandRGB/STATE = {"Time":"2021-05-18T19:30:47","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":92,"MqttCount":1,"POWER":"ON","Dimmer":74,"Color":"22,189,0,0,0","HSBColor":"113,100,74","White":0,"CT":153,"Channel":[9,74,0,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:30:48.481 APP: Boot Count 65
19:30:48.597 CFG: Saved to flash at FB, Count 537, Bytes 4096
19:30:52.422 MQT: tasmota/discovery/A020A61AB111/config = {"ip":"172.16.20.31","dn":"IslandRGB","fn":["Island RGB",null,null,null,null,null,null,null],"hn":"IslandRGB","mac":"A020A61AB111","md":"H801","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.4.0","t":"IslandRGB","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[2,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":1,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":5,"sho":[0,0,0,0],"ver":1} (retained)
19:30:52.428 MQT: tasmota/discovery/A020A61AB111/sensors = {"sn":{"Time":"2021-05-18T19:30:52"},"ver":1} (retained)
19:31:01.753 WIF: Checking connection...

blackdiamondsoftware, 2021-05-18:
when it blips the Web UI stops responding and the device restarts

ascillato2, 2021-05-18:
when it restarts, what is the new output of the console (mainly the restart reason)?

blackdiamondsoftware, 2021-05-18:
dont really see much of a reason

00:00:00.001 HDW: ESP8266EX
00:00:00.049 CFG: Loaded from flash at F8, Count 564
00:00:00.055 QPC: Count 1
00:00:00.060 SRC: Restart
00:00:00.066 Project tasmota IslandRGB Version 9.4.0(tasmota)-2_7_4_9(2021-04-23T10:07:22)
00:00:00.178 WIF: Checking connection...
00:00:00.178 WIF: Attempting connection...
00:00:00.515 WIF: Connecting to AP1 inFused Channel 11 BSSId 02:EC:DA:3E:EC:0F in mode 11n as IslandRGB...
00:00:01.758 WIF: Checking connection...
00:00:01.758 WIF: Connected
00:00:02.011 HTP: Web server active on IslandRGB with IP address 172.16.20.31
00:00:02.504 RTC: UTC 2021-05-18T19:15:13, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
20:15:14.002 MQT: Attempting connection...
20:15:14.014 MQT: Connected
20:15:14.017 MQT: tele/IslandRGB/LWT = Online (retained)
20:15:14.020 MQT: cmnd/IslandRGB/POWER = 
20:15:14.022 MQT: Subscribe to cmnd/IslandRGB/#
20:15:14.025 MQT: Subscribe to cmnd/tasmotas/#
20:15:14.027 MQT: Subscribe to cmnd/DVES_1AB111_fb/#
20:15:14.030 MQT: Unsubscribe from homeassistant/status
20:15:14.035 MQT: tele/IslandRGB/INFO1 = {"Info1":{"Module":"H801","Version":"9.4.0(tasmota)","FallbackTopic":"cmnd/DVES_1AB111_fb/","GroupTopic":"cmnd/tasmotas/"}}
20:15:14.043 MQT: tele/IslandRGB/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"IslandRGB","IPAddress":"172.16.20.31"}}
20:15:14.054 MQT: tele/IslandRGB/INFO3 = {"Info3":{"RestartReason":"Hardware Watchdog"}}
20:15:14.060 MQT: stat/IslandRGB/RESULT = {"POWER":"ON"}
20:15:14.064 MQT: stat/IslandRGB/POWER = ON
20:15:15.064 SRC: MQTT
20:15:15.066 CMD: Group 0, Index 1, Command "STATE", Data ""
20:15:15.073 MQT: tele/IslandRGB/STATE = {"Time":"2021-05-18T20:15:15","Uptime":"0T00:00:06","UptimeSec":6,"Heap":28,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":62,"MqttCount":1,"POWER":"ON","Dimmer":36,"Color":"92,23,0,0,0","HSBColor":"15,100,36","White":0,"CT":153,"Channel":[36,9,0,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:15:15.109 MQT: stat/IslandRGB/RESULT = {"Time":"2021-05-18T20:15:15","Uptime":"0T00:00:06","UptimeSec":6,"Heap":28,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":62,"MqttCount":1,"POWER":"ON","Dimmer":36,"Color":"92,23,0,0,0","HSBColor":"15,100,36","White":0,"CT":153,"Channel":[36,9,0,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:15:17.479 QPC: Reset
20:15:18.414 MQT: tele/IslandRGB/STATE = {"Time":"2021-05-18T20:15:18","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":77,"MqttCount":1,"POWER":"ON","Dimmer":36,"Color":"92,23,0,0,0","HSBColor":"15,100,36","White":0,"CT":153,"Channel":[36,9,0,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:15:19.413 APP: Boot Count 78
20:15:19.596 CFG: Saved to flash at F7, Count 565, Bytes 4096

ascillato, 2021-05-18:
Your restart reason is:

```lua
MQT: tele/IslandRGB/INFO3 = {"Info3":{"RestartReason":"Hardware Watchdog"}}
```

Please, can you try latest Tasmota from the development branch ( http://ota.tasmota.com/tasmota/tasmota.bin.gz ) ?

blackdiamondsoftware, 2021-05-18:
Console log from Dev Branch 9.4.0.4 below, this seems to have resolved this issue.  I will keep testing and will post results in a few hours if there still seems to be an issue but i think this is resolved.

21:16:25.428 CMD: Group 0, Index 1, Command "BACKLOG", Data "NoDelay;Power1 ON;NoDelay;Color2 255,255,255"
21:16:25.433 SRC: Backlog
21:16:25.436 CMD: Group 0, Index 1, Command "POWER", Data "ON"
21:16:25.445 MQT: tele/IslandRGB/STATE = {"Time":"2021-05-18T21:16:25","Uptime":"0T00:04:18","UptimeSec":258,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":54,"Color":"103,1,138,0,0","HSBColor":"285,100,54","White":0,"CT":153,"Channel":[40,1,54,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:16:25.485 MQT: stat/IslandRGB/RESULT = {"POWER":"ON"}
21:16:25.489 MQT: stat/IslandRGB/POWER = ON
21:16:25.495 SRC: Backlog
21:16:25.498 CMD: Group 0, Index 2, Command "COLOR", Data "255,255,255"
21:16:25.509 MQT: tele/IslandRGB/STATE = {"Time":"2021-05-18T21:16:25","Uptime":"0T00:04:18","UptimeSec":258,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":105,"MqttCount":1,"POWER":"ON","Dimmer":54,"Color":"138,138,138,0,0","HSBColor":"0,0,54","White":0,"CT":153,"Channel":[54,54,54,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:16:25.550 MQT: stat/IslandRGB/RESULT = {"POWER":"ON","Dimmer":54,"Color":"138,138,138,0,0","HSBColor":"0,0,54","White":0,"CT":153,"Channel":[54,54,54,0,0]}
21:16:26.096 CFG: Saved to flash at F7, Count 594, Bytes 4096
21:16:31.257 WIF: Checking connection...
21:16:51.253 WIF: Checking connection...
21:17:11.260 WIF: Checking connection...
21:17:16.430 MQT: tele/IslandRGB/STATE = {"Time":"2021-05-18T21:17:16","Uptime":"0T00:05:09","UptimeSec":309,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":54,"Color":"138,138,138,0,0","HSBColor":"0,0,54","White":0,"CT":153,"Channel":[54,54,54,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:17:31.260 WIF: Checking connection...
21:17:36.926 SRC: MQTT
21:17:36.930 CMD: Group 0, Index 1, Command "BACKLOG", Data "NoDelay;Power1 ON;NoDelay;Color2 186,0,255"
21:17:36.936 SRC: Backlog
21:17:36.939 CMD: Group 0, Index 1, Command "POWER", Data "ON"
21:17:36.951 MQT: tele/IslandRGB/STATE = {"Time":"2021-05-18T21:17:36","Uptime":"0T00:05:29","UptimeSec":329,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":54,"Color":"138,138,138,0,0","HSBColor":"0,0,54","White":0,"CT":153,"Channel":[54,54,54,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:17:36.989 MQT: stat/IslandRGB/RESULT = {"POWER":"ON"}
21:17:36.994 MQT: stat/IslandRGB/POWER = ON
21:17:36.999 SRC: Backlog
21:17:36.002 CMD: Group 0, Index 2, Command "COLOR", Data "186,0,255"
21:17:37.001 MQT: tele/IslandRGB/STATE = {"Time":"2021-05-18T21:17:36","Uptime":"0T00:05:29","UptimeSec":329,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":105,"MqttCount":1,"POWER":"ON","Dimmer":54,"Color":"101,0,138,0,0","HSBColor":"284,100,54","White":0,"CT":153,"Channel":[40,0,54,0,0],"Scheme":0,"Fade":"OFF","Speed":20,"LedTable":"ON","Wifi":{"AP":1,"SSId":"inFused","BSSId":"02:EC:DA:3E:EC:0F","Channel":11,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:17:37.042 MQT: stat/IslandRGB/RESULT = {"POWER":"ON","Dimmer":54,"Color":"101,0,138,0,0","HSBColor":"284,100,54","White":0,"CT":153,"Channel":[40,0,54,0,0]}
21:17:37.134 CFG: Saved to flash at F6, Count 595, Bytes 4096
21:17:51.258 WIF: Checking connection...
21:17:52.463 HTP: Main Menu
21:17:54.122 HTP: Console

ascillato2, 2021-05-21:
Glad to know that your issue is solved.

closed 2021-05-21 by ascillato2