arendst/Tasmota #12132: STATUS6 doesn't show MQTT SSL

jeroenst, 2021-05-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

When I request status info the state of using ssl/tls is missing from STATUS6

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff zbbridge
- [x] Tasmota binary firmware version number used: 94.0.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

send CMD STATUS 0 with MQTT to Tasmota and see result.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

STATUS6 should show the ssl/tls state on mqtt

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-05-18:
> TO REPRODUCE
> Steps to reproduce the behavior:
> send CMD STATUS 0 with MQTT to Tasmota and see result

Please, can you share what is the result you are obtaining and which is the result you expect?

jeroenst, 2021-05-18:
Of course, 
This is the current result:
home/TASMOTA_ZIGBEE/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BA274","MqttUser":"jeroen","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}

I would expect:
home/TASMOTA_ZIGBEE/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxx","MqttPort":1883,"MqttSSLt":0,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BA274","MqttUser":"jeroen","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}

ascillato, 2021-05-18:
Ah, ok. That is not implemented. So, this is a feature request right?

jeroenst, 2021-05-18:
I think it's more a bug than a feature request but a minor one.

ascillato2, 2021-05-18:
Moving to Ideas in Discussions.

Thanks.

closed 2021-05-18 by ascillato2