arendst/Tasmota #12134: ZMAI-90 with Tasmota 9.4.0.4 wont shown energy yesterday

testpaul999, 2021-05-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
- ZMAI-90 with Tasmota 9.4.0.4 wont shown energy yesterday
- have set setoption72 0 and see only the toltal and today kWh

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _ZMAI-90____
- [ x] Tasmota binary firmware version number used: __9.4.0.4 sensors___
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: _OTA____
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:03:01.458 CMD: Backlog Template; Module; GPIO 255;
08:03:01.460 SRC: WebConsole from 192.168.2.247
08:03:01.463 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255;"
08:03:02.140 MQT: tele/Solar_D41136/RESULT = {"TuyaReceived":{"Data":"55AA0302000004","Cmnd":2}}
08:03:02.147 TYA: RX MCU configuration Mode=0

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
08:00:33.800 CMD: Backlog Rule1; Rule2; Rule3
08:00:33.803 SRC: WebConsole from 192.168.2.247
08:00:33.806 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
08:00:34.320 SRC: Backlog
08:00:34.324 CMD: Group 0, Index 3, Command "RULE", Data ""
08:00:34.334 MQT: stat/Solar_D41136/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":9,"Free":502,"Rules":"`: ```lua"}}
08:00:34.686 RUL: RULES#TIMER=1 performs "backlog TuyaSend8; RuleTimer1 2"
08:00:34.689 SRC: Rule
08:00:34.692 CMD: Group 0, Index 1, Command "BACKLOG", Data "TuyaSend8; RuleTimer1 2"
08:00:34.721 SRC: Backlog
08:00:34.724 CMD: Group 0, Index 8, Command "TUYASEND", Data ""
```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
07:58:42.973 CMD: status 0
07:58:42.976 SRC: WebConsole from 192.168.2.247
07:58:42.980 CMD: Group 0, Index 1, Command "STATUS", Data "0"
07:58:42.986 MQT: stat/Solar_D41136/STATUS = {"Status":{"Module":54,"DeviceName":"SolarStrom","FriendlyName":["SolarStrom"],"Topic":"Solar_D41136","ButtonTopic":"Solar_D41136","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
07:58:43.001 MQT: stat/Solar_D41136/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T02:33:28","StartupUTC":"2021-05-19T03:25:14","Sleep":50,"CfgHolder":4617,"BootCount":1585,"BCResetTime":"2021-05-14T18:14:37","SaveCount":1698,"SaveAddress":"F4000"}}
07:58:43.014 MQT: stat/Solar_D41136/STATUS2 = {"StatusFWR":{"Version":"9.4.0.4(sensors)","BuildDateTime":"2021-05-17T08:35:09","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"345/699"}}
07:58:43.027 MQT: stat/Solar_D41136/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["1JUF2",""],"TelePeriod":10,"Resolution":"551500C0","SetOption":["0080800B","2805C8000100060000005A0A000000000000","00810280","00006000","00000000"]}}
07:58:43.046 MQT: stat/Solar_D41136/STATUS4 = {"StatusMEM":{"ProgramSize":656,"Free":344,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
07:58:43.062 MQT: stat/Solar_D41136/STATUS5 = {"StatusNET":{"Hostname":"Solar_D41136-4406","IPAddress":"192.168.2.239","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"3C:61:05:D4:11:36","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
07:58:43.075 MQT: stat/Solar_D41136/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.247","MqttPort":1883,"MqttClientMask":"Solar_%06X","MqttClient":"Solar_D41136","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:58:43.088 MQT: stat/Solar_D41136/STATUS7 = {"StatusTIM":{"UTC":"2021-05-19T05:58:43","Local":"2021-05-19T07:58:43","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:03","Sunset":"21:30"}}
07:58:43.099 MQT: stat/Solar_D41136/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
07:58:43.115 MQT: stat/Solar_D41136/STATUS10 = {"StatusSNS":{"Time":"2021-05-19T07:58:43","ENERGY":{"TotalStartTime":"2021-05-18T07:40:35","Total":0.08,"Yesterday":0.00,"Today":0.08,"Power":0.00,"ApparentPower":0.00,"ReactivePower":0.00,"Factor":0.00,"Voltage":236.50,"Current":0.00}}}
07:58:43.129 MQT: stat/Solar_D41136/STATUS11 = {"StatusSTS":{"Time":"2021-05-19T07:58:43","Uptime":"0T02:33:29","UptimeSec":9209,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"1JUF2","BSSId":"DC:15:C8:BB:47:52","Channel":6,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}}
07:58:43.729 RUL: RULES#TIMER=1 performs "backlog TuyaSend8; RuleTimer1 2"
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
08:04:53.132 TYA: fnId=32 is set for dpId=18
08:04:53.692 MQT: tele/Solar_D41136/STATE = {"Time":"2021-05-19T08:04:53","Uptime":"0T02:39:39","UptimeSec":9579,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"1JUF2","BSSId":"DC:15:C8:BB:47:52","Channel":6,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}
08:04:53.706 MQT: tele/Solar_D41136/SENSOR = {"Time":"2021-05-19T08:04:53","ENERGY":{"TotalStartTime":"2021-05-18T07:40:35","Total":0.08,"Yesterday":0.00,"Today":0.08,"Period":0.00,"Power":0.00,"ApparentPower":0.00,"ReactivePower":0.00,"Factor":0.00,"Voltage":236.30,"Current":0.00}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
see this every day on count yesterday energy

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
in my mind the must at day change the file today energy switch to the filed yesterday and reset to "0"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
the same problem are there with setoption72 1

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-05-19:
Hi,

Setoption72 is just for choosing to use the energy total stored in the power measurement chip or to use the energy total counted by Tasmota.

When you use the one from Tasmota, your device must be online at midnight in order to save the energy today and put it on energy yesterday.

If your device is restarting, it will never save the values in the flash. To prevent flash wear out, Tasmota only saves the energy values at midnight.

From your logs, your device uptime is just 2 hours, so, please, do not restart your device and leave it measuring for more than 24 hours.

If you need further assistance on how to setup your device, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-05-19 by ascillato2

testpaul999, 2021-05-19:
Hi, now way have run this the whole night and have this result.😩
Before I write this issue.
Come back tomorrow and post the same issue again.

Viele Grüße

Jens-Uwe Franz

ascillato, 2021-05-19:
Try to set a syslog to catch what is happening. Your device is rebooting for some reason.

testpaul999, 2021-05-19:
👍🏻 Done

Viele Grüße

Jens-Uwe Franz

testpaul999, 2021-05-19:
Dose Tasmota reboot by itself if wifi is off at night?

Viele Grüße

Jens-Uwe Franz

testpaul999, 2021-05-19:
And only for Dokumentation screenshot from today
[cid:0BD9EFDA-2BE0-4FF1-855E-37E5A0EDAFAC-L0-001]
At 21:25

Viele Grüße

Jens-Uwe Franz

ascillato, 2021-05-19:
That's it. You nail it.

You have wificonfig set to 2 (in your status 0)

So that is restart if there is no Wi-Fi

You should use the default `wificonfig 4` that is retry connection without restarting.

testpaul999, 2021-05-20:
👍🏻 Have change and wait one more day.
Thx for help

Viele Grüße

Jens-Uwe Franz