arendst/Tasmota #12161: mqtt passwords

Bernd9000, 2021-05-22:
### PROBLEM DESCRIPTION
Four-digit passwords are no longer accepted in the MQTT settings
There must be at least five 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): all
- [ ] Tasmota binary firmware version number used: 9.4.0.4
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: all
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:35:55.044 MQT: stat/Steckdose-3/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:35:55.253 MQT: stat/Steckdose-3/RESULT = {"Module":{"1":"Sonoff Basic"}}
17:35:55.507 MQT: stat/Steckdose-3/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
17:36:44.163 MQT: stat/Steckdose-3/STATUS = {"Status":{"Module":1,"DeviceName":"Steckdose-3","FriendlyName":["Steckdose-3"],"Topic":"Steckdose-3","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:36:44.169 MQT: stat/Steckdose-3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-DE.bin.gz","RestartReason":"Software/System restart","Uptime":"2T17:36:39","StartupUTC":"2021-05-19T23:00:05","Sleep":50,"CfgHolder":4617,"BootCount":88,"BCResetTime":"2021-04-23T09:00:25","SaveCount":282,"SaveAddress":"F5000"}}
17:36:44.174 MQT: stat/Steckdose-3/STATUS2 = {"StatusFWR":{"Version":"9.4.0.4(tasmota)","BuildDateTime":"2021.05.19 15:25:44","Boot":6,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"336/699"}}
17:36:44.178 MQT: stat/Steckdose-3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Zentrale",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
17:36:44.188 MQT: stat/Steckdose-3/STATUS4 = {"StatusMEM":{"ProgramSize":604,"Free":396,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:36:44.197 MQT: stat/Steckdose-3/STATUS5 = {"StatusNET":{"Hostname":"Steckdose-3-3479","IPAddress":"192.168.0.122","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"68:C6:3A:D5:AD:97","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:36:44.202 MQT: stat/Steckdose-3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D5AD97","MqttUser":"mqtt","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:36:44.209 MQT: stat/Steckdose-3/STATUS7 = {"StatusTIM":{"UTC":"2021-05-22T16:36:44","Local":"2021-05-22T17:36:44","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:59","Sunset":"20:33"}}
17:36:44.215 MQT: stat/Steckdose-3/STATUS10 = {"StatusSNS":{"Time":"2021-05-22T17:36:44"}}
17:36:44.221 MQT: stat/Steckdose-3/STATUS11 = {"StatusSTS":{"Time":"2021-05-22T17:36:44","Uptime":"2T17:36:39","UptimeSec":236199,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"ON","Wifi":{"AP":1,"SSId":"Zentrale","BSSId":"C8:0E:14:0D:16:DA","Channel":6,"RSSI":80,"Signal":-60,"LinkCount":2,"Downtime":"0T00:00:08"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-05-22:
Correct

ascillato2, 2021-05-22:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-05-22 by ascillato2

Bernd9000, 2021-05-22:
Where can I find the conditions for a mqtt password?
I didn't find that in the Docs.
Thank you

Jason2866, 2021-05-24:
Documentation is for release version.