arendst/Tasmota #12169: Makegood MG-AUWF01 - Small corrections and questions

victorhooi, 2021-05-22:
### PROBLEM DESCRIPTION
Device template for Makegood MG-AUWF01  contains some small errors.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Makegood MG-AUWF01 
- [X] Tasmota binary firmware version number used: 9.4.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tuya-convert for initial, then used Tasmota Web UI to upgrade to latest
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:19:48.739 CMD: Backlog Template; Module; GPIO 255
00:19:48.764 RSL: RESULT = {"NAME":"MG-AUWF01","GPIO":[320,161,544,323,2720,2656,0,0,2624,225,321,224,160,0],"FLAG":0,"BASE":18}
00:19:48.987 RSL: RESULT = {"Module":{"0":"MG-AUWF01"}}
00:19:49.241 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"161":"Switch2"},"GPIO2":{"544":"LedLink"},"GPIO3":{"323":"Led_i4"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"225":"Relay2"},"GPIO14":{"321":"Led_i2"},"GPIO15":{"224":"Relay1"},"GPIO16":{"160":"Switch1"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
N/A

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:20:12.514 CMD: Status 0
00:20:12.521 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_FD72E0","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[5,5,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:20:12.529 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:21:49","StartupUTC":"2021-05-22T22:58:23","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2021-05-22T23:48:04","SaveCount":24,"SaveAddress":"F4000"}}
00:20:12.536 RSL: STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021-04-23T10:07:22","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"321/699"}}
00:20:12.541 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MIMO",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
00:20:12.549 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:20:12.555 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_FD72E0-4832","IPAddress":"10.5.1.187","Gateway":"10.5.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.5.1.1","Mac":"8C:AA:B5:FD:72:E0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:20:12.560 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FD72E0","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:20:12.567 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-05-22T23:20:12","Local":"2021-05-23T00:20:12","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:58","Sunset":"20:34"}}
00:20:12.572 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:20:12.580 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-05-23T00:20:12","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2021-05-22T23:48:04","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":8,"ReactivePower":8,"Factor":0.05,"Voltage":290,"Current":0.027}}}
00:20:12.586 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-05-23T00:20:12","Uptime":"0T00:21:49","UptimeSec":1309,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"MIMO","BSSId":"00:4E:35:13:86:20","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Use the on-screen web UI to toggle the power plugs.

### EXPECTED BEHAVIOUR
The left on-screen control should correspond to the left-physical socket, and the right on-screen control should correspond to the right physical socket.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

I recently purchased some smart powerpoint (Makegood MG-AUWF01), based on the following Tasmota template:

https://templates.blakadder.com/makegood_MG-AUWF01.html

First thing - the page lists two Aliexpress links given there to purchase the item. I purchased units from both links.

The [first link](https://www.aliexpress.com/item/4000252892030.html) is actually for a different item - the seller is a little shady/deceptive when you talk to them, but it's basically just a touch-button wall socket, with no smart functionality at all.

Is it possible to add a note here, or even remove from the wiki entry?

The [second link](https://www.aliexpress.com/i/4000224808833.html) I can confirm does appear to be the MG-AUWF01, with smart functionality.

As of units purchased at May 2021, I was able to flash these using [tuya-convert](https://github.com/ct-Open-Source/tuya-convert).

On the control board, there is the version string `REV1.2-211019`. The chip on there is a `TYWE3S`.

[image]

[image]

Second thing, the wiki refers to having to set `Gpio03` low - for a Tasmota newbie like me, I didn't realise this is already done as part of the template. However, you do still need to apply the SwitchMode comments. Might be worth adding a small sentence in the device wiki entry to clarify?

Third thing - I have applied the template from the device wiki page, using the instructions here.

I then ran the two commands listed on the wiki:
```
00:01:37.633 CMD: SwitchMode1 5
00:01:37.638 RSL: RESULT = {"SwitchMode1":5}
00:01:46.562 CMD: SwitchMode2 5
00:01:46.567 RSL: RESULT = {"SwitchMode2":5}
```
One thing that is weird - in the web interface afterwards, the left UI toggle corresponds to the right physical power point, and the right UI toggle coresponds to the left physical power point.

Is there any way to flip these around, so it's correct and matches the UI?

Finally - for future reference - there's some discussion on flashing MakeGood devices via serial headers [here](https://github.com/sillyfrog/contact/issues/4) - might be useful in future, if they somehow change it so it's harder to flash with tuya-convert.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-05-23:
Please avoid cross-posting. Gave an answer in #12170

Jason2866, 2021-05-23:
In any case not a Tasmota issue. Please close issue.

ascillato2, 2021-05-23:
Closing in favour of #12170

closed 2021-05-23 by ascillato2