arendst/Tasmota #12203: Security: Tasmota & ESP32-CAM stream is not covered by password protection

samazw, 2021-05-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The Tasmota version for ESP32-Camera module has the option "set password".
But this password protection does not cover the direct link to the video stream.

Meaning: The password protect the webinterface of tasmota itself but not http://DEVICE_IP:81/stream.
So correct would be the need of user+password also for the stream.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

martink1979, 2021-08-21:
Gibt es schon eine Lösung für dieses Problem ?  Das die Stream URL nicht mit einem Passwort geschützt ist ?

sfromis, 2021-08-21:
It is not a "problem", merely a wish for an enhancement.

martink1979, 2021-08-23:
> It is not a "problem", merely a wish for an enhancement.

Gibt es denn schon einen workaround dazu ? Das man den stream irgendwie schützen kann ?

ascillato2, 2021-08-23:
> GOOGLE TRANSLATE: Is there already a workaround for this? That you can somehow protect the stream?

Not at this moment, that is why this issue is open to anyone that want to jump in and make a PR.