arendst/Tasmota #12207: Osram Zigbee Device doesn't remember state

spcqike, 2021-05-27:
### PROBLEM DESCRIPTION
Osram Smart+ bulbs don't remember state (dimmer value and on/off) after power loss.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Osram Smart+ (https://www.amazon.de/dp/B074KJ72MP/) (https://zigbee.blakadder.com/OSRAM_4058075816510.html)
- [x] Tasmota binary firmware version number used: 9.4.0 zbbridge
  - [x] Pre-compiled
  - [ ] Self-compiled

### TO REPRODUCE
switch light on (main power), dimm to value x, switch off (main power), switch on again.

### EXPECTED BEHAVIOUR
keep dimmer value

### ADDITIONAL CONTEXT
Osram seems do have an option to save the present values. so that the bulb starts with these values after power loss. (like mentioned here https://www.digitalzimmer.de/artikel/news/lightify-update-osram-lampen-speichern-ihre-einstellung/ and here https://www.zigbee2mqtt.io/devices/AC03645.html)
I'm not sure, if it works with mine particular bulb. anyway it would be nice for one or the other if we could add "remember_state" as command. (and maybe "set_transition", as mentioned in zigbee2mqtt)

**(Please, remember to close the issue when the problem has been addressed)**

spcqike, 2021-06-02:
hi,

i found some more code/info for my problem within the zigbee-herdsman code.
(https://github.com/Koenkk/zigbee-herdsman-converters/blob/a9a26353842e1f5507b70359fc239205f838b857/converters/toZigbee.js row 1884+, https://github.com/Koenkk/zigbee-herdsman/blob/c48583ad04699e34bc4cbc055c715f2c3006ad65/src/zcl/definition/cluster.ts row 3754+)

as we can send zigbee low-level commands and i guess this is a rather specific use case, maybe someone can point me towards the right direction or give me the corresponding low-level code, as i cant "read" these two files... i dont find useable code/commands :(

:)

s-hadinger, 2021-06-02:
That's doable. But I can't help you right now

s-hadinger, 2021-06-28:
Bump to keep it open