arendst/Tasmota #12208: Tasmota Lost aftert Sleep 0 set

LarryB49, 2021-05-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After Sleep 0 command after reboot up Device becomes unreachable

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ___Sonoff Basic__
- [ x] Tasmota binary firmware version number used: _9.4 I tested and same with 9.1____
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: ___esptool.py v3.0__
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
Can not do since unreachable

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

N/A
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

Can not do since unreachable
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Can not do since unreachable

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Enter command Sleep 0.  On next boot device LED in blinking state.  The device is not at last ip (I have a router reservation, static IP). Can not find device on network. In wireless a device with SSID of ESP-with last 6 characters of Mac address shows available from my phone.  But this device will not accept a connection.  Only fix I have found is to erase flash and then reflash with Tasmota and start over. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Why would setting sleep off cause device to no longer work?  I was turning sleep off thinking it would make state changes report faster.  Is Sleep 0 a bad idea?

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-05-27:
Just tested on a device, setting Sleep 0 and restarting. Came up and online no problem, still with the Sleep 0 setting active. 

If you judge state change reporting speed looking at the console, don't. This can and will often be a bit behind. You can get better indications by watching MQTT, or of course any switched outputs.

ascillato2, 2021-05-28:
Sleep 0 make your device to use more energy. Sonoff Basics are known to have a very poor power supply. So you are killing your device with overpower.

You should use the default sleep value for a Sonoff Basic. If you continue having issues (meaning that you have one of the known batch of faulty ones) you should use Tasmota-lite.bin instead to lower even more the CPU power consumption.

closed 2021-05-28 by ascillato2