arendst/Tasmota #12218: DS18B20 sensor on ESP32 not forced to 12bit resolution like on ESP8266

serg-ku, 2021-05-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
DS18B20 sensor on ESP32 not forced to 12bit resolution like on ESP8266

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32
- [x] Tasmota binary firmware version number used: v9.4.0 Leslie
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP-Flasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect 18B20 sensor initially configured to <12bits resolution to tasmota

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
In commit https://github.com/arendst/Tasmota/commit/512566c7278ebaf62173c9358e1ae43d5cba50b8 explicit 12bit resolution setting was introduced, we can see it here: https://github.com/arendst/Tasmota/blob/ed5b048b6ee2c69c1d3ea7fa1183b94e348ce5b3/tasmota/xsns_05_ds18x20.ino#L378-L379

But if i understand correctly ESP32 uses another 18x20 implementation which is in file https://github.com/arendst/Tasmota/blob/ed5b048b6ee2c69c1d3ea7fa1183b94e348ce5b3/tasmota/xsns_05_ds18x20_esp32.ino and it doesn't have 12bit setting in Ds18x20Read() function as I can see.

I see this issue in Tasmota v9.4.0 Leslie.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-05-29:
Looking at this https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_05_ds18x20_esp32.ino#L129
i think it is 12 bit resolution
Compared to this https://github.com/arendst/Tasmota/blob/development/lib/lib_basic/OneWire-Stickbreaker-20190506-1.1/examples/DS18x20_Temperature/DS18x20_Temperature.pde#L86-L112

serg-ku, 2021-05-29:
If I understand correctly 
https://github.com/arendst/Tasmota/blob/9ab38d8452f6eda7401173927e832168e671f996/tasmota/xsns_05_ds18x20_esp32.ino#L128-L129
this part you mentioned is for DS18**S**20 with id 0x10 (also interesting what's the reason to set 12bit on DS18**S**20, which is 9bit by specs).
For DS18**B**20 (which is much more common) with id 0x28 code starts from
https://github.com/arendst/Tasmota/blob/9ab38d8452f6eda7401173927e832168e671f996/tasmota/xsns_05_ds18x20_esp32.ino#L135
and it doesn't have 12bit setting like I mentioned in esp8266 code.

sfromis, 2021-05-29:
While that conversion does indeed seem to support 12-bit (and I'm getting DS18B20 readings on ESP32 with delta values of 0.06°C), the point was about a sensor initially configured to < 12 bit not being "forced" to 12 bit by the Tasmota driver, like what is done in the ESP8266 version of the driver.

serg-ku, 2021-05-29:
@sfromis you are right, if DS18B20 is set to 12bit using arduino it outputs 12bit values in tasmota too.

closed 2021-06-04 by ascillato2