arendst/Tasmota #12221: Cross-site scripting vulnerability

TheRabbitX, 2021-05-28:
### PROBLEM DESCRIPTION
I've identified a Cross-site scripting vulnerability (XSS) in the web interface of Tasmota 6.5.0. Unfortunately, I can not check if the current release is also affected. It would be nice if someone could check this out and give me feedback.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Navigate to "Configuration" - "Configure Other" and insert the following sting in the field "Friendly Name 1":
"/><script>alert(1)</script>

After that, a JavaScript alert box should appear if the version is vulnerable.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-05-28:
Testing this on 9.4.0.4, pasting `"/><script>alert(1)</script>` into friendly name and hitting save, I see nothing changed. In the console I'm seeing `{"FriendlyName1":""}` instead of the `{"FriendlyName1":"asdfasdf"}` when actually updating friendly name in the form. If I instead use `/><script>alert(1)</script>` (without the initial `"` from your suggestion) I do get the alert when reentering "Configure Other" after the restart.

However, to me this does not look like much of a "real" problem. If someone hostile gets access to screwing with my Tasmota devices, I'd likely have much bigger problems than this "detail". I'd certainly not be exposing them to the Internet, meaning that someone would have to breach my network in other ways to play around with Tasmota devices. And then they need no cross-site scriptring stuff, as they'd have much more straightforward ways of messing with me.

Jason2866, 2021-05-29:
The ESP8266 is lacking any security feature by design. IMHO every ESP8266 device which is reachable from non authorized people or machines is a high security risk. A real hacker will find a way to break in a ESP8266 driven device. **NEVER use Tasmota in a not secured wifi environment.**
Keep this always in mind when using a ESP8266 device.

digiblur, 2021-06-07:
Tuya figured out the security for now.

Jason2866, 2021-06-07:
@digiblur No real hacker is working on to break Tuya again on a ESP8266 device.
No new designed TUYA device is with a ESP82xx. There are reasons for ;-)

digiblur, 2021-06-07:
Plenty of work has been done. No exploit found so far. The change of chipset isn't due to esp8266 security though.

sfromis, 2021-06-07:
For a long time, some Tuya vulnerabilities were found, and actively exploited, at least by Tuya-convert. Seems closed for newer iterations.

arendst, 2023-12-18:
Fixed in v13.3.0.1 https://github.com/arendst/Tasmota/commit/f65ae0694b99f454a0525bc387ef8cac0162b5d7