arendst/Tasmota #12229: PWM Dimmer Fade out works, but Fade in too fast (almost instant power on)

Tobias007, 2021-05-29:
### PROBLEM DESCRIPTION
PWM Fade/Speed option only works for fade out, but not fade in.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): MagicHome LED strip controller
- [ ] Tasmota binary firmware version number used: 9.4.0(knx) - 2_7_4_9/2.2.2-dev(38a443e)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: NodeMCU Flasher
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:52:16.848 MQT: stat/livingroom/dim3/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
23:52:17.056 MQT: stat/livingroom/dim3/RESULT = {"Module":{"0":"Generic"}}
23:52:17.266 MQT: stat/livingroom/dim3/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"416":"PWM1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
23:52:47.253 MQT: stat/livingroom/dim3/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":135,"Free":376,"Rules":"on Switch1#state=1 do Backlog Publish stat/%topic%/PIR1 ON; RuleTimer1 30 endon on Rules#Timer=1 do Publish stat/%topic%/PIR1 OFF endon"}}
23:52:47.461 MQT: stat/livingroom/dim3/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
23:52:47.666 MQT: stat/livingroom/dim3/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:54:11.701 MQT: stat/livingroom/dim3/STATUS = {"Status":{"Module":0,"DeviceName":"LR-Dim3","FriendlyName":["LR-Dim3"],"Topic":"livingroom/dim3","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"PIR","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:54:11.717 MQT: stat/livingroom/dim3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-knx.bin","RestartReason":"Software/System restart","Uptime":"7T09:13:44","StartupUTC":"2021-05-22T13:40:27","Sleep":50,"CfgHolder":4617,"BootCount":48,"BCResetTime":"2020-10-15T15:31:14","SaveCount":2963,"SaveAddress":"F6000"}}
23:54:11.733 MQT: stat/livingroom/dim3/STATUS2 = {"StatusFWR":{"Version":"9.4.0(knx)","BuildDateTime":"2021-04-23T10:07:18","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"403/699"}}
23:54:11.748 MQT: stat/livingroom/dim3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["D2Offline",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C8000100060000005A0A000000000000","00000080","00006200","00000000"]}}
23:54:11.766 MQT: stat/livingroom/dim3/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC587","04268011","000000CF","010013C0","8000F181","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,22,24,26,27,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
23:54:11.779 MQT: stat/livingroom/dim3/STATUS5 = {"StatusNET":{"Hostname":"LR-Dim3","IPAddress":"192.168.14.40","Gateway":"192.168.14.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.14.1","Mac":"10:52:1C:CC:D1:CB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:54:11.792 MQT: stat/livingroom/dim3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.14.144","MqttPort":1883,"MqttClientMask":"LR-Dim3","MqttClient":"LR-Dim3","MqttUser":"fhem","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:54:11.807 MQT: stat/livingroom/dim3/STATUS7 = {"StatusTIM":{"UTC":"2021-05-29T22:54:11","Local":"2021-05-29T23:54:11","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:53","Sunset":"20:41"}}
23:54:11.822 MQT: stat/livingroom/dim3/STATUS10 = {"StatusSNS":{"Time":"2021-05-29T23:54:11","Switch1":"OFF"}}
23:54:11.838 MQT: stat/livingroom/dim3/STATUS11 = {"StatusSTS":{"Time":"2021-05-29T23:54:11","Uptime":"7T09:13:44","UptimeSec":638024,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":80,"MqttCount":3,"POWER":"ON","Dimmer":53,"Fade":"ON","Speed":6,"LedTable":"ON","Wifi":{"AP":1,"SSId":"D2Offline","BSSId":"D6:FB:E4:C7:30:99","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":3,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
23:56:24.076 MQT: stat/livingroom/dim3/RESULT = {"WebLog":4}
23:56:24.726 CFG: Saved to flash at F5, Count 2964, Bytes 4096
23:56:26.181 HTP: Main Menu
23:56:28.153 SRC: WebGui from 192.168.14.142
23:56:28.160 MQT: stat/livingroom/dim3/RESULT = {"POWER":"OFF"}
23:56:28.165 MQT: stat/livingroom/dim3/POWER = OFF
23:56:30.683 SRC: WebGui from 192.168.14.142
23:56:30.689 MQT: stat/livingroom/dim3/RESULT = {"POWER":"ON"}
23:56:30.693 MQT: stat/livingroom/dim3/POWER = ON
23:56:32.594 HTP: Console
23:56:36.374 CMD: weblog 2

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Config:
23:59:19.576 CMD: fade
23:59:19.584 MQT: stat/livingroom/dim3/RESULT = {"Fade":"ON"}
23:59:23.480 CMD: speed
23:59:23.488 MQT: stat/livingroom/dim3/RESULT = {"Speed":6}

Via WebGui:
Dimmer is in state off: 
Press Toggle button -> light goes on quickly (within 0.5sec, very slight fade in visible)
Press Toggle button again -> light goes softly off (fade out within few seconds

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Fade in and out in same speed. (6 sec)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-05-30:
What is your value for `LedTable `?

Tobias007, 2021-05-30:
I tried both - first with LedTable = 1. (same behavior, very quick fade in, slow fade out).
With LedTable = 0, similar behavior, anyhow the transitions a bit smother, but still by far too fast fade in.

I also tried different LEDs, as I assumed it has something to do with the LED in the MR12 bulbs. Even removed the driver and replaced it with a simple resistor (as I have a clean, stable 12V DC power source).

s-hadinger, 2021-05-30:
Did you TRY `SetOption117 1`?

Tobias007, 2021-05-30:
Yes, tried it. But still same. Ramp up very very fast. Fade out perfectly slow.
When I dim manually with the slider, I could create a slow ramp up :) .. so it doesn't seem to be a hardware issue ..

Tobias007, 2021-05-30:
I tried as well another device and LED, just to be sure it's not the hardware ...

s-hadinger, 2021-05-30:
I won't have time to look at it for some days. In the meantime can you share a short video?

Tobias007, 2021-05-31:
Thanks for the amazing quick support and response.

Attached the video:

https://user-images.githubusercontent.com/37033298/120128296-45133600-c1f4-11eb-9013-74647ecfea37.mov

Interestingly in the video the ramp up is visible, but the difference of ramp up and down is obvious. In reality it looks even quicker/nearly instant light up. I assume a much longer delay time for the first few dim steps is required (instead of the table an exponential formula might be useful?).

sfromis, 2021-05-31:
Testing `Fade 1; Speed 10` with a bulb, I do see how power on fade-in is faster than power off fade-out. This is a good fit for my psychological expectation when hitting a switch, seeing some quick action instead of it initially barely visible. OTOH, when turning off, a slower slide to dark is preferable to me. The difference I see is the lower part of fade-in being fast-forwarded. Makes sense to me.

Tobias007, 2021-06-03:
yes, when using a switch  I agree it is beneficial if there is a faster ramp up. However when using light scenes a slower ramp up would be preferable. Excellent would be a setting to enforce a slow ramp up.

ascillato2, 2021-06-03:
> However when using light scenes a slower ramp up would be preferable.

Ok, moving this issue to a Feature Request in Discussions.

closed 2021-06-03 by ascillato2