arendst/Tasmota #12235: WifiManager leaves WifiConfig 0 instead of expected 4

barbudor, 2021-05-30:
### PROBLEM DESCRIPTION

Problem reported by @NsinghP in https://github.com/arendst/Tasmota/discussions/11845

WifiManager web portal set WifiConfig to 0

@ascillato I think you are the latest one who worked on the WifiManager portal. Would you have a look please ?

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): NodeMCU
- [X] Tasmota binary firmware version number used: 9.4.0.4
  - [W] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotizer 1.2, "erase flash" checked
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
Problem occurs even before a template/Module is configured
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
No rules yet set, fresh flashing
```
- [X] Provide the output of this command: `Status 0`:
```lua
15:46:50.417 CMD: status 0
15:46:50.423 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_2AD017","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:46:50.449 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:07:14","StartupUTC":"2021-05-30T14:39:36","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2021-05-30T15:39:45","SaveCount":5,"SaveAddress":"F7000"}}
15:46:50.482 RSL: STATUS2 = {"StatusFWR":{"Version":"9.4.0.4(tasmota)","BuildDateTime":"2021-05-30T09:34:24","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"370/699"}}
15:46:50.501 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BNETWLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
15:46:50.530 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":604,"Free":396,"Heap":28,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:46:50.561 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_2AD017-4119","IPAddress":"192.168.168.52","Gateway":"192.168.168.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.168.244","Mac":"3C:71:BF:2A:D0:17","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
15:46:50.583 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2AD017","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:46:50.606 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-05-30T14:46:50","Local":"2021-05-30T15:46:50","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:52","Sunset":"20:42"}}
15:46:50.624 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-05-30T15:46:50"}}
15:46:50.631 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-05-30T15:46:50","Uptime":"0T00:07:14","UptimeSec":434,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"88:C3:97:F2:25:EC","Channel":1,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
No requiered in this case
```

### TO REPRODUCE

Flashing tasmota development 9.4.0.4 using tasmotizer 1.2 on a ModeMCU with "Erase flash" checked
Then run Termite serial terminal
Once tasmota restart, WifiConfig is set to expected WIFI_RETRY (4)
```lua 
Serial console logs:
00:00:00.001 HDW: ESP8266EX
00:00:00.017 CFG: Use defaults
00:00:00.457 QPC: Reset
00:00:00.467 Project tasmota Tasmota Version 9.4.0.4(tasmota)-2_7_4_9(2021-05-30T09:34:24)
00:00:00.765 WIF: WifiManager active for 3 minutes
00:00:00.275 HTP: Web server active on tasmota_2AD017-4119 with IP address 192.168.4.1
wificonfig
00:00:21.966 CMD: wificonfig
00:00:21.970 RSL: RESULT = {"WifiConfig":{"4":"Retry"}}
```

Then enter credentials on the portal, device connects to the provided AP but WifiConfig is set to WIFI_RESTART (0)
```
00:00:58.847 WIF: Connecting to AP BNETWLAN as tasmota_2AD017-4119 ...
00:01:06.487 WIF: SSId1 BNETWLAN: Connected - IP Address 192.168.168.52
wificonfig
00:01:08.889 CMD: wificonfig
00:01:08.893 RSL: RESULT = {"WifiConfig":{"0":"Restart"}}
```

After restart, same:
```
00:00:00.002 HDW: ESP8266EX
00:00:00.038 CFG: Loaded from flash at F7, Count 5
00:00:00.044 QPC: Count 1
00:00:00.051 Project tasmota Tasmota Version 9.4.0.4(tasmota)-2_7_4_9(2021-05-30T09:34:24)
00:00:00.541 WIF: Connecting to AP1 BNETWLAN in mode 11n as tasmota_2AD017-4119...
00:00:06.694 QPC: Reset
00:00:07.002 WIF: Connected
00:00:07.254 HTP: Web server active on tasmota_2AD017-4119 with IP address 192.168.168.52
15:39:46.283 RSL: INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"9.4.0.4(tasmota)","FallbackTopic":"cmnd/DVES_2AD017_fb/","GroupTopic":"cmnd/tasmotas/"}}
15:39:46.288 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_2AD017-4119","IPAddress":"192.168.168.52"}}
15:39:46.299 RSL: INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
15:39:46.306 RSL: RESULT = {"POWER":"OFF"}
15:39:46.309 RSL: POWER = OFF
wificonfig
15:39:48.706 CMD: wificonfig
15:39:48.710 RSL: RESULT = {"WifiConfig":{"0":"Restart"}}
```

### EXPECTED BEHAVIOUR

WifiConfig should stick on the value hardcoded in the binary (WIFI_RETRY by default or any other value defined by user_config_override.h)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

mlavelle42, 2021-05-30:
This happens to me too.

barbudor, 2021-05-30:
I tested with older versions:
9.3.1.1 => 4 (Retry)
9.3.1.2 => 2 (WifiManager)
9.3.1.3 => 0 (Restart)

barbudor, 2021-05-30:
Adding some logs
Here : https://github.com/arendst/Tasmota/blob/322c056758007c40f13ea6e5e7804807b04e9887/tasmota/xdrv_01_webserver.ino#L1801
`TasmotaGlobal.wifi_state_flag` is 0
`Settings.sta_config` is 4
So `Web.old_wificonfig` get 0

And there:
https://github.com/arendst/Tasmota/blob/322c056758007c40f13ea6e5e7804807b04e9887/tasmota/xdrv_01_webserver.ino#L3298
`Settings.sta_config` is restored to 0

At 1st glance, saving `Settings.sta_config` instead of `TasmotaGlobal.wifi_state_flag` seems to work ... if connection to AP works fine:
```
00:00:00.765 WIF: WifiManager active for 3 minutes
00:00:00.275 HTP: Web server active on tasmota_2AD017-4119 with IP address 192.168.4.1
00:00:30.171 WFM: save wificonfig 4 and set to 2 (settings=4)
00:00:30.174 WIF: Connecting to AP BNETWLAN as tasmota_2AD017-4119 ...
00:00:38.382 WIF: SSId1 BNETWLAN: Connected - IP Address 192.168.168.52
00:00:38.383 WFM: Restore wificonfig 4
00:00:43.360 APP: Restarting
```

However, if the credentials are not correct and wifi doesn't connect, Tasmota remains in the WifiManager and re-save value 2 (RESTART)
```
00:00:00.759 WIF: WifiManager active for 3 minutes
00:00:00.269 HTP: Web server active on tasmota_2AD017-4119 with IP address 192.168.4.1
00:00:42.593 WFM: save wificonfig 4 and set to 2 (settings=4)
00:00:42.596 WIF: Connecting to AP BNETWLAN as tasmota_2AD017-4119 ...
00:01:03.702 WIF: Connect failed with AP timeout
00:01:20.503 WFM: save wificonfig 2 and set to 2 (settings=2)
00:01:20.506 WIF: Connecting to AP BNETWLAN as tasmota_2AD017-4119 ...
00:01:25.813 WIF: SSId1 BNETWLAN: Connected - IP Address 192.168.168.52
00:01:25.815 WFM: Restore wificonfig 2
00:01:33.885 APP: Restarting
```

If we save WIFI_CONFIG_TOOL then it would always restore the value defined in `my_user_config.h`/`user_config_override.h`
In that case if a user goes back to WifiManager while WifiConifg has already been changed from the console, it would revert to the initial value.

No more idea right now

barbudor, 2021-05-30:
Proposed a PR!: https://github.com/arendst/Tasmota/pull/12242

ascillato2, 2021-06-01:
Thanks, Let me do some tests.

ascillato2, 2021-06-04:
Fix merged. Thanks for reporting and for fixing!!!

closed 2021-06-04 by ascillato2