arendst/Tasmota #12244: Incorrect Data after DeepSleepTime change

n0valis, 2021-05-31:
### PROBLEM DESCRIPTION
I change the DeepsleepTime of an 8266 D1 mini every night and every morning. I increase the value from 600s to 1800s during night and vice versa in the morning. Exactly at this time I have incorrect data being transmitted from the BME280 sensor.
This a Tasmota issue?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): generic
- [ ] Tasmota binary firmware version number used: latest _____
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: espeasy
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:00:02.059 CMD: Backlog Template; Module; GPIO 255
00:00:02.113 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:30:04.109 RSL: RESULT = {"Module":{"0":"Generic"}}
11:30:04.362 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"640":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
11:30:05.038 MQT: Attempting connection...
11:30:05.058 MQT: Connected
11:30:05.061 MQT

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**
[image]

sfromis, 2021-05-31:
Have the temperature sensor close the the ESP8266? The chip can get noticeably warmer if on more than very briefly.

n0valis, 2021-05-31:
It is near yes but all the other data is ok.

sfromis, 2021-05-31:
Of course, all the other data would not be affected if it is a case of the temperature sensor being warmed by a longer than usual on period. What I mean is that changing deep sleep timing might result in the current on-period getting longer. 

Anyway, Tasmota can only repeat the data it got from the sensor, without modification. Thus, there has to be a reason for the sensor being warmed enough to be reporting a higher value. And my experience with BME280 does not suggest it being liable to report spurious wrong data.

n0valis, 2021-05-31:
As I said:  It would makes sense if the TelePeriod time (which is actually the awake time) was noticeable longer. But as you can see, it isn't. Look at the chart, the yellow chart indicates the time the sensor is awake. It is not longer than usual. ALso, I do not only get incorrect data from the temperature sensor but also from the humidity and air pressure sensor:

[image]

sfromis, 2021-06-01:
I did not talk about teleperiod time, but if the awake time is longer due to you changing parameters. You did not show any indication of what your physical power-on time actually was.

Anyway, now that you add further info, this may make the hardware-related issue different, but still *Tasmota can only repeat the data it got from the sensor, without modification.* Whatever numbers are not created by Tasmota but by the sensor.

It is not impossible that some power-related issue may throw off BME280 from working properly.

n0valis, 2021-06-01:
I tried to say that the TelePeriod time is actually the awake time. It is shown in the legend as teleperiod, but it is the actual time the esp is alive between two sleep periods.
And yes the interesting question is, why does the hardware suddenly report false values exactly at the time the software changes its opertaing mode.

sfromis, 2021-06-01:
What I'm trying to get across is that while teleperiod is *normally* coupled to controlling on-time, yes, you can't automatically say that you changing deepsleep parameters could not possibly cause that *single* on-period to differ. In general, do not assume, verify.

Your screendump does not show what (how much) happened to pressure, but on a few occasions (with dubious power), I've seem BME280 produce values way off, not even within the plausible range. Nothing like the rather small spike in your temperature screendump, which is entirely consistent with what could happen with longer on-time and temperature sensor too close to the radiator.

stefanbode, 2021-06-17:
I agree that the sensor reading should be the same. Do you have the issue ALL the time when you change the deepsleeptime? For example, change it every hour and increase it by 30 seconds. 
I assume you use the RETAIN flag on your MQTT broker to store the new deepsleeptime when the device wakes up and connects.
I quite confident that this is not an issue by changing the deepsleeptime. Very unlikely. BEFORE the esp goes into deepsleep it reports the uptime. Maybe you can grab this too is a graph and post it here. A too short uptime could the reason. I have a similar setup with a light-distance sensor. No issues there, even if changing the deepsleep like you between night and day.

n0valis, 2021-07-03:
Yes all the time when I change the deepsleeptime. I use the sonoff adapter in iobroker as MQTT broker and I update the deepsleeptime value at a given time. I leave the rest to iobroker. The TelePeriod value you can see in the first graph is the uptime. I just got the naming wrong, and since I didn't know there was actually a datapoint for it I calculated it myself.

stefanbode, 2021-07-04:
Please take a Lok in the Doku. Just before deepsleep you get a Json including the uptime in seconds. Here we can see if the device after a change is doing something completely different. Let me know if you don't find it. It's the last mqqt message before the device turns off.

n0valis, 2021-07-05:
I don't know how to do that. Can you please explain. I can check the value UptimeSec in IoBroker. But I think it is not transmitted, it does not change. Maybe the uptime is too short, my calculations say it is around 12s.

n0valis, 2021-07-06:
> Do you have the issue ALL the time when you change the deepsleeptime? 

I have to correct myself here. I do not have the problem if I change it "manually". If I do it via console it doesn't affect the data.
The two times I get the  wrong values are when my day-night-shift script in iobroker is doing it. If the illumination sensor reports 0 Lux a javascript is called that sets the deepsleeptime to 1800 and if it reports >0 Lux it is set back to 600.

I can also trigger the change by changing a user datapoint in iobroker to a new value which then set deepsleeptime to this value.
This would also not affect in incorrect data from the sensors. Only if the script does it.
I don't know how the script could be the issue, but it seems it is.

closed 2021-07-23 by github-actions[bot]