arendst/Tasmota #12255: HD-38

rgerr, 2021-06-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I recently purchased a few HD-38 moist sensors. Or atleast, that's what is says on the board:

[image]

I have tried literally all options on my ESP8266 with Tasmota, I have used D6 and A0 as pin. No result whatsoever. Any ideas on this, or is it possible to add this sensor?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-06-01:
Wired correctly, you can select between digital and analog outputs. If you want digitial, you need to make sure that ADC on ESP8266 gets a voltage in the supported range, 0-1V, not more. Dev boards will normally have a voltage divider to accept up to 3.3V on A0. If correctly setup, and configured on the analog pin (gpio17), Tasmota will be able to show a number, and you have `AdcParam` to enable extra analog features.

rgerr, 2021-06-01:
Thanks for your reply. The A0 does give me a value (range). So that’s good. 

Next I would like this to be pushed to Domoticz, however, in the domoticz configuration on Tasmota, the range of the sensor is not being pushed to domoticz since there is no option for that. I tried all fields with a dummy idx, but it’s not coming through. 

Any thoughts on this?

sfromis, 2021-06-01:
Use rules. [Example](https://tasmota.github.io/docs/Rules/#use-of-variables-and-tele-in-domoticz).

rgerr, 2021-06-01:
> Use rules. [Example](https://tasmota.github.io/docs/Rules/#use-of-variables-and-tele-in-domoticz).

Not sure how to use this, via console? And do you have an example for A0: ADC Range to Domoticz IDX?

closed 2021-06-01 by ascillato2