arendst/Tasmota #12275: WHERE THE FUCK IS THE DOWNLOAD BUTTON FOR SONOFF 4CH??????????

portiktamas, 2021-06-03:
I ONLY SEE THE DOWNLOAD FOR THE FUCKIN SOURCE CODE WHICH I DON'T NEED!!!!!
WHY IS THIS SOOOOOOO FUCKING HARD TO PUT A BIIIIIGGGGG BUTTTONN ON THE FRONT OF THE PAGE AND CALL IT A DAY????? YOU ARE COMPLICATING EVERYBODY'S LIFE! I DON'T WANT TO MIGRATE FROM AN OLDER VERSION, JUST INSTALL THE FUCKING .BIN

closed 2021-06-03 by close-issue-app[bot]

Jason2866, 2021-06-06:
?
[image]

ascillato2, 2021-06-09:
@portiktamas

Hi, the button you are looking for is just a bit lower than the download source button and it is called RELEASE 9.4.0 and also you can reach the .bin files by clicking also on the DOWNLOADS button next to the release button. If you need help, you have the Tasmota discord chat to kindly ask for assistance.

Also, your post has been edited and moderated to comply with the Code of Conduct and so as to have a civilizated behaviour required by community projects.

------

User **portiktamas** has been reported to github.