arendst/Tasmota #12278: PWM with zero cross does not work properly after 8.4.0

Tinky1233, 2021-06-03:
### PROBLEM DESCRIPTION
PWM does not work with zero cross detection after version 8.4.0. The output works ok from about dimmer 5% to dimmer 80%, outside that range output behavior is erratic. Output cannot be turned OFF 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP12E and aliexpress ac dimmer module
- [] Tasmota binary firmware version number used: all versions above 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotizer1.2
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

GPIO4  PWM1
GPIO5  COUNTER1

Setoption99 1
Pwmfrequency 100
Ledtable 0

adjust slider below 5% or try to turn output off

module used:
https://a.aliexpress.com/_m0qzkH7

### EXPECTED BEHAVIOUR
Normal dimming

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
Dimming works normal on 8.4.0

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-06-03:
Can you try with changing from `DWAVEFORM_LOCKED_PWM` to `DWAVEFORM_LOCKED_PHASE`
this change can be done here https://github.com/arendst/Tasmota/blob/development/platformio.ini#L117-L119

Tinky1233, 2021-06-03:
Its now properly synced, but still cannot turn-off, and erratic at extremes 0-10%  90-100%

stefanbode, 2021-06-13:
I did the code for this device. Therefore I assume I have most knowledge. Can you confirm your  frequency is correct. I have a 9.2 running without problems. 

With high debug level we should be able to find the problem. Between 0% and 5% it will be off. Very difficult to do the timing right. Same between 95% and 100%. This should be full on without any pWM.

Tinky1233, 2021-07-13:
I assembled a test rig. Installed latest 9.5.0, i can confirm that is not working.

I tried various pin locations no effect.

When i configure ouput as relay, Triac module works fine (see video):
https://youtu.be/ofH9gg8Z4L0

When i configure PWM with zero cross, It's not working and it cannot turn off (see video):
-setoption 99 1
-pwmfrequency 100
-ledtable 0
https://youtu.be/3aUiHk8S_s4

closed 2021-08-12 by github-actions[bot]