arendst/Tasmota #12283: DDS238-2 Reactive Power wrong value

sebitnt, 2021-06-03:
### PROBLEM DESCRIPTION
Hello,
I am using 9.4.0(sensors) with an DDS238-2 attached via Modbus. Everything is working as expected except for the Reactive Power which is displayed wrong (much too high). It seems to me that it is an issue with an 16bit value being rolled over. Thank you for your help in advance!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP07
- [ ] Tasmota binary firmware version number used: 9.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Not necessary.

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
 None

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  Not necessary.

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Not necessary.

```

### TO REPRODUCE
See description.

### EXPECTED BEHAVIOUR
Calculate and show the correct value.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
Nothing

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2021-06-10:
Hi,
I wonder, that noone already replied.
For me, it looks like we have a negative value overflow here.
Due to some effect like rounding, the value is negative, but the variable only expects positive values.
Haven't looked into the code yet, but I suggest to set the value to zero, when it becomes negative.
A negative value doesn't make sense indeed.
Joerg

chaveiro, 2021-06-14:
Check if this is the reason : https://github.com/arendst/Tasmota/discussions/12272

sebitnt, 2021-06-17:
I have done a few tests. The problem is only present when the power flow is backwards through the meter. Connecting it the other way, tasmota is reporting correct values. This seems to me to be a overflow of a unsigned variable. Reactive power might be negative but tasmota does not expect this.

arendst, 2021-06-19:
Thx for the investigation.

Tasmota does negative reactive power as can be seen here:

[image]

The issue is with the DDS2382 driver (xnrg_09_dds2382.ino - line 70) which decodes reactive power like:
```
      Energy.reactive_power[0] = (float)((buffer[33] << 8) + buffer[34]);
```
You might want to try the below change dline:
```
      Energy.reactive_power[0] = (float)(int16_t)((buffer[33] << 8) + buffer[34]);
```

closed 2021-06-21 by ascillato2