arendst/Tasmota #12290: Ruleset doesn't stop on BREAK, if using one-shot detection

dragospascale, 2021-06-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff SV
- [ x] Tasmota binary firmware version number used: 9.4.0.4
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: tasmotizer.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:24:03.251 CMD: Backlog Template; Module; GPIO 255
09:24:03.303 MQT: stat/tasmota_859E55/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:24:03.552 MQT: stat/tasmota_859E55/RESULT = {"Module":{"3":"Sonoff SV"}}
09:24:03.805 MQT: stat/tasmota_859E55/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"640":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
09:26:15.468 MQT: stat/tasmota_859E55/RESULT = {"Rule1":{"State":"ON","Once":"ON","StopOnError":"OFF","Length":198,"Free":313,"Rules":"ON VEML7700#Illuminance >= 1500 DO Var1 1500 BREAK ON VEML7700#Illuminance >= 1000 DO Var1 1000 BREAK ON VEML7700#Illuminance >= 500 DO Var1 500 ENDON ON VEML7700#Illuminance < 500 DO Var1 100 ENDON"}}
09:26:15.679 MQT: stat/tasmota_859E55/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:26:15.929 MQT: stat/tasmota_859E55/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:27:14.022 MQT: stat/tasmota_859E55/STATUS = {"Status":{"Module":3,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_859E55","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:27:14.065 MQT: stat/tasmota_859E55/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:27:16","StartupUTC":"2021-06-04T07:59:58","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2021-06-04T08:29:16","SaveCount":20,"SaveAddress":"F8000"}}
09:27:14.115 MQT: stat/tasmota_859E55/STATUS2 = {"StatusFWR":{"Version":"9.4.0.4(tasmota)","BuildDateTime":"2021-06-04T07:27:26","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"359/699"}}
09:27:14.145 MQT: stat/tasmota_859E55/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
09:27:14.188 MQT: stat/tasmota_859E55/STATUS4 = {"StatusMEM":{"ProgramSize":618,"Free":384,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00005004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,71"}}
09:27:14.237 MQT: stat/tasmota_859E55/STATUS5 = {"StatusNET":{"Hostname":"tasmota_859E55-7765","IPAddress":"192.168.111.27","Gateway":"192.168.111.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.50.21","Mac":"EC:FA:BC:85:9E:55","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
09:27:14.273 MQT: stat/tasmota_859E55/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.111.24","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_859E55","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:27:14.307 MQT: stat/tasmota_859E55/STATUS7 = {"StatusTIM":{"UTC":"2021-06-04T08:27:14","Local":"2021-06-04T09:27:14","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:49","Sunset":"20:47"}}
09:27:14.339 MQT: stat/tasmota_859E55/STATUS10 = {"StatusSNS":{"Time":"2021-06-04T09:27:14","VEML7700":{"Illuminance":459,"WhiteContent":905}}}
09:27:14.361 MQT: stat/tasmota_859E55/STATUS11 = {"StatusSTS":{"Time":"2021-06-04T09:27:14","Uptime":"0T00:27:16","UptimeSec":1636,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IT","BSSId":"58:8D:09:E3:15:01","Channel":2,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Rule 1 ON VEML7700#Illuminance >= 1500 DO Var1 1500 BREAK ON VEML7700#Illuminance >= 1000 DO Var1 1000 BREAK ON VEML7700#Illuminance >= 500 DO Var1 500 ENDON ON VEML7700#Illuminance < 500 DO Var1 100 ENDON
Rule1 5
Rule1 1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
When Illuminance is over 1500 I expect Var1 to be 1500 instead I get from rule to rule in sequence 1500 1000 and finaly 500. BREAK does not break as expected.
If I change to one-shot (Rule1 4) then it works as expected

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-06-04:
I see no error there. What happens sounds very logical assuming that one-shot acts for each trigger individually, meaning that when the first trigger is dormant with one-shot after firing one time, the next trigger will be allowed to act on next iteration of the main loop (about 1 Hz rate), and so on.

Rules and triggers are independent and event-driven, in no way similar to having if statements in a programming language.

dragospascale, 2021-06-04:
According to documentation:
Using BREAK to simulate IF..ELSEIF..ELSE..ENDIF~
BREAK is an alternative to ENDON. BREAK will stop the execution for the triggers that follow. If a trigger that ends with BREAK fires, then the following triggers of that rule will not be executed. This allows to simulate IF..ELSEIF..ELSE..ENDIF

Example:

IF temp > 85 then
  VAR1 more85
ELSEIF temp > 83 then
  VAR1 more83
ELSEIF temp > 81 then
  VAR1 more81
ELSEIF temp = 81 then
  VAR1 equal81
ELSE
  VAR1 less81
ENDIF
With the actual rules, if we use a set like the following:

Rule1
  ON event#temp>85 DO VAR1 more85 ENDON
  ON event#temp>83 DO VAR1 more83 ENDON
  ON event#temp>81 DO VAR1 more81 ENDON
  ON event#temp=81 DO VAR1 equal81 ENDON
  ON event#temp<81 DO VAR1 less81 ENDON
This is the output in the console:

CMD: rule
MQT: stat/living/RESULT = {"Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":322,"Rules":"ON event#temp>85 do VAR1 more85 ENDON ON event#temp>83 do VAR1 more83 ENDON on event#temp>81 do VAR1 more81 ENDON on event#temp=81 do VAR1 equal81 ENDON on event#temp<81 DO VAR1 less81 ENDON"}
CMD: event temp=10
MQT: stat/living/RESULT = {"Event":"Done"}
RUL: EVENT#TEMP<81 performs "VAR1 less81"
MQT: stat/living/RESULT = {"Var1":"less81"}
CMD: event temp=100
MQT: stat/living/RESULT = {"Event":"Done"}
RUL: EVENT#TEMP>85 performs "VAR1 more85"
MQT: stat/living/RESULT = {"Var1":"more85"}
RUL: EVENT#TEMP>83 performs "VAR1 more83"
MQT: stat/living/RESULT = {"Var1":"more83"}
RUL: EVENT#TEMP>81 performs "VAR1 more81"
MQT: stat/living/RESULT = {"Var1":"more81"}
So, all the triggers where TEMP>100, are firing. With the BREAK statement the rule set can be changed to:

Rule
  on event#temp>85 do VAR1 more85 break
  on event#temp>83 do VAR1 more83 break
  ON event#temp>81 DO VAR1 more81 ENDON
  ON event#temp=81 DO VAR1 equal81 ENDON
  ON event#temp<81 DO VAR1 less81 ENDON
Which will result in the following output:

CMD: rule
RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":321,"Rules":"ON event#temp>85 do VAR1 more85 break ON event#temp>83 do VAR1 more83 break on event#temp>81 do VAR1 more81 ENDON on event#temp=81 do VAR1 equal81 ENDON on event#temp<81 DO VAR1 less81 ENDON"}
CMD: event temp=10
RSL: RESULT = {"Event":"Done"}
RUL: EVENT#TEMP<81 performs "VAR1 less81"
RSL: RESULT = {"Var1":"less81"}
CMD: event temp=100
RSL: RESULT = {"Event":"Done"}
RUL: EVENT#TEMP>85 performs "VAR1 more85"
RSL: RESULT = {"Var1":"more85"}
CMD: event temp=83
RSL: RESULT = {"Event":"Done"}
RUL: EVENT#TEMP>81 performs "VAR1 more81"
RSL: RESULT = {"Var1":"more81"}

sfromis, 2021-06-04:
The example in the documentation is significantly different, in that it is not using a sensor continuously updated, but instead one single value. Just not the same as what you're doing.

The main point is that your rules are run with multiple independent 1 Hz iterations of sensor readings, instead of it being the same value. When one-shot for the first trigger has been fulfilled, that trigger is basically out of the loop till no longer satisfied, opening up the next trigger when the next sensor reading arrives next second.

sfromis, 2021-06-04:
But if you actually did fill in the most interesting part of the template, it would be more clear :-)

dragospascale, 2021-06-04:
And now is clear ?

sfromis, 2021-06-04:
No, you you have still not documented your issue, and simply ignored my explanations.

dragospascale, 2021-06-05:
Ok . Thank you. I will use the example of thermostat because is very similar to my case.

closed 2021-06-05 by dragospascale