arendst/Tasmota #12293: rst looping when flashing ESP32_CAM with tasmota32_webcam.bin using ESP-Flasher-Windows-x64.exe

lecocqe, 2021-06-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
i'm flashing tasmota32_webcam and in the serial at the end of the flashing.
it reboot
and get this in loop

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:350180
ets Jun  8 2016 00:22:57

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32AI Thinker
- [x] Tasmota binary firmware version number used: 9.4.0 (stable) and 9.4.0.4 (dev)
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: **ESP-Flasher-Windows-x64.exe**
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- flashing the esp32 tasmota32_webcam with the tool ESP-Flasher-Windows-x64.exe

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
starting tasmota on device

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:350180
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:350180
ets Jun  8 2016 00:22:57

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2021-06-04:
[image]

https://tasmota.github.io/docs/ESP32/

lecocqe, 2021-06-04:
maybe, yes. after flashing for the 5th time. it worked.

closed 2021-06-04 by ascillato2