arendst/Tasmota #12294: ESP 32 webcam - MQTT socket error

Enjoy-Combi, 2021-06-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Regular socket error on client with disconnection & reconnection

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): AITHINKER ESP32 CAM
- [ ] Tasmota binary firmware version number used: 9.4.0.4(webcam)
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:37:56.229 MQT: stat/webcam_salon_2/RESULT = {"NAME":"AITHINKER CAM","GPIO":[4992,1,672,1,416,5088,1,1,1,6720,736,704,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,576,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":2}
16:37:56.530 MQT: stat/webcam_salon_2/RESULT = {"Module":{"0":"AITHINKER CAM"}}
16:37:56.826 MQT: stat/webcam_salon_2/RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"}}
16:37:56.837 MQT: stat/webcam_salon_2/RESULT = {"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"576":"LedLink_i"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:38:37.575 MQT: stat/webcam_salon_2/STATUS11 = {"StatusSTS":{"Time":"2021-06-04T16:38:37","Uptime":"1T19:50:17","UptimeSec":157817,"Heap":120,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":287,"MqttCount":23,"POWER":"OFF","Dimmer":30,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SFR_B7A8","BSSId":"E4:5D:51:69:B7:AE","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
16:13:02.255 MQT: tele/webcam_salon_2/SENSOR = {"Time":"2021-06-04T16:13:02","ESP32":{"Temperature":70.0},"TempUnit":"C"}
16:13:58.572 MQT: Attempting connection...
16:13:58.606 MQT: Connected
16:13:58.612 MQT: tele/webcam_salon_2/LWT = Online (retained)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Nothing special, it appears in a unpredictable way. Nohting special to do

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Do not have MQTT disconnection (broker on a raspberry pi, if mosquitto v1.6.x the devices is usable, if mosquitto v 2.x.x the devices is not usable as there is two must disconnections and connections)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-06-04:
You may have unstable wifi for the board.

Enjoy-Combi, 2021-06-04:
I don't think so. I did a test with two raspberry:
- One with where domoticz is running and where mosquitto is. I have issue only with the AIthinker I have (total of two). Please note that I have plenty of esp8266 devices without any issue.
- One where there is home assistant for testing, when I connect the Aithinker on this one I have no issue.

ascillato, 2021-06-04:
Please, provide the complete output of the console for the command `STATUS 0`. Thanks.

Also check that you don't have several devices with the same MQTT CLIENT name.

Jason2866, 2021-06-05:
Can not reproduce with my AI-Tinker CAM
```
00:00:00.001 HDW: ESP32-D0WDQ6
00:00:00.047 UFS: FlashFS mounted with 52 kB free
00:00:00.114 CFG: Loaded from File, Count 0
00:00:00.206 CFG: Use defaults
00:00:00.572 Project sonoff Tasmota Version 9.4.0.4(webcam)-1_0_6(2021-06-05T15:56:06)
00:00:00.738 CMD: 
00:00:04.661 WIF: Connecting to AP1 Jason_Home_WLAN Channel 4 BSSId 88:C3:97:B1:1D:56 in mode 11n as sonoff-952728-1832...
00:00:12.515 WIF: Connected
16:01:12.244 HTP: Web server active on sonoff-952728-1832 with IP address 192.168.2.174
16:01:13.778 MQT: Attempting connection...
16:01:13.808 MQT: Connected
16:01:13.816 MQT: tele/sonoff-952728/LWT = Online (retained)
16:01:13.830 MQT: cmnd/sonoff-952728/POWER = 
16:01:13.843 MQT: tele/sonoff-952728/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"9.4.0.4(webcam)","FallbackTopic":"cmnd/sonoff-952728_fb/","GroupTopic":"cmnd/sonoffs/"}}
16:01:13.853 MQT: tele/sonoff-952728/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sonoff-952728-1832","IPAddress":"192.168.2.174"}}
16:01:13.868 MQT: tele/sonoff-952728/INFO3 = {"Info3":{"RestartReason":"Software reset digital core"}}
16:01:18.553 MQT: tele/sonoff-952728/STATE = {"Time":"2021-06-05T16:01:18","Uptime":"0T00:00:18","UptimeSec":18,"Heap":198,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"88:C3:97:B1:1D:56","Channel":4,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:11"}}
16:01:18.580 MQT: tele/sonoff-952728/SENSOR = {"Time":"2021-06-05T16:01:18","ESP32":{"Temperature":52.2},"TempUnit":"C"}
```

Jason2866, 2021-06-05:
I do not think it is (WebCam)ESP32 related, have you tried the ESP8266 to connect to the mosquitto v.2.0 ? Is there a specific reason for using mosquitto v2.0? Tasmota does not use features of the v2.0 and will not do in the future.

sfromis, 2021-06-05:
By now, the main reason for using Mosquitto v2.0 is probably that it is the current version, generally flagged as "latest". Extra work to switch to an older one. For now, I've postponed upgrading my old setup running very smoothly.

Enjoy-Combi, 2021-06-05:
I was using mosquitto v2 simply because it the stable verison on raspberry OS right now. When having this I enabled listening on port 1883 & anoymous connection.
the modules (two at the total) were operating good for sometime and start continiously disconnecting and reconnecting (MQTT, not wifi).
I downgraded to mosquitto v1 and the modules are usable but with few mqtt disconnections & reconnections. I do not have this issue at with sononf devices, esp-01 devices I have (and I have roughly 15 devices like this).

Here is the log with more data when the issue appears:
Jun  5 12:50:06 webcam-salon-2 ESP-WIF: Checking connection...
Jun  5 12:50:11 webcam-salon-2 ESP-DOM: idx 9, nvalue 0
Jun  5 12:50:11 webcam-salon-2 ESP-DOM: idx 32, nvalue 0
Jun  5 12:50:21 webcam-salon-2 ESP-DOM: idx 8, nvalue 0
Jun  5 12:50:26 webcam-salon-2 ESP-WIF: Checking connection...
Jun  5 12:50:31 webcam-salon-2 ESP-DOM: idx 7, nvalue 0
Jun  5 12:50:31 webcam-salon-2 ESP-DOM: idx 105, nvalue 0
Jun  5 12:50:31 webcam-salon-2 ESP-DOM: idx 106, nvalue 0
Jun  5 12:50:32 webcam-salon-2 ESP-DOM: idx 107, nvalue 0
Jun  5 12:50:33 webcam-salon-2 ESP-DOM: idx 24, nvalue 0
Jun  5 12:50:33 webcam-salon-2 ESP-DOM: idx 26, nvalue 0
Jun  5 12:50:33 webcam-salon-2 ESP-DOM: idx 25, nvalue 0
Jun  5 12:50:33 webcam-salon-2 ESP-DOM: idx 23, nvalue 0
Jun  5 12:50:46 webcam-salon-2 ESP-WIF: Checking connection...
Jun  5 12:51:01 webcam-salon-2 ESP-DOM: idx 7, nvalue 0
Jun  5 12:51:01 webcam-salon-2 ESP-DOM: idx 105, nvalue 0
Jun  5 12:51:02 webcam-salon-2 ESP-DOM: idx 106, nvalue 0
Jun  5 12:51:02 webcam-salon-2 ESP-DOM: idx 107, nvalue 0
Jun  5 12:51:06 webcam-salon-2 ESP-WIF: Checking connection...
Jun  5 12:51:11 webcam-salon-2 ESP-HTP: Main Menu
Jun  5 12:51:26 webcam-salon-2 ESP-CAM: Handle camserver
Jun  5 12:51:26 webcam-salon-2 ESP-CAM: Create client
Jun  5 12:51:26 webcam-salon-2 ESP-CAM: Start stream
Jun  5 12:51:26 webcam-salon-2 ESP-DOM: idx 11, nvalue 0
Jun  5 12:51:27 webcam-salon-2 ESP-HTP: Consoles
Jun  5 12:51:27 webcam-salon-2 ESP-CAM: Handle camserver
Jun  5 12:51:27 webcam-salon-2 ESP-CAM: Create client
Jun  5 12:51:27 webcam-salon-2 ESP-CAM: Client fail
Jun  5 12:51:27 webcam-salon-2 ESP-CAM: Stream exit
Jun  5 12:51:27 webcam-salon-2 ESP-CAM: Handle camserver
Jun  5 12:51:27 webcam-salon-2 ESP-CAM: Create client
Jun  5 12:51:27 webcam-salon-2 ESP-CAM: Start stream
Jun  5 12:51:27 webcam-salon-2 ESP-MQT: Attempting connection...
Jun  5 12:51:27 webcam-salon-2 ESP-MQT: Connected
Jun  5 12:51:27 webcam-salon-2 ESP-MQT: tele/webcam_salon_2/LWT = Online (retained)
Jun  5 12:51:27 webcam-salon-2 ESP-MQT: cmnd/webcam_salon_2/POWER = 
Jun  5 12:51:27 webcam-salon-2 ESP-MQT: Subscribe to cmnd/webcam_salon_2/#
Jun  5 12:51:27 webcam-salon-2 ESP-MQT: Subscribe to cmnd/tasmotas/#
Jun  5 12:51:27 webcam-salon-2 ESP-MQT: Subscribe to cmnd/DVES_EAA038_fb/#
Jun  5 12:51:27 webcam-salon-2 ESP-MQT: Subscribe to domoticz/out/#
Jun  5 12:51:27 webcam-salon-2 ESP-MQT: Unsubscribe from homeassistant/status
Jun  5 12:51:31 webcam-salon-2 ESP-HTP: Console
Jun  5 12:51:31 webcam-salon-2 ESP-DOM: idx 8, nvalue 0
Jun  5 12:51:32 webcam-salon-2 ESP-DOM: idx 7, nvalue 0
Jun  5 12:51:32 webcam-salon-2 ESP-DOM: idx 105, nvalue 0
Jun  5 12:51:32 webcam-salon-2 ESP-DOM: idx 106, nvalue 0
Jun  5 12:51:32 webcam-salon-2 ESP-DOM: idx 107, nvalue 0
Jun  5 12:51:32 webcam-salon-2 ESP-DOM: idx 9, nvalue 0
Jun  5 12:51:32 webcam-salon-2 ESP-DOM: idx 32, nvalue 0
Jun  5 12:51:33 webcam-salon-2 ESP-DOM: idx 24, nvalue 0
Jun  5 12:51:33 webcam-salon-2 ESP-DOM: idx 26, nvalue 0
Jun  5 12:51:33 webcam-salon-2 ESP-DOM: idx 25, nvalue 0
Jun  5 12:51:33 webcam-salon-2 ESP-DOM: idx 23, nvalue 0
Jun  5 12:51:45 webcam-salon-2 ESP-WIF: Checking connection...
Jun  5 12:51:47 webcam-salon-2 ESP-HTP: Consoles
Jun  5 12:51:47 webcam-salon-2 ESP-DOM: idx 128, nvalue 44
Jun  5 12:51:48 webcam-salon-2 ESP-HTP: Main Menu
Jun  5 12:51:51 webcam-salon-2 ESP-CAM: Handle camserver
Jun  5 12:51:51 webcam-salon-2 ESP-CAM: Create client
Jun  5 12:51:51 webcam-salon-2 ESP-CAM: Start stream
Jun  5 12:51:52 webcam-salon-2 ESP-CAM: Handle camserver
Jun  5 12:51:52 webcam-salon-2 ESP-CAM: Create client
Jun  5 12:51:52 webcam-salon-2 ESP-CAM: Start stream
Jun  5 12:51:55 webcam-salon-2 ESP-DOM: idx 51, nvalue 0
Jun  5 12:52:01 webcam-salon-2 ESP-DOM: idx 7, nvalue 0
Jun  5 12:52:01 webcam-salon-2 ESP-DOM: idx 105, nvalue 0
Jun  5 12:52:01 webcam-salon-2 ESP-DOM: idx 106, nvalue 0
Jun  5 12:52:01 webcam-salon-2 ESP-DOM: idx 107, nvalue 0
Jun  5 12:52:08 webcam-salon-2 ESP-WIF: Checking connection...

Apparently, the disconnection follow a client fail.

Enjoy-Combi, 2021-06-05:
Status 0 gives:

21:33:59.342 CMD: STATUS 0
21:33:59.350 MQT: stat/webcam_salon_2/STATUS = {"Status":{"Module":0,"DeviceName":"Webcam salon 2","FriendlyName":["Tasmota"],"Topic":"webcam_salon_2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:33:59.383 MQT: stat/webcam_salon_2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"3T00:45:39","StartupUTC":"2021-06-02T19:48:20","Sleep":50,"CfgHolder":4617,"BootCount":94,"BCResetTime":"2021-05-20T14:08:41","SaveCount":229}}
21:33:59.420 MQT: stat/webcam_salon_2/STATUS2 = {"StatusFWR":{"Version":"9.4.0.4(webcam)","BuildDateTime":"2021-05-19T15:28:36","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6","CR":"390/699"}}
21:33:59.439 MQT: stat/webcam_salon_2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"serveur","LogPort":514,"SSId":["SFR_B7A8",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
21:33:59.468 MQT: stat/webcam_salon_2/STATUS4 = {"StatusMEM":{"ProgramSize":1357,"Free":1856,"Heap":122,"PsrMax":4095,"PsrFree":3345,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC78F","00140001","000000CF","010013C0","40000981","80004000","00200000","00000024"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,50,81","Sensors":"1,2,3,5,6,87"}}
21:33:59.501 MQT: stat/webcam_salon_2/STATUS5 = {"StatusNET":{"Hostname":"webcam-salon-2","IPAddress":"192.168.1.22","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"9C:9C:1F:EA:A0:38","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
21:33:59.525 MQT: stat/webcam_salon_2/STATUS6 = {"StatusMQT":{"MqttHost":"serveur","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EAA038","MqttUser":"DVES_USER","MqttCount":33,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:33:59.547 MQT: stat/webcam_salon_2/STATUS7 = {"StatusTIM":{"UTC":"2021-06-05T20:33:59","Local":"2021-06-05T21:33:59","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:48","Sunset":"20:48"}}
21:33:59.568 MQT: stat/webcam_salon_2/STATUS10 = {"StatusSNS":{"Time":"2021-06-05T21:33:59","ESP32":{"Temperature":66.7},"TempUnit":"C"}}
21:33:59.582 MQT: stat/webcam_salon_2/STATUS11 = {"StatusSTS":{"Time":"2021-06-05T21:33:59","Uptime":"3T00:45:39","UptimeSec":261939,"Heap":118,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":170,"MqttCount":33,"POWER":"OFF","Dimmer":42,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SFR_B7A8","BSSId":"E4:5D:51:69:B7:AE","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:06"}}}

Enjoy-Combi, 2021-06-05:
Regarding the topic name, there is only this device having webcam_salon_2

ascillato, 2021-06-05:
Sorry, it is known that are some issues with mosquitto V2. If you can use previous version of mosquitto you won't have this issue. 
If not, please do a new clean install of mosquitto. If the issue continues, I don't have the exact issue number, but there is an issue in home assistant where it is explained how to reconfigure mosquitto to not have this issue.

Jason2866, 2021-06-06:
Using mosquitto v1.6 with Raspberry. Rock solid with every device. I do NOT use Home Assistant.
It is not a Tasmota issue.

Enjoy-Combi, 2021-06-06:
> Using mosquitto v1.6 with Raspberry. Rock solid with every device. I do NOT use Home Assistant.
> It is not a Tasmota issue.

I use it with domoticz. I just do some test with home assistant.
FYI, if I use mosquitti broker with home assistant (on a separate raspberry) no issue.
I have the issue when using it with the raspberry which hosts domoticz (even with the raspberry shutted down where home assistant is)

Jason2866, 2021-06-06:
You just wrote yourself you have a setup issue when using Domoticz / mosquitto together.
Tasmota github is not the place to solve these issues.

Enjoy-Combi, 2021-06-06:
I understand, but how to be sure this is not an issue with the tasmota firmware ?

Jason2866, 2021-06-06:
Tasmota is working in many many mqtt installations without issues. We have no open issue regarding mqtt problems. Since we do not have issues we cant isolate where your issues do came from. You are the only one who can do.

Enjoy-Combi, 2021-06-06:
OK, then I can close this one. When finding the issue I will update it for information.
Thanks for all your comments

closed 2021-06-06 by Enjoy-Combi