arendst/Tasmota #12304: Not booting without internet access

chris7topher, 2021-06-06:
### PROBLEM DESCRIPTION
I flashed Sonoff Dual R3 modules with the current Tasmota ESP32 release. When I try to boot the devices without internet access, the web frontend hangs and MQTT requests are received only very delayed.
When booting the devices with internet access everything works fine. Even if the internet access is removed after booting, the modules continue to work without any problems. To me it looks like a web request is blocking other activities at startup.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R3
- [x] Tasmota binary firmware version number used: 9.4.0 and 9.4.0.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP-Flasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:30:15.204 MQT: stat/Sonoff-DualR3-Rollo-06/RESULT = {"NAME":"Sonoff Dual R3","GPIO":[32,0,0,0,0,0,3232,3200,0,576,225,0,0,0,0,0,0,0,0,0,0,1,1,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
00:30:15.426 MQT: stat/Sonoff-DualR3-Rollo-06/RESULT = {"Module":{"0":"Sonoff Dual R3"}}
00:30:15.683 MQT: stat/Sonoff-DualR3-Rollo-06/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"3232":"Serial Rx"},"GPIO10":{"3200":"Serial Tx"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"224":"Relay1"}}
00:30:15.701 MQT: stat/Sonoff-DualR3-Rollo-06/RESULT = {"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:30:58.777 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"Sonoff-DualR3-Rollo-06","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:30:58.793 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"Rollos","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:29:09","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2021-04-11T13:12:05","SaveCount":147}}
00:30:58.806 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021-04-23T10:10:10","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3","CR":"481/699"}}
00:30:58.817 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":2,"SysLog":4,"LogHost":"192.168.178.100","LogPort":514,"SSId":["LollipopRep",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A0A000000000000","400000C0","00006000","00000002"]}}
00:30:58.836 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS4 = {"StatusMEM":{"ProgramSize":1472,"Free":1984,"Heap":199,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","00084052","20200000","0000002D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,87"}}
00:30:58.846 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-DualR3-Rollo-06","IPAddress":"192.168.178.121","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"3C:61:05:4B:36:58","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:30:58.856 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.99","MqttPort":1883,"MqttClientMask":"Sonoff-DualR3-Rollo-06","MqttClient":"Sonoff-DualR3-Rollo-06","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:30:58.869 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:30:58","Local":"1970-01-01T00:30:58","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:30:58.886 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:30:58","Switch1":"ON","Switch2":"ON","ESP32":{"Temperature":54.4},"Shutter1":{"Position":23,"Direction":0,"Target":23},"TempUnit":"C"}}
00:30:58.898 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:30:58","Uptime":"0T00:29:09","UptimeSec":1749,"Heap":198,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"LollipopRep","BSSId":"DC:15:C8:7B:4B:0C","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:14"}}}
00:30:58.909 MQT: stat/Sonoff-DualR3-Rollo-06/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log (From Syslog, since weblog is not working because webserver hangs):
```lua
  Console output here:
Jun  6 13:18:16 Sonoff-DualR3-Rollo-06 ESP-MQT: tele/Sonoff-DualR3-Rollo-06/STATE = {"Time":"1970-01-01T00:01:56","Uptime":"0T00:00:20","UptimeSec":20,"Heap":209,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":9470,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"LollipopRep","BSSId":"DC:15:C8:7B:4B:0C","Channel":6,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:12"}}
Jun  6 13:18:16 Sonoff-DualR3-Rollo-06 ESP-MQT: tele/Sonoff-DualR3-Rollo-06/SENSOR = {"Time":"1970-01-01T00:01:56","Switch1":"ON","Switch2":"ON","ESP32":{"Temperature":53.9},"Shutter1":{"Position":23,"Direction":0,"Target":23},"TempUnit":"C"}
Jun  6 13:19:41 Sonoff-DualR3-Rollo-06 ESP-HTP: Consoles

```

### TO REPRODUCE
Boot tasmota without internet access

### EXPECTED BEHAVIOUR
Boot without internet access should work without problems

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-06-06:
Tasmota needs access to a NTP server, without it will lag. If you provide a NTP server in your LAN,
Tasmota needs NO access to the Internet.

chris7topher, 2021-06-06:
The local NTP server has solved the problem! Thanks for the info!
Wouldn't it be better to just continue working without NTP after a timeout? For many use cases it does not necessarily need an NTP server?

Jason2866, 2021-06-06:
Tasmota needs correct time. Without most functions do not work. So your suggestion is no alternative.

closed 2021-06-06 by chris7topher