arendst/Tasmota #12313: dht22 reporting via SI7021 not AM2301 on D4

Welsh-1, 2021-06-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____WEMOS_D1_MINI_PRO
- [ ] Tasmota binary firmware version number used: _____9.4
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____TASMOTIZER V1.2
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:41:57.904 CMD: Backlog Template; Module; GPIO 255
09:41:57.973 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:41:58.181 RSL: RESULT = {"Module":{"18":"Generic"}}
09:41:58.432 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"1248":"SI7021"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"224":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:43:35.496 CMD: Status 0
09:43:35.502 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E99BA5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:43:35.508 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:11:11","StartupUTC":"2021-06-07T08:32:24","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2021-06-07T08:38:43","SaveCount":38,"SaveAddress":"F6000"}}
09:43:35.512 RSL: STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021-04-23T10:07:22","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"330/699"}}
09:43:35.517 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Virus_4_Dad",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008001","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
09:43:35.525 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16301C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
09:43:35.530 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_E99BA5-7077","IPAddress":"10.0.0.44","Gateway":"10.0.0.254","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.254","Mac":"3C:61:05:E9:9B:A5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:43:35.536 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-06-07T08:43:35","Local":"2021-06-07T09:43:35","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:47","Sunset":"20:50"}}
09:43:35.542 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-06-07T09:43:35","SI7021":{"Temperature":25.7,"Humidity":53.4,"DewPoint":15.5},"TempUnit":"C"}}
09:43:35.547 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-06-07T09:43:35","Uptime":"0T00:11:11","UptimeSec":671,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Virus_4_Dad","BSSId":"FE:EC:DA:A1:90:03","Channel":11,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
plug DHT22 into D4 and Set to AM2301 - Null reported
plug DHT22 into D4 and Set to SI7201 - Data Reported (SI7021 Temperature | 25.8 °C -- | -- 53.3 % 15.6 °C)

OFF
--

</div>
Toggle
--

<div id="but3d" style="padding: 5px; font-size: 16px; color: rgb(234, 234, 234); font-family: verdana, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(37, 37, 37); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; display: block;"></div><p style="margin: 0.5em 0px; color: rgb(234, 234, 234); font-family: verdana, sans-serif; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(37, 37, 37); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"></p><form id="but3" action="http://10.0.0.44/cn" method="get" style="color: rgb(234, 234, 234); font-family: verdana, sans-serif; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(37, 37, 37); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; display: block;"><button name="" style="border: 0px; border-radius: 0.3rem; background: rgb(31, 163, 236); color: rgb(250, 255, 255); line-height: 2.4rem; font-size: 1.2rem; width: 340px; transition-duration: 0.4s; cursor: pointer;">Configuration</button></form><p style="margin: 0.5em 0px; color: rgb(234, 234, 234); font-family: verdana, sans-serif; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(37, 37, 37); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"></p><p style="margin: 0.5em 0px; color: rgb(234, 234, 234); font-family: verdana, sans-serif; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(37, 37, 37); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"></p><form id="but4" action="http://10.0.0.44/in" method="get" style="color: rgb(234, 234, 234); font-family: verdana, sans-serif; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(37, 37, 37); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; display: block;"><button name="" style="border: 0px; border-radius: 0.3rem; background: rgb(31, 163, 236); color: rgb(250, 255, 255); line-heig [... cut ...]  initial; text-align: right;"><hr><a href="https://bit.ly/tasmota" target="_blank" style="color: rgb(170, 170, 170); text-decoration: none;">Tasmota 9.4.0 by Theo Arends</a></div>

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-06-07:
What do you mean by "not on D0" in the updated title? Assuming that you mean GPIO16 (D0), instead of AM2301 vs SI7021, just do not use that pin. It does not work for another one-wire sensor DS18B20, as it is notably different from the regular GPIO pins.

Welsh-1, 2021-06-07:
I read that DHT22 should now be connected to GPIO16 (D0). Was just trying to find a link to the recommendation. If the recommendation is wrong then there is potentially no problem.

sfromis, 2021-06-07:
Searching Tasmota docs, I did not find any such recommendation or even suggestion. To me, GPIO16 is an encumbered pin, while not good for general usage, still workable for some things (especially simpler) after testing. Not using it till I have to.

ascillato2, 2021-06-07:
GPIO16 restrictions are Hardware restrictions. Those can not be changed by software. Sorry.

closed 2021-06-07 by ascillato2