arendst/Tasmota #12321: "TCPBaudRate" & "TCPStart" commands are unknown?

adharyanto, 2021-06-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I tried follow this link "https://tasmota.github.io/docs/Serial-to-TCP-Bridge/", but why the "TCPBaudRate" & "TCPStart" commands are unknown?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2021-06-08:
First, you did not fill out the issue template, as requested,
second, you seem to be unable to read the first sentence of the documentation you are referring to:
`This is available in the zbbridge or...`

sfromis, 2021-06-08:
Many commands needs two thing to be recognized. First, the feature needs to be included in the specific binary you use (self-compiled or precompiled), lots of features are not generally available. Second, the configuration (like gpio pins) needs to to be correct for the driver to become active.

closed 2021-06-09 by ascillato2