arendst/Tasmota #12323: Pulsetime counter reset after tele period

chriskmn, 2021-06-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The pulsetime counter gets reset on tele report if pulsetime is larger than tele period.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff s20
- [x ] Tasmota binary firmware version number used: 9.4.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: OTA
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

22:26:58.543 CMD: Backlog Template; Module; GPIO 255
22:26:58.574 MQT: stat/sonoff-3/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
22:26:58.791 MQT: stat/sonoff-3/RESULT = {"Module":{"8":"Sonoff S2X"}}
22:26:59.045 MQT: stat/sonoff-3/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

22:28:52.341 CMD: Status 0
22:28:52.349 MQT: stat/sonoff-3/STATUS = {"Status":{"Module":8,"DeviceName":"Sonoff-3","FriendlyName":["Bogenlampe"],"Topic":"sonoff-3","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:28:52.380 MQT: stat/sonoff-3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"3T07:30:52","StartupUTC":"2021-06-05T12:58:00","Sleep":250,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2021-03-29T19:42:42","SaveCount":21,"SaveAddress":"FB000"}}
22:28:52.413 MQT: stat/sonoff-3/STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021-04-23T10:07:22","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"422/699"}}
22:28:52.433 MQT: stat/sonoff-3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["malibu","malibu"],"TelePeriod":600,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","000000C0","00006000","00000000"]}}
22:28:52.464 MQT: stat/sonoff-3/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
22:28:52.497 MQT: stat/sonoff-3/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3-3090","IPAddress":"192.168.0.17","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:9F:4C:12","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:28:52.519 MQT: stat/sonoff-3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.180","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F4C12","MqttUser":"mosquitto","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:28:52.544 MQT: stat/sonoff-3/STATUS7 = {"StatusTIM":{"UTC":"2021-06-08T20:28:52","Local":"2021-06-08T22:28:52","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"05:47","Sunset":"21:50"}}
22:28:52.562 MQT: stat/sonoff-3/STATUS10 = {"StatusSNS":{"Time":"2021-06-08T22:28:52"}}
22:28:52.571 MQT: stat/sonoff-3/STATUS11 = {"StatusSTS":{"Time":"2021-06-08T22:28:52","Uptime":"3T07:30:52","UptimeSec":286252,"Heap":24,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":3,"MqttCount":3,"POWER":"ON","Wifi":{"AP":1,"SSId":"malibu","BSSId":"44:4E:6D:4C:E4:37","Channel":11,"RSSI":88,"Signal":-56,"LinkCount":2,"Downtime":"0T00:01:24"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

22:31:23.874 MQT: stat/sonoff-3/POWER = ON
22:31:28.355 CMD: Pulsetime 
22:31:28.358 SRC: WebConsole from 192.168.0.110
22:31:28.362 CMD: Group 0, Index 1, Command "PULSETIME", Data ""
22:31:28.369 MQT: stat/sonoff-3/RESULT = {"PulseTime1":{"Set":300,"Remaining":295},"PulseTime2":{"Set":0,"Remaining":0},"PulseTime3":{"Set":0,"Remaining":0},"PulseTime4":{"Set":0,"Remaining":0},"PulseTime5":{"Set":0,"Remaining":0},"PulseTime6":{"Set":0,"Remaining":0},"PulseTime7":{"Set":0,"Remaining":0},"PulseTime8":{"Set":0,"Remaining":0}}
22:31:36.107 WIF: Checking connection...
22:31:46.114 CMD: Pulsetime 
22:31:46.117 SRC: WebConsole from 192.168.0.110
22:31:46.121 CMD: Group 0, Index 1, Command "PULSETIME", Data ""
22:31:46.127 MQT: stat/sonoff-3/RESULT = {"PulseTime1":{"Set":300,"Remaining":277},"PulseTime2":{"Set":0,"Remaining":0},"PulseTime3":{"Set":0,"Remaining":0},"PulseTime4":{"Set":0,"Remaining":0},"PulseTime5":{"Set":0,"Remaining":0},"PulseTime6":{"Set":0,"Remaining":0},"PulseTime7":{"Set":0,"Remaining":0},"PulseTime8":{"Set":0,"Remaining":0}}
22:31:56.141 WIF: Checking connection...
22:32:16.148 WIF: Checking connection...
22:32:36.144 WIF: Checking connection...
22:32:51.913 MQT: tele/sonoff-3/STATE = {"Time":"2021-06-08T22:32:51","Uptime":"3T07:34:51","UptimeSec":286491,"Heap":26,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":4,"MqttCount":3,"POWER":"ON","Wifi":{"AP":1,"SSId":"malibu","BSSId":"44:4E:6D:4C:E4:37","Channel":11,"RSSI":98,"Signal":-51,"LinkCount":2,"Downtime":"0T00:01:24"}}
22:32:52.161 SRC: MQTT
22:32:52.164 CMD: Group 0, Index 1, Command "POWER", Data "ON"
22:32:52.170 MQT: stat/sonoff-3/RESULT = {"POWER":"ON"}
22:32:52.174 MQT: stat/sonoff-3/POWER = ON
22:32:57.419 CMD: Pulsetime 
22:32:57.422 SRC: WebConsole from 192.168.0.110
22:32:57.425 CMD: Group 0, Index 1, Command "PULSETIME", Data ""
22:32:57.430 MQT: stat/sonoff-3/RESULT = {"PulseTime1":{"Set":300,"Remaining":294},"PulseTime2":{"Set":0,"Remaining":0},"PulseTime3":{"Set":0,"Remaining":0},"PulseTime4":{"Set":0,"Remaining":0},"PulseTime5":{"Set":0,"Remaining":0},"PulseTime6":{"Set":0,"Remaining":0},"PulseTime7":{"Set":0,"Remaining":0},"PulseTime8":{"Set":0,"Remaining":0}}
22:32:57.914 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set pulsetime higher than tele period and power on

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device should switch off after pulsetime

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-06-08:
What your output shows is not PulseTime being reset in relation to TelePeriod, but the expected behavior of a new Power On command restarting PulseTime. As designed.

ascillato2, 2021-06-09:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-06-09 by ascillato2

chriskmn, 2021-06-09:
Thank you. I did not realise that. It seems to be sent out by a homebridge plugin. Opened an issue there.