arendst/Tasmota #12331: Sonoff B1 7-bit max white values for RGBWWTable

jan-gerard, 2021-06-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff B1
- [x] Tasmota binary firmware version number used: 9.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA (and stared with Tasmotizer)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
00:16:40.520 MQT: stat/tasmota_483F5C/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:16:40.730 MQT: stat/tasmota_483F5C/RESULT = {"Module":{"26":"Sonoff B1"}}
00:16:40.937 MQT: stat/tasmota_483F5C/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"3008":"MY92x1 DI"},"GPIO13":{"0":"None"},"GPIO14":{"3040":"MY92x1 DCKI"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
```
- [x] Provide the output of this command: `Status 0`:
```lua
00:17:32.234 MQT: stat/tasmota_483F5C/STATUS = {"Status":{"Module":26,"DeviceName":"SonoffB1a","FriendlyName":["SonoffB1a"],"Topic":"tasmota_483F5C","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:17:32.251 MQT: stat/tasmota_483F5C/STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021-04-23T10:07:22","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"395/699"}}
[0,0,0,0,70],"Scheme":0,"Fade":"ON","Speed":4,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Hector","BSSId":"3C:A6:2F:45:BA:3E","Channel":1,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:22:34.351 MQT: tele/tasmota_483F5C/STATE = {"Time":"2021-06-10T00:22:34","Uptime":"0T00:27:49","UptimeSec":1669,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"0,0,0,0,255","HSBColor":"0,0,0","White":100,"CT":500,"Channel":[0,0,0,0,100],"Scheme":0,"Fade":"ON","Speed":4,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxxx","BSSId":"3C:A6:2F:45:BA:3E","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:05"}}
00:22:34.392 MQT: stat/tasmota_483F5C/RESULT = {"POWER":"ON","Dimmer":100,"Color":"0,0,0,0,255","HSBColor":"0,0,0","White":100,"CT":500,"Channel":[0,0,0,0,100]}
```

### TO REPRODUCE
1) Use a Sonoff B1 bulb
2) Type in the console: Fade on
3) Type in the console: RGBWWTable 255,255,255,127,127
4) Set brightness to 100% using the web UI. Set the White-temperatuur to max COOL, and then to max WARM. 
5) Type in the console: RGBWWTable 255,255,255,128,128
6) Set brightness to 100% using the web UI. Set the White-temperatuur to max COOL, and then to max WARM.
5) Type in the console: RGBWWTable 255,255,255,255,255
6) Set brightness to 100% using the web UI. Set the White-temperatuur to max COOL, and then to max WARM.

### EXPECTED BEHAVIOUR
During step 4), the light turns bright cool white, then bright warm white. This is OK.
During step 6), the light goes off. Then, when switching from cool to warm, it goes on, then fades from cool to warm, and then switches off immediately.
During step 8), the light goes on. Then, when switching from cool to warm, it fades from cool to warm, then starts over at cool and fades to warm again.
I would expect that the 0...255 values of RGBWWTable define the ratios of the different channels, and that 255 is the maximum value. For the RGB channels, that is true, but for this bulb, the white channels seem to be off by a factor of two. Probably, on hardware level, fading is 7-bit for the white channels, but that should be transparent on the interface level. So a value of 255 in the console command, should internally be converted to 127 if applicable. 

### SCREENSHOTS

### ADDITIONAL CONTEXT
It seems like the cool and warm white LEDs in the B1 are controlled with a 7-bit PWM, while the firmware thinks (when the RGBWWTable is set to 255,255,255,255,255) that it is 8-bit. So the ...,128,128 setting makes it fade from 0 to 127 and then to 128 which is 0 again if the MSB is ignored. And when the table is set to ...,255,255, it fades from 0 to 127 and via 128 (=0) to 255 (=127). So you see a double fading pattern.

s-hadinger, 2021-06-10:
Sonoff B1 uses two chained MY92x1, so PWM is supposed to be 8 bits. It may be a protocol issue when using 2 chained MY92x1 that we've never detected.

Honestly the Sonoff B1 is not a great device, there is so much difference between RGB and White luminosity that is it hardly usable. I found many better bulbs than this one. I don't plan to spend time on solving this bug. Sorry

blakadder, 2021-06-15:
Its also discontinued

arendst, 2021-06-19:
Did the above tests with my "old" B1 and did not observe your issue. The light fades from cool to warm without issues.

Do you oberve this on just one bulb or on different bulbs?

jan-gerard, 2021-06-19:
> Did the above tests with my "old" B1 and did not observe your issue. The light fades from cool to warm without issues.
> Do you oberve this on just one bulb or on different bulbs?
I've got only one B1. I can fade from warm to cool and vice versa, the problem occurs when I fade past 50%. So actually, 50% is max, then 51% is off again, and 100% is again max. Weird, but if the white leds are 7-bit, that would explain this behaviour.

sfromis, 2021-06-19:
What is your `DimmerRange`? Maybe you could adjust that to have "less surprising" results.

jan-gerard, 2021-06-19:
> What is your `DimmerRange`? Maybe you could adjust that to have "less surprising" results.

MQT: stat/tasmota_483F5C/RESULT = {"DimmerRange":{"Min":0,"Max":100}}

sfromis, 2021-06-19:
I'd check the outcome with `DimmerRange 0 50` then. Easy to try out some values.

closed 2021-07-09 by ascillato2