arendst/Tasmota #12334: PN532 Reader :  unexpected termination

Noschvie, 2021-06-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
A PN532 reader connected to a NodeMCU terminates unexpected. A reboot doesn't repair, you have to save the template again to get the PN532 back to the web GUI and working.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU ESP8266
- [x] Tasmota binary firmware version number used: 9.4.0.5
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tasmotizer, OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:04:41.707 MQT: stat/tasmota_A45E62/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:04:41.929 MQT: stat/tasmota_A45E62/RESULT = {"Module":{"0":"Generic"}}
09:04:42.140 MQT: stat/tasmota_A45E62/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"2880":"PN532 Rx"},"GPIO14":{"1216":"AM2301"},"GPIO15":{"2848":"PN532 Tx"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
09:05:45.547 MQT: stat/tasmota_A45E62/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":39,"Free":472,"Rules":"on PN532#UID=12345678 do power on endon"}}
09:05:45.771 MQT: stat/tasmota_A45E62/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":54,"Free":457,"Rules":"ON Power1#State=1 DO BACKLOG DELAY 15; POWER OFF ENDON"}}
09:05:45.994 MQT: stat/tasmota_A45E62/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:07:03.494 MQT: stat/tasmota_A45E62/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_A45E62","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:07:03.507 MQT: stat/tasmota_A45E62/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:49:42","StartupUTC":"2021-06-10T06:17:21","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2021-04-21T18:08:11","SaveCount":145,"SaveAddress":"F7000"}}
09:07:03.518 MQT: stat/tasmota_A45E62/STATUS2 = {"StatusFWR":{"Version":"9.4.0.5(sensors)","BuildDateTime":"2021-06-08T20:11:43","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"407/699"}}
09:07:03.529 MQT: stat/tasmota_A45E62/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT_5G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["02008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
09:07:03.552 MQT: stat/tasmota_A45E62/STATUS4 = {"StatusMEM":{"ProgramSize":657,"Free":344,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
09:07:03.568 MQT: stat/tasmota_A45E62/STATUS5 = {"StatusNET":{"Hostname":"tasmota_A45E62-7778","IPAddress":"192.168.1.144","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:CC:A8:A4:5E:62","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:07:03.581 MQT: stat/tasmota_A45E62/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.43","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A45E62","MqttUser":"xyz","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:07:03.598 MQT: stat/tasmota_A45E62/STATUS7 = {"StatusTIM":{"UTC":"2021-06-10T07:07:03","Local":"2021-06-10T09:07:03","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"05:46","Sunset":"21:52"}}
09:07:03.614 MQT: stat/tasmota_A45E62/STATUS10 = {"StatusSNS":{"Time":"2021-06-10T09:07:03","AM2301":{"Temperature":27.4,"Humidity":37.5,"DewPoint":11.6},"TempUnit":"C"}}
09:07:03.630 MQT: stat/tasmota_A45E62/STATUS11 = {"StatusSTS":{"Time":"2021-06-10T09:07:03","Uptime":"0T00:49:42","UptimeSec":2982,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":36,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"IoT_5G","BSSId":"34:FC:B9:B7:14:80","Channel":6,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
.) restart device with PN532 reader
.) test reading
.) wait for hours...
.) retest reading not working

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
PN532 reader without termination

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
A AM2301 device is connected to the NodeMCU in parallel and works fine.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-06-10:
Without log info, details are scarce, but a board removing partial config details is usually a sign of a boot loop, which easily can happen if the power source is not stable.

Noschvie, 2021-06-13:
Yesterday the PN532 reader died and I have ordered a new one to continue the tests...

Noschvie, 2021-06-16:
New PN532 device connected and restarted the test with "9.4.0.6(sensors)"...

Noschvie, 2021-06-16:
Writing to a tag isn't possible:
```
16:54:55.473 CMD: status 0
16:54:55.481 MQT: stat/tasmota_A45E62/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_A45E62","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
16:54:55.495 MQT: stat/tasmota_A45E62/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T06:54:25","StartupUTC":"2021-06-16T09:00:30","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2021-06-10T10:31:45","SaveCount":46,"SaveAddress":"FA000"}}
16:54:55.505 MQT: stat/tasmota_A45E62/STATUS2 = {"StatusFWR":{"Version":"9.4.0.6(sensors)","BuildDateTime":"2021-06-16T06:48:00","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"383/699"}}
16:54:55.515 MQT: stat/tasmota_A45E62/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gentzgasse_5G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
16:54:55.539 MQT: stat/tasmota_A45E62/STATUS4 = {"StatusMEM":{"ProgramSize":662,"Free":340,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
16:54:55.555 MQT: stat/tasmota_A45E62/STATUS5 = {"StatusNET":{"Hostname":"tasmota_A45E62-7778","IPAddress":"192.168.1.144","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:CC:A8:A4:5E:62","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:54:55.571 MQT: stat/tasmota_A45E62/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.43","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A45E62","MqttUser":"Top24IoTDeviceUser","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:54:55.589 MQT: stat/tasmota_A45E62/STATUS7 = {"StatusTIM":{"UTC":"2021-06-16T15:54:55","Local":"2021-06-16T16:54:55","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:55"}}
16:54:55.605 MQT: stat/tasmota_A45E62/STATUS10 = {"StatusSNS":{"Time":"2021-06-16T16:54:55"}}
16:54:55.622 MQT: stat/tasmota_A45E62/STATUS11 = {"StatusSTS":{"Time":"2021-06-16T16:54:55","Uptime":"0T06:54:25","UptimeSec":24865,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":37,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Gentzgasse_5G","BSSId":"34:FC:B9:B7:14:80","Channel":6,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}
16:55:02.314 CMD: Sensor40 S,ILOVETASMOTA
16:55:02.321 MQT: stat/tasmota_A45E62/RESULT = {"Command":"Error"}
```
and erasing seems broken too:
```
16:56:34.696 CMD: Sensor40 E
16:56:34.703 MQT: stat/tasmota_A45E62/RESULT = {"Command":"Error"}
```

sfromis, 2021-06-16:
Did you compile with `#define USE_PN532_DATA_FUNCTION` to enable support for the Sensor40 command? As you ticked "pre-compiled", I'm assuming the answer to be no.

Noschvie, 2021-06-16:
You are right, I have to compile ... thanks.

Noschvie, 2021-06-17:
.. recompiled and Sensor40 command works fine.
Changed power supply to 3A, test restarted ...
PN532 device is connected to 3.3V (and not to 5V) isn't it ?

sfromis, 2021-06-17:
Read the specs before connecting anything....

Googling for a random PN532 devices suggests both voltages working, but that may be up to the design of the specific board.

Noschvie, 2021-06-18:
Last night the PN532 was disconnected again... will replace the NodeMCU with a new one (or an ESP32) and will config the syslog...

Noschvie, 2021-06-26:
Have replaced the NodeMCU with a new one, but same behaviour, after few hours the PN532 gets broken like this: "RestartReason":"Hardware Watchdog"
```
2021-06-26	12:07:36	Notice	tasmota_a4500c-4108.neunzehn.at	user	tasmota_A4500C-4108	ESP-MQT: tele/tasmota_A4500C/INFO3 = {"Info3":{"RestartReason":"Hardware Watchdog"}}
2021-06-26	12:07:36	Notice	tasmota_a4500c-4108.neunzehn.at	user	tasmota_A4500C-4108	ESP-MQT: tele/tasmota_A4500C/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_A4500C-4108","IPAddress":"192.168.1.150"}}
2021-06-26	12:07:36	Notice	tasmota_a4500c-4108.neunzehn.at	user	tasmota_A4500C-4108	ESP-MQT: tele/tasmota_A4500C/INFO1 = {"Info1":{"Module":"Generic","Version":"9.5.0.1(sensors)","FallbackTopic":"cmnd/DVES_A4500C_fb/","GroupTopic":"cmnd/tasmotas/"}}
```
But a "restart" with the GUI is needed to reconnect the PN532.

As a next step I will use a ESP32...

Noschvie, 2021-06-28:
... same with an ESP32.

Noschvie, 2021-07-02:
It seems that the PN532 is longer available in case of closed Web GUI.
Has anybody an idea of how to debug ? Shall I add debug outputs or checks of return values at xsns_40_pn532.ino ?

closed 2021-07-23 by github-actions[bot]