arendst/Tasmota #12390: How to set ESP32 Frequency

helmar74, 2021-06-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32
- [X ] Tasmota binary firmware version number used: _9.5.0____
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: VSC with PlatformIO Addon
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compile Tasmota 9.5.0 with the following platformio_override.h:

....
[common]
platform_packages           = ${core.platform_packages}
build_unflags               = ${core.build_unflags}
build_flags                 = ${core.build_flags}
; *** Optional Debug messages
;                            -DDEBUG_TASMOTA_CORE
;                            -DDEBUG_TASMOTA_DRIVER
;                            -DDEBUG_TASMOTA_SENSOR

; *** CAUTION *** This setting is ONLY possible since 12.01.2021 with development version !!!
; *** Enable only if you exactly know what are you doing
; *** If you try with earlier builds a serial erase and flash is probably needed
;
; Build variant 1MB = 1MB firmware no filesystem (default)
;board                       = esp8266_1M
; Build variant 2MB = 1MB firmware, 1MB filesystem (most Shelly devices)
;board                       = esp8266_2M1M
; Build variant 4MB = 1MB firmware, 1MB OTA, 2MB filesystem (WEMOS D1 Mini, NodeMCU, Sonoff POW)
;board                       = esp8266_4M2M

; set CPU frequency to 80MHz (default) or 160MHz
; CPU Frequenz auf 240 MHz einstellen:
board_build.f_cpu           = 240000000L
.....

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
ESP32 Running on 240MHz, but still running on 80 Mhz:
17:39:01.985 MQT: stat/esp32_1/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021.06.17 16:40:35","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":80,"Hardware":"ESP32-D0WD","CR":"463/699"}}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-06-17:
You are right. This entry does not work anymore since the change doing all config via boards.
I have to think about a easy solution for users to provide a different speed.

helmar74, 2021-06-17:
@Jason2866 :
Thank you, is there any possibility to get it set to 240 MHz at the moment?

helmar74, 2021-06-17:
esp32

Jason2866, 2021-06-17:
Change this line https://github.com/arendst/Tasmota/blob/development/boards/esp32_4M.json#L8
from `"f_cpu": "80000000L",` to `"f_cpu": "240000000L",`

helmar74, 2021-06-17:
Thanks, I'm trying, will provide feedback

helmar74, 2021-06-17:
It worked like charmed ;-)

`17:04:11.185 MQT: stat/esp32_1/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021.06.17 16:40:04","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":240,"Hardware":"ESP32-D0WD","CR":"463/699"}}`

closed 2021-06-17 by helmar74

Jason2866, 2021-06-18:
@helmar74  after merging the PR it is again possible to override settings with `platformio_override.ini` as it is intended to do!
Thx for opening the issue.

helmar74, 2021-06-18:
Thanks, great to hear. But currently in the development branch only?

Jason2866, 2021-06-18:
Yes, until next release. At the moment there are just minimal minor changes between release and development version.

helmar74, 2021-06-18:
Great Thanks a lot for fixing it so quickly

helmar74, 2021-06-21:
@Jason:
I tried the current development version with the following platformio_override.ini:
`[env:tasmota32_base]
framework               = ${common.framework}
platform                = ${common32.platform}
platform_packages       = ${common32.platform_packages}
; Build variant ESP32 4M Flash, Tasmota 1856k Code/OTA, 320k LITTLEFS (default)
board                   = ${common32.board}
; Build variant ESP32 8M Flash, Tasmota 2944k Code/OTA, 2112k LITTLEFS
;board                   = esp32_8M
; Build variant ESP32 16M Flash, Tasmota 2944k Code/OTA, 10M LITTLEFS
;board                   = esp32_16M
board_build.f_cpu       = 240000000L
;board_build.f_flash     = 40000000L
monitor_speed           = ${common32.monitor_speed}
upload_resetmethod      = ${common32.upload_resetmethod}
; *** Serial port used for erasing/flashing the ESP32
upload_port             = ${common32.upload_port}
;upload_port             = COM4
; upload_speed           = 115200
extra_scripts           = ${common32.extra_scripts}
build_unflags           = ${common32.build_unflags}
build_flags             = ${common32.build_flags} -DBOARD_HAS_PSRAM
lib_ldf_mode            = ${common32.lib_ldf_mode}
lib_compat_mode         = ${common32.lib_compat_mode}`

I only changed these entries:
`board_build.f_cpu       = 240000000L
build_flags             = ${common32.build_flags} -DBOARD_HAS_PSRAM
`

The CPU is running on 240 MHz, which is fine, but it isn't using PSRAM.
It worked with current master 9.5.0.

Do I have to change it on a other location?

Thank you in advance

helmar74, 2021-06-21:
I also tried this build flags:
build_flags             = ${common32.build_flags} -DBOARD_HAS_PSRAM __-mfix-esp32-psram-cache-issue -lc-psram-workaround -lm-psram-workaround__

I have connect a 10,1" TFT Display.
Now I can show pictures in size 180 x 180 dpi, but not with size 216 x 400. It worked well in Master 9.5.0 version.
Do you have a tip, where I can look in to get 9.5.0.1 working with bigger picture size=

Jason2866, 2021-06-21:
Change `board` to `esp32-cam` and add a additional entry `board_build.f_flash  = 40000000L` in your override if your board crashes since flash speed is set to 80Mhz.
If it runs well keep the high frequency. No change.
It doesnt matter that you have no esp32cam connected. Board definition is for choosing the correct settings. The settings are what your need

helmar74, 2021-06-22:
Hi again, I tried to dig a little bit deeper...
I set the following in platformio_override.ini:

[env:tasmota32_base]
framework               = ${common.framework}
platform                = ${common32.platform}
platform_packages       = ${common32.platform_packages}
; Build variant ESP32 4M Flash, Tasmota 1856k Code/OTA, 320k LITTLEFS (default)
board                   = ${common32.board}
; Build variant ESP32 8M Flash, Tasmota 2944k Code/OTA, 2112k LITTLEFS
;board                   = esp32_8M
; Build variant ESP32 16M Flash, Tasmota 2944k Code/OTA, 10M LITTLEFS
;board                   = esp32_16M
board_build.f_cpu       = 240000000L
;board_build.f_flash     = 40000000L
monitor_speed           = ${common32.monitor_speed}
upload_resetmethod      = ${common32.upload_resetmethod}
; *** Serial port used for erasing/flashing the ESP32
upload_port             = ${common32.upload_port}
;upload_port             = COM4
; upload_speed           = 115200
extra_scripts           = ${common32.extra_scripts}
build_unflags           = ${common32.build_unflags}
; Aktivierung von PSRAM mit build Flag -DBOARD_HAS_PSRAM, damit auch große Bilder angezeigt werden können
; WEitere Infos siehe https://github.com/arendst/Tasmota/issues/10392
build_flags             = ${common32.build_flags} __-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -lc-psram-workaround -lm-psram-workaround__
lib_ldf_mode            = ${common32.lib_ldf_mode}
lib_compat_mode         = ${common32.lib_compat_mode}

... this are the same buildflags as in webcam version.
But this doesn't compile in PSRAM usage:
[image]

If I compile master version of tasmota (9.5.0) the with __-DBOARD_HAS_PSRAM__ it's working:
[image]

Maybe there's a bug in the current development version, that the switch __-DBOARD_HAS_PSRAM__ is ignored in platformio_override.ini?

helmar74, 2021-06-22:
I'm said to say, but I think im right.

I changed platformio_override.ini back to "default":
[env:tasmota32_base]
framework               = ${common.framework}
platform                = ${common32.platform}
platform_packages       = ${common32.platform_packages}
; Build variant ESP32 4M Flash, Tasmota 1856k Code/OTA, 320k LITTLEFS (default)
board                   = ${common32.board}
; Build variant ESP32 8M Flash, Tasmota 2944k Code/OTA, 2112k LITTLEFS
;board                   = esp32_8M
; Build variant ESP32 16M Flash, Tasmota 2944k Code/OTA, 10M LITTLEFS
;board                   = esp32_16M
board_build.f_cpu       = 240000000L
;board_build.f_flash     = 40000000L
monitor_speed           = ${common32.monitor_speed}
upload_resetmethod      = ${common32.upload_resetmethod}
; *** Serial port used for erasing/flashing the ESP32
upload_port             = ${common32.upload_port}
;upload_port             = COM4
; upload_speed           = 115200
extra_scripts           = ${common32.extra_scripts}
build_unflags           = ${common32.build_unflags}
__build_flags             = ${common32.build_flags} __
lib_ldf_mode            = ${common32.lib_ldf_mode}
lib_compat_mode         = ${common32.lib_compat_mode}

Than I changed __platformio_tasmota_env32.ini__ that way:
[env:tasmota32]
extends                 = env:tasmota32_base
build_flags             = ${common32.build_flags} __-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -lc-psram-workaround -lm-psram-workaround -DFIRMWARE_TASMOTA32__

This brings back PSRAM in Version 9.5.0.1:
[image]

So I think, that changing build_flags in platformio_override.ini isn't recognized at the moment.
Maybe you can have a look on it?

Thank you in advance.

Jason2866, 2021-06-22:
@helmar74  Cant try today (not on PC). It should work. There is no reason for not working.
Maybe Platformio parsing related. Scanning for more options in one line do sometimes not work.
Please try:
```
build_flags = ${common32.build_flags}
                        -DBOARD_HAS_PSRAM
                        -mfix-esp32-psram-cache-issue
                        -lc-psram-workaround
                        -lm-psram-workaround
```
If you look in the other parts of the setup. We avoid more settings in one line.

helmar74, 2021-06-22:
@Jason2866 :
Dividing it up, as you suggested, doesn't work either.

Jason2866, 2021-06-22:
Mhh, no idea at the moment. Will have a look when i am back to PC and some free time...

helmar74, 2021-06-22:
Great to hear, if I can test something, just tell me.

Jason2866, 2021-06-22:
think i found it :-)  Since with the last needed changes the section `common` and `common32` are not redefined in override anymore. All [env] uses for build_flags the source `common` and `common32`. Since this is NOT changed in override anymore this has no effect!!
Had some time and could use a Laptop from a friend PR #12442

helmar74, 2021-06-22:
OK, if I download current development 9.5.0.1 newly, can I then try it?
(If so, will do it tomorrow and provide feedback)

Thanks a lot for your good work!!!

Jason2866, 2021-06-22:
If the change is merged, you can try. If you want to try now use my branch with the changes.
https://github.com/Jason2866/Tasmota/tree/fix_override

helmar74, 2021-06-23:
@Jason2866 

I tried it. ;-)
I used my own user_config_override.h (same one as for Tasmota 9.5.0)
I changed platformio_override.ini that way:
.....
[env:tasmota32_base]
framework               = ${common.framework}
platform                = ${core32.platform}
platform_packages       = ${core32.platform_packages}
; Build variant ESP32 4M Flash, Tasmota 1856k Code/OTA, 320k LITTLEFS (default)
board                   = esp32_4M
; Build variant ESP32 8M Flash, Tasmota 2944k Code/OTA, 2112k LITTLEFS
;board                   = esp32_8M
; Build variant ESP32 16M Flash, Tasmota 2944k Code/OTA, 10M LITTLEFS
;board                   = esp32_16M
__board_build.f_cpu       = 240000000L__
;board_build.f_flash     = 40000000L
monitor_speed           = 115200
; *** Serial port used for erasing/flashing the ESP32
;upload_port             = ${common.upload_port}
upload_port             = COM4
;upload_speed            = 115200
upload_resetmethod      = ${common.upload_resetmethod}
extra_scripts           = ${common.extra_scripts}
build_unflags           = ${esp32_defaults.build_unflags}
build_flags             = ${esp32_defaults.build_flags} __-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -lc-psram-workaround -lm-psram-workaround__
lib_ldf_mode            = ${common.lib_ldf_mode}
lib_compat_mode         = ${common.lib_compat_mode}
.....

Now everything works as expected, CPU is running on 240MHz and PSRAM is present.
[image]

Thank you very much for reacting and fixing it so fast.

Kind regards

helmar

Jason2866, 2021-06-23:
It is the new core. I stripped uneeded code out. No fear, nothing is missing

Jason2866, 2021-06-23:
@helmar74 One request could you try, if PSRAM works correctly only with this option? 
```
build_flags = ${esp32_defaults.build_flags} -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue
```
I have found very different statements what is correct with Platformio. Latest Platformio (we use) should?! add the needed correct PSRAM fixing code with this. Unnecessary code is added with the sequence we use now. Would be great if you could test.
Since I don't have a module that has any PSRAM error, I can't test it myself.

helmar74, 2021-06-23:
@Jason2866 
I tried it. The compiled firmware is 5 KB smaller.
PSRAM is shown in Info of the device like before:
[image]

I also tried my use case. If PSRAM isn't present "bigger" pictures (e.g. 400x400 px) aren't shown. But this also worked with your suggested settings. So I think adding only these 2 build flags should enable PSRAM.

Kind Regards

helmar