arendst/Tasmota #12391: Trigger Tele-Wifi#SSId not fired when MQTT server disconnected

lalo-uy, 2021-06-17:
### PROBLEM DESCRIPTION
the trigger Tele-Wifi#SSId  is not fired if connected to an AP without MQTT server

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Wemos D1
- [X ] Tasmota binary firmware version number used: 9.4.0(display)
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:54:57.965 CMD: Backlog Template; Module; GPIO 255
14:54:58.023 MQT: stat/Maserador
/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}

14:54:58.273 MQT: stat/Maserador/RESULT = {"Module":{"18":"Gener
ic"}}

14:54:58.528 MQT: stat/Maserador/RESULT = {"GPIO0":{"0":"None"},
"GPIO1":{"0":"None"},"GPIO2":{"256":"Relay_i1"},"GPIO3":{"0":"None"},"GPIO4":{"6624":"TM1637 CLK"},"GPIO5":{"6656":"TM1637 DIO"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1312":"DS18x20"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
Rule1":{"State":"ON"
,"Once":"OFF","StopOnError":"OFF","Length":187,"Free":324,"Rules":"ON Tele-DS18B20#Temperature DO  Backlog Var1 %value% ; ADD1 0.5 ; DisplayText %value%^ ; RuleTimer1 5 ENDON ON rules#timer=1 DO Backlog DisplayNumber %Var1%,1,0,3 ; DisplayTextNC  ^ ENDON"}}

"Rule2":{"State":"OFF
","Once":"OFF","StopOnError":"OFF","Length":47,"Free":464,"Rules":"ON tele-wifi#SSId $! MyMainAp do Ap 0 endone"}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:57:33.026 CMD: status 0
14:57:33.038 MQT: stat/Maserador
/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"Maserador","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:57:33.075 MQT: stat/Maserador/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:11:19","StartupUTC":"2021-06-17T13:46:14","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2021-06-05T14:35:10","SaveCount":124,"SaveAddress":"F8000"}}
14:57:33.108 MQT: stat/Maserador/STATUS2 = {"StatusFWR":{"Version":"9.4.0(display)","BuildDateTime":"2021-04-23T10:07:21","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"376/699"}}
14:57:33.134 MQT: stat/Maserador/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["pokerplanta","Lalo"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
14:57:33.164 MQT: stat/Maserador/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":30,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAA858E","04049FA1","000000C3","00000000","00004080","00000020","401F8000","00000000"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6"}}
14:57:33.200 MQT: stat/Maserador/STATUS5 = {"StatusNET":{"Hostname":"Maserador-6042","IPAddress":"192.168.1.6","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:6E:97:9A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:57:33.224 MQT: stat/Maserador/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6E979A","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:57:33.250 MQT: stat/Maserador/STATUS7 = {"StatusTIM":{"UTC":"2021-06-17T13:57:33","Local":"2021-06-17T14:57:33","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:55"}}
14:57:33.272 MQT: stat/Maserador/STATUS10 = {"StatusSNS":{"Time":"2021-06-17T14:57:33"}}
14:57:33.286 MQT: stat/Maserador/STATUS11 = {"StatusSTS":{"Time":"2021-06-17T14:57:33","Uptime":"0T00:11:19","UptimeSec":679,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":2,"SSId":"Lalo","BSSId":"E4:47:B3:B4:B6:B8","Channel":7,"RSSI":100,"Signal":-50,"LinkCount":2,"Downtime":"0T00:00:32"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Connect to an AP and turn off the MQTT server.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The Rule2 fiers the AP switch

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-06-17:
Your rule is wrong. It is `ENDON` not `endone`

lalo-uy, 2021-06-17:
You are right, but it works anyway, as long as the MQTT server is connected.
May be should use Wifi#Connected instead of Tele-Wifi?

lalo-uy, 2021-06-17:
Wifi#Connected returns 1 . I need the SSID

arendst, 2021-06-17:
Execute command ``Teleperiod`` to start regular teleperiod reporting when mqtt is down.

lalo-uy, 2021-06-17:
yes, the teleperiod is started, but yet the rule do not match.
[cutecom.log](https://github.com/arendst/Tasmota/files/6671195/cutecom.log)

arendst, 2021-06-17:
No problem here with a rule like ``on wifi#connected do teleperiod endon on tele-wifi#ssid do publish cmnd/theo/ssid %value% endon`` an no valid mqtthost:

```
00:00:00.001 HDW: ESP8266EX
00:00:00.051 CFG: Loaded from flash at F6, Count 155
00:00:00.056 QPC: Count 1
00:00:00.061 SRC: Restart
00:00:00.063 Project tasmota Wemos10 Version 9.5.0.1(theo)-(2021-06-17T16:44:23)
00:00:00.079 DSB: Sensors found 1
00:00:00.114 I2C: HTU21 found at 0x40
00:00:00.439 WIF: Attempting connection...
00:00:00.778 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as demo...
00:00:01.760 WIF: Connected
00:00:01.817 RUL: WIFI#CONNECTED performs "teleperiod"
00:00:01.818 SRC: Rule
00:00:01.819 CMD: Grp 0, Cmnd 'TELEPERIOD', Idx 1, Len 0, Data ''
00:00:01.821 RSL: RESULT = {"TelePeriod":300}
00:00:02.014 HTP: Web server active on demo with IP address 192.168.2.153
00:00:02.165 RSL: STATE = {"Time":"1970-01-01T00:00:02","Uptime":"0T00:00:04","UptimeSec":4,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}
00:00:02.193 RUL: TELE-WIFI#SSID performs "publish cmnd/theo/ssid INDEBUURT_IOT"
00:00:02.194 SRC: Rule
00:00:02.195 CMD: Grp 0, Cmnd 'PUBLISH', Idx 1, Len 28, Data 'cmnd/theo/ssid INDEBUURT_IOT'
00:00:02.202 RSL: ssid = INDEBUURT_IOT
00:00:02.209 RSL: SENSOR = {"Time":"1970-01-01T00:00:02","DS18B20":{"Id":"01144A0CB2AA","Temperature":29.50},"HTU21":{"Temperature":29.54,"Humidity":52.2,"DewPoint":18.68},"TempUnit":"C"}
00:00:02.243 RTC: UTC 2021-06-17T15:05:37, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
17:05:38.014 MQT: Attempting connection...
17:05:38.020 MQT: Connect failed to theo1:1883, rc -2. Retry in 10 sec
17:05:41.226 QPC: Reset
17:05:43.170 APP: Boot Count 66
17:05:43.583 CFG: Saved to flash at F5, Count 156, Bytes 4096
17:05:49.007 MQT: Attempting connection...
```

lalo-uy, 2021-06-18:
Thanks Theo, now this is working:

rule1 on Wifi#Connected do teleperiod  endon ON tele-wifi#SSId $! MyInternalAp do backlog websend [myExternalServer.com] /internalAlarm; ap 0 endon

closed 2021-06-18 by lalo-uy