arendst/Tasmota #12413: Issue with Sub command with bit integer

ggirou, 2021-06-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sub command doesn't work with big integer.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 2.5
- [X] Tasmota binary firmware version number used: 9.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: esptool.py
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

21:41:11.953 MQT: stat/tasmota_bureau/RESULT = {"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}
21:41:12.192 MQT: stat/tasmota_bureau/RESULT = {"Module":{"0":"Shelly 2.5"}}
21:41:12.447 MQT: stat/tasmota_bureau/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"32":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"193":"Switch_n2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"192":"Switch_n1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ"},"GPIO17":{"4736":"ADC Temp"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

21:43:44.621 MQT: stat/tasmota_bureau/RESULT = {"Rule1":{"State":"ON","Once":"ON","StopOnError":"OFF","Length":132,"Free":379,"Rules":"ON ENERGY#Current[1]>0 DO Var1 %utctime% ENDON   ON ENERGY#Current[1]<0.400 DO Backlog Var2 %utctime%; Sub2 %Var1%; Event up=1 ENDON"}}
21:43:44.844 MQT: stat/tasmota_bureau/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
21:43:45.045 MQT: stat/tasmota_bureau/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":60,"Free":451,"Rules":"ON event#up DO publish stat/shutter/calibration %Var1% ENDON"}}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

21:44:20.732 MQT: stat/tasmota_bureau/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota_bureau","FriendlyName":["tasmota_bureau","Tasmota2"],"Topic":"tasmota_bureau","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:44:20.752 MQT: stat/tasmota_bureau/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:18:57","StartupUTC":"2021-06-18T20:25:23","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2021-04-21T18:31:05","SaveCount":668,"SaveAddress":"F8000"}}
21:44:20.769 MQT: stat/tasmota_bureau/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"526/699"}}
21:44:20.787 MQT: stat/tasmota_bureau/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXX",""],"TelePeriod":300,"Resolution":"558180F0","SetOption":["0000C009","2805C8000100060000005A0A000000000000","40000080","00006000","00000000"]}}
21:44:20.809 MQT: stat/tasmota_bureau/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
21:44:20.827 MQT: stat/tasmota_bureau/STATUS5 = {"StatusNET":{"Hostname":"tasmota_bureau","IPAddress":"192.168.1.42","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"E8:DB:84:A9:B1:05","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:44:20.843 MQT: stat/tasmota_bureau/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A9B105","MqttUser":"iot","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:44:20.862 MQT: stat/tasmota_bureau/STATUS7 = {"StatusTIM":{"UTC":"2021-06-18T20:44:20","Local":"2021-06-18T21:44:20","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:56"}}
21:44:20.875 MQT: stat/tasmota_bureau/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:44:20.898 MQT: stat/tasmota_bureau/STATUS10 = {"StatusSNS":{"Time":"2021-06-18T21:44:20.889","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":62.1},"ENERGY":{"TotalStartTime":"2021-04-21T18:31:05","Total":0.089,"Yesterday":0.002,"Today":0.005,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":0,"Direction":0,"Target":0},"TempUnit":"C"}}
21:44:20.919 MQT: stat/tasmota_bureau/STATUS11 = {"StatusSTS":{"Time":"2021-06-18T21:44:20.913","Uptime":"0T00:18:57","UptimeSec":1137,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"XXX","BSSId":"88:C3:97:83:E8:FD","Channel":1,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
21:44:20.936 MQT: stat/tasmota_bureau/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":290,"Close":260,"50perc":66,"Delay":0,"Opt":"0100","Calib":[1860,394,4114,900,1000],"Mode":"0"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

21:45:47.136 CMD: Backlog Var1 1624046871; Sub1 1624046861
21:45:47.138 SRC: WebConsole from 192.168.1.5
21:45:47.141 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 32, Data 'Var1 1624046871; Sub1 1624046861'
21:45:47.163 SRC: Backlog
21:45:47.166 CMD: Grp 0, Cmnd 'VAR', Idx 1, Len 10, Data '1624046871'
21:45:47.172 MQT: stat/tasmota_bureau/RESULT = {"Var1":"1624046871"}
21:45:47.209 ADE: 6486190, 4478, [1773, -94, 685, 4], [1773, 9, 685, 1]
21:45:47.211 ADE: U 6486190, C 4478, I 0 + 0 = 0, P 0 + 0 = 0
21:45:47.394 SRC: Backlog
21:45:47.397 CMD: Grp 0, Cmnd 'SUB', Idx 1, Len 10, Data '1624046861'
21:45:47.404 MQT: stat/tasmota_bureau/RESULT = {"Sub1":"0.000"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

```
Backlog Var1 1624046871; Sub1 1624046861
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Result of `Var1 1624046871; Sub1 1624046861` should be `10` but is `0`

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I'm trying to calculate elapsed time between two events in rules using `%utctime%` value.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-06-18:
Not a bug, merely a limitation due to the math being single precision floating point, meaning 7.1 digits. Your values are being truncated before the subtraction, thus you lose digits from the right.

ascillato2, 2021-06-19:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-06-19 by ascillato2

ggirou, 2021-06-19:
I wasn't aware of this limitation because I'm working with integer.
I found this related issue https://github.com/arendst/Tasmota/issues/3083 that adds decimals support, but we lost in precision for integer.

Is it possible to support both integer and double precision for ADD, SUB, MULT, ... commands?
Maybe looking for a dot in the value to differentiate integer and double?

sfromis, 2021-06-19:
Well, the phrase "not possible" is not used lightly, but it would be a significant change to add typing to numbers, and the simple concept of detecting a dot in values would be a breaking change to current usage of the math functions.

The Berry scripting language now part of Tasmota on ESP32 does support 31-bit integers (9.1 digits), along with single precision floating point.

Perhaps your use case has a simpler alternate way of doing it.

ascillato, 2021-06-19:
>Is it possible to support both integer and double precision for ADD, SUB, MULT, ... commands?
>Maybe looking for a dot in the value to differentiate integer and double?

That is good idea but how to solve the issue when you use mixed types, like adding a float to a double integer? in that case you will lose precision anyway.

ggirou, 2021-06-19:
I'll like to autocalibrate my shutters with few rules, so I want to calculate elapsed time between opened and closed events, then call `ShutterOpenDuration` or `ShutterCloseDuration`. It would be a pleasure to share these rules if they work :)

One way is to substract end time to start time using `%utctime%` value. But I have this precision lost issue.
Another way could be using a stopwatch, but I didn't find a command for this.

> That is good idea but how to solve the issue when you use mixed types, like adding a float to a double integer? in that case you will lose precision anyway.

I agree but in my case I have only integers.

sfromis, 2021-06-19:
You can have a stopwatch function using `PulseTime` with a "long enough" interval. At the stop event, you can reissue the command to get the remaining time, which you can pick up using a trigger like `PulseTime1#Remaining`. Now you have two small numbers to subtract from each other, and if no more than 11 seconds, you can even get it in 0.1 second increments for much better resolution than using %utctime%. 

You can use `RuleTimer` a very similar way, without needing a pin (could be dummy) for the PulseTime, but you do not get the 0.1 second increments.

ggirou, 2021-06-20:
I'll give it a try, thanks for your help :)

sfromis, 2021-06-20:
BTW. I found a trick for 31-bit integer add/subtract. Config a dummy counter pin, and
`on event#start do backlog Counter1 0; Counter1 -%utctime% endon`
`on event#stop do Counter1 +%utctime% endon`
After invoking start and stop, result in format: `{"Counter1":4}`
Still does not give you the good resolution available with PulseTime.

hoepfner, 2021-07-11:
I've a similar problem to get number of seconds between two events. The two workarounds (Counter, Pulsetime) are good ideas but have some difficulties. How can I solve the switch from Seconds to millseconds in rules, or how do I get the value from counter to a varX to use it for calculate. It would be easier to use the %Uptime% instead of %utctime% , because the number of seconds is not so big. but %uptime% is in minutes.
Is it possible to add a new Replacement in Rules to get UptimeS in Seconds
  https://github.com/arendst/Tasmota/blob/5a5843b5ab9be3e1b93f468a939b48c582c36aed/tasmota/xdrv_10_rules.ino#L761
  RulesVarReplace(commands, F("%UPTIMES%"), String(Uptime()));

sfromis, 2021-07-11:
Using regular rule facilities, JSON-based triggers, you can easily get a counter value into a var. And the Mult command allows multiplication (like from seconds to milliseconds), even without needing a binary built to include support for expressions.

hoepfner, 2021-07-12:
Thank you for the fast answer, but i dont get it - can you please help me, with the steps in rule-syntax to get the number of seconds between two events in a varX?

hoepfner, 2021-07-13:
Thank you for the help! I've found a solution with ruletimer and JSON-based triggers. 
so dont have a switch from seconds to microseconds, and dont need a gpio for counter.