arendst/Tasmota #12414: Error settings default module

loxK, 2021-06-19:
### PROBLEM DESCRIPTION
I am trying to automatically build for multiple devices, with their configuration being the default configuration, so in case of a reset, they keep the proper settings. I have been following the documentation "Customize your build".

I have this: 

```
#ifdef MODULE
#undef MODULE
#endif
#define MODULE SONOFF_POW_R2
```

And get this error when compiling:

```
tasmota/../../configs/sonoff-pow-tv.h:7:32: error: 'SONOFF_POW_R2' was not declared in this scope
 #define MODULE                 SONOFF_POW_R2
```

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Pow R2
- [x] Tasmota binary firmware version number used: 9.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: WebUI
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- [ ] Provide the output of this command: `Status 0`:
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:

### TO REPRODUCE

**user_config_override.h**
```
#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

// force the compiler to show a warning to confirm that this file is included
#warning **** user_config_override.h: Using Settings from this File ****

#include "../../configs/common.h"
#include "../../configs/sonoff-pow-bay.h"
#include "../../configs/sonoff-pow-tv.h"
#endif  // _USER_CONFIG_OVERRIDE_H_
```

**sonoff-pow-tv.h**
```
#ifdef _GAIC_CONFIG_SONOFF_POW_TV_H_

// force the compiler to show a warning to confirm that this file is included
#warning **** sonoff-pow-tv.h: Using Settings from this File ****

#ifdef MODULE
#undef MODULE
#endif
#define MODULE SONOFF_POW_R2

#ifdef FRIENDLY_NAME
#undef FRIENDLY_NAME
#endif
#define FRIENDLY_NAME "SonOff TV"

#ifdef MQTT_CLIENT_ID
#undef MQTT_CLIENT_ID
#endif
#define MQTT_CLIENT_ID "sonoffp-tv"

#ifdef MQTT_TOPIC
#undef MQTT_TOPIC
#endif
#define MQTT_TOPIC "sonoffp-tv"

//#undef OTA_URL
//#define OTA_URL "http://ota.tasmota.com/tasmota/release/tasmota.bin"  // [OtaUrl]

#ifdef APP_POWERON_STATE
#undef APP_POWERON_STATE
#endif
#define APP_POWERON_STATE POWER_ALL_ALWAYS_ON

#endif  // _GAIC_CONFIG_SONOFF_POW_TV_H_
```

### EXPECTED BEHAVIOUR
It should compile without error

sfromis, 2021-06-19:
Did you try it the documented way, having your settings in user_config_override.h?

I'd not expect usual C++ programming techniques to work the same, due to Tasmota originally being built with the Arduino IDE which changes how scoping works, with global scope encompassing a lot.

Jason2866, 2021-06-19:
As @sfromis  wrote not a bug. Platformio has to mimic the Arduino IDE build way to be compatible.
Includes work different and are not possible everywhere. If you find a solution for your tried attempt.
let us know. Until know, i know no way.

ascillato2, 2021-06-19:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-06-19 by ascillato2

loxK, 2021-06-21:
Hello, thanks.

The includes weren't the issue, I was using `extends = env:tasmota32_base` in `platformio_tasmota_cenv.ini`. Changing it to `extends = env:tasmota` works as it.