arendst/Tasmota #12415: Switchmode 9 does not work

Dome2075, 2021-06-19:
### PROBLEM DESCRIPTION
I use a Blitzwolf SS5  device with switchmode 1 and SwitchDebounce 609. All is working fine
Now i want to add the hold state so changed dt switchmode 9.

After i press the switch nothing happens any more and there is also no activity in the log.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Blitzwolf SS5
- [x] Tasmota binary firmware version number used: 9.5.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
11:12:19.609 CMD: SwitchMode1 9
11:12:19.615 MQT: stat/%topic%/RESULT = {"SwitchMode1":9}
11:12:38.629 CMD: SwitchMode2 9
11:12:38.635 MQT: stat/%topic%/RESULT = {"SwitchMode2":9}
Switch pressed
Switch pressed
11:15:41.329 CMD: SwitchMode1 1
11:15:41.335 MQT: stat/%topic%/RESULT = {"SwitchMode1":1}
11:15:47.186 CMD: SwitchMode2 1
11:15:47.193 MQT: stat/%topic%/RESULT = {"SwitchMode2":1}
Switch pressed
11:15:51.821 MQT: stat/%topic%/RESULT = {"POWER1":"ON"}
11:15:51.827 MQT: stat/%topic%/POWER1 = ON
Switch pressed
11:15:55.482 MQT: stat/%topic%/RESULT = {"POWER1":"OFF"}
11:15:55.486 MQT: stat/%topic%/POWER1 = OFF
Switch pressed
11:15:57.394 MQT: stat/%topic%/RESULT = {"POWER1":"ON"}
11:15:57.398 MQT: stat/%topic%/POWER1 = ON

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-06-23:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2021-06-25:
Please, when you have time, complete the template to have more information so as to properly help you. Thanks.

closed 2021-06-25 by ascillato2

Dome2075, 2021-07-05:
Post already updated!