arendst/Tasmota #12435: Serial buffer overrun

aug70, 2021-06-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I'm seeing below messages repeatedly on the Tasmota console.
```
21:40:09.190 CMD: 21:40:09.130 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}
21:40:09.197 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}
21:40:09.211 CMD: Serial buffer overrun
21:40:09.221 CMD: Serial buffer overrun
21:40:09.231 CMD: Serial buffer overrun
21:40:09.243 CMD: Serial buffer overrun
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: _9.4.0.6_
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:43:00.112 CMD: Backlog Template; Module; GPIO 255
21:43:01.839 CMD: Serial buffer overrun
21:43:01.851 CMD: 21:43:01.796 CMD: Serial buffer overrun

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:43:53.195 CMD: status 0
21:43:53.204 MQT: stat/tasmota_867FC6/STATUS = {"Status":{"Module":0,"DeviceName":"Plug 3","FriendlyName":["Plug 3-1","Plug 3-2","Plug 3-3","USB Right","USB Left"],"Topic":"tasmota_867FC6","ButtonTopic":"0","Power":25,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:43:53.243 MQT: stat/tasmota_867FC6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:47","StartupUTC":"2021-06-21T20:30:06","Sleep":50,"CfgHolder":4617,"BootCount":45,"BCResetTime":"2021-02-08T20:03:59","SaveCount":709,"SaveAddress":"F5000"}}
21:43:53.276 MQT: stat/tasmota_867FC6/STATUS2 = {"StatusFWR":{"Version":"9.4.0.6(tasmota)","BuildDateTime":"2021-06-16T06:48:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"394/699"}}
21:43:53.295 MQT: stat/tasmota_867FC6/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["pingpong",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
21:43:53.329 MQT: stat/tasmota_867FC6/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
21:43:53.365 MQT: stat/tasmota_867FC6/STATUS5 = {"StatusNET":{"Hostname":"tasmota_867FC6-8134","IPAddress":"192.168.1.85","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.200","Mac":"68:C6:3A:86:7F:C6","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:43:53.387 MQT: stat/tasmota_867FC6/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.199","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_867FC6","MqttUser":"DVES_USER","MqttCount":14,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:43:53.412 MQT: stat/tasmota_867FC6/STATUS7 = {"StatusTIM":{"UTC":"2021-06-21T20:43:53","Local":"2021-06-21T21:43:53","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:57"}}
21:43:53.431 MQT: stat/tasmota_867FC6/STATUS10 = {"StatusSNS":{"Time":"2021-06-21T21:43:53"}}
21:43:53.440 MQT: stat/tasmota_867FC6/STATUS11 = {"StatusSTS":{"Time":"2021-06-21T21:43:53","Uptime":"0T00:13:47","UptimeSec":827,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":14,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","POWER5":"ON","Wifi":{"AP":1,"SSId":"pingpong","BSSId":"54:A0:50:59:91:28","Channel":2,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:03"}}}
21:43:55.160 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-06-21:
Don't worry about those component numbers, the bigger numbers are simply a new numbering system, as the old system with values less than 255 could no longer accommodate all features of Tasmota. The old numbers are automatically converted to the new system if you input an old template.

Looks like the device may inappropriately have crosstalk from tx to rx. I'd suggest `SerialLog 0`

aug70, 2021-06-21:
I saw similar tickets and `SerialLog 0` recommendations. I tried it already couple times, it didn't seem to make any difference...

ascillato, 2021-06-21:
> I saw similar tickets and SerialLog 0 recommendations. I tried it already couple times, it didn't seem to make any difference...

Please, complete the step of the template that say:

**Set weblog to 4 and then, when you experience your issue, provide the output of the Console log:**

aug70, 2021-06-22:
@ascillato updated the ticket...

ascillato, 2021-06-22:
Great, Thanks.

Please, can you now post the output of the console of the command 'SerialLog 0' ?

aug70, 2021-06-22:
See below
```
20:30:49.375 CMD: SerialLog 0
20:30:49.378 SRC: WebConsole from xxx.xxx.xxx.xxx
20:30:49.380 CMD: Grp 0, Cmnd 'SERIALLOG', Idx 1, Len 1, Data '0'
20:30:49.384 MQT: stat/tasmota_867FC6/RESULT = {"SerialLog":{"0":{"Active":"0"}}}
20:30:51.674 CMD: Serial buffer overrun
20:30:51.692 CMD: 20:30:51.630 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}
20:30:51.694 SRC: Serial
20:30:51.697 CMD: Grp 0, Cmnd '', Idx 20, Len 66, Data ':30:51.630 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}'
20:30:51.702 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}
20:30:51.716 CMD: Serial buffer overrun
20:30:51.725 CMD: Serial buffer overrun
20:30:51.735 CMD: Serial buffer overrun
20:30:51.747 CMD: Serial buffer overrun
```

ascillato, 2021-06-22:
Thanks.

Please, can you post the output of `Backlog Template; Module; GPIO 255`

aug70, 2021-06-23:
Can you tell me what exactly are you looking at? Because not sure any of those seems to be helpful...
```
00:58:11.559 CMD: Backlog Template; Module; GPIO 255
00:58:11.562 SRC: WebConsole from xxx.xxx.xxx.xxx
00:58:11.565 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
00:58:13.793 SRC: Serial
00:58:13.795 CMD: Grp 0, Cmnd '', Idx 0, Len 37, Data ':58:13.733 CMD: Serial buffer overrun'
00:58:13.799 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}
00:58:13.812 CMD: Serial buffer overrun
00:58:13.823 CMD: Serial buffer overrun
00:58:13.834 CMD: 00:58:13.779 CMD: Serial buffer overrun
00:58:13.836 SRC: Serial
00:58:13.839 CMD: Grp 0, Cmnd '', Idx 0, Len 37, Data ':58:13.779 CMD: Serial buffer overrun'
00:58:13.843 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}
00:58:13.857 CMD: Serial buffer overrun
00:58:13.874 CMD: 00:58:13.812 CMD: Serial buffer overrun
00:58:13.877 SRC: Serial
00:58:13.879 CMD: Grp 0, Cmnd '', Idx 0, Len 37, Data ':58:13.812 CMD: Serial buffer overrun'
00:58:13.884 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}
00:58:13.897 CMD: Serial buffer overrun
00:58:13.909 CMD: Serial buffer overrun
00:58:13.920 CMD: 00:58:13.857 CMD: Serial buffer overrun
00:58:13.922 SRC: Serial
00:58:13.924 CMD: Grp 0, Cmnd '', Idx 0, Len 37, Data ':58:13.857 CMD: Serial buffer overrun'
00:58:13.929 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}
00:58:13.943 CMD: Serial buffer overrun
00:58:13.954 CMD: 00:58:13.897 CMD: Serial buffer overrun
00:58:13.956 SRC: Serial
00:58:13.959 CMD: Grp 0, Cmnd '', Idx 0, Len 37, Data ':58:13.897 CMD: Serial buffer overrun'
00:58:13.962 MQT: stat/tasmota_867FC6/RESULT = {"Command":"Unknown"}
00:58:13.976 CMD: Serial buffer overrun
00:58:13.988 CMD: Serial buffer overrun
00:58:13.998 CMD: 00:58:13.943 CMD: Serial buffer overrun
```

ascillato, 2021-06-23:
Hi, I'm trying to find where the configuration is wrong.

In your template or in the previous comment, the output of the GPIO assignment was not there because of that terrible crosstalk you have.

Please, post the screenshoot of the module selection, and the GPIO assignment in the template. 

All that in order to find, where that crosstalk is coming from.

aug70, 2021-06-23:
https://templates.blakadder.com/WP40.html

[image]
[image]
[image]

ascillato, 2021-06-23:
Ok. So your settings are correct.

Please, add in GPIO 1 and in GPIO 3 configurations, LED outputs to avoid Tasmota reading those pins.

In your hardware seems that TX and RX pins are soldered together.

aug70, 2021-06-23:
Thank you very much. I made the changes you recommended, it stopped throwing those messages. I have to say I'm really impressed with Tasmota and thank you very much for your help!!!. This kind of dedication makes this project very successful. When I find the time I will inspect the hardware to find if I caused a short-circuit between TX and RX. I will report back on that later. Once again, thank you for your time and help.

closed 2021-06-23 by ascillato2

RaunoVV, 2023-01-15:
> Ok. So your settings are correct.
> 
> Please, add in GPIO 1 and in GPIO 3 configurations, LED outputs to avoid Tasmota reading those pins.
> 
> In your hardware seems that TX and RX pins are soldered together.

Gpio 1 & 3 assigning fixed indeed! Using Nedis IR remote