arendst/Tasmota #12440: Rules stop working with NO VALID JSON after 9.5.0

joeH68, 2021-06-22:
### PROBLEM DESCRIPTION
Large rule set created under 9.4.0 stops working after upgrading to 9.5.0.  Rule1 command returns NO VALID JSON

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):  SONOFF RFbridge, SONOFF ZBbridge, AUVISIO IR bridge
- [x] Tasmota binary firmware version number used: 9.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

04:23:28.680 CMD: Backlog Template; Module; GPIO 255
04:23:29.001 MQT: mimb758/SonoffRFbridge_C0CCE3/TEMPLATE = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
04:23:29.289 MQT: mimb758/SonoffRFbridge_C0CCE3/MODULE = {"Module":{"25":"Sonoff Bridge"}}
04:23:29.574 MQT: mimb758/SonoffRFbridge_C0CCE3/GPIO = {"GPIO0":{"32":"Button1"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

04:20:44.693 CMD: Backlog Rule1; Rule2; Rule3
04:20:45.609 MQT: mimb758/SonoffRFbridge_C0CCE3/RULE = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
04:20:45.893 MQT: mimb758/SonoffRFbridge_C0CCE3/RULE = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

04:15:44.506 CMD: status 0
04:15:44.518 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS = {"Status":{"Module":25,"DeviceName":"MIMB758-TR-SONOFF-RF-BRIDGE","FriendlyName":["MIMB758-TR-SONOFF-RF-BRIDGE"],"Topic":"SonoffRFbridge_C0CCE3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1,"InfoRetain":0,"StateRetain":0}}
04:15:44.613 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://github.com/arendst/Tasmota/releases/download/v9.4.0/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T23:12:44","StartupUTC":"2021-06-21T10:03:00","Sleep":50,"CfgHolder":4617,"BootCount":722,"BCResetTime":"2020-04-07T09:31:03","SaveCount":4380,"SaveAddress":"FB000"}}
04:15:44.701 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"505/699"}}
04:15:44.783 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":1,"SysLog":3,"LogHost":"10.120.11.4","LogPort":514,"SSId":["LRS-WiFi",""],"TelePeriod":900,"Resolution":"558180C0","SetOption":["8400A039","2805FF000100060000965A0A000000000000","0080808D","00086102","00000000"]}}
04:15:44.871 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
04:15:44.958 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS5 = {"StatusNET":{"Hostname":"MIMB758-SONOFF-RFBRIDGE-C0CCE3","IPAddress":"10.120.88.30","Gateway":"10.120.88.1","Subnetmask":"255.255.255.0","DNSServer":"10.120.88.1","Mac":"2C:F4:32:C0:CC:E3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
04:15:45.002 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS6 = {"StatusMQT":{"MqttHost":"10.120.11.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C0CCE3","MqttUser":"TASMOTA","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":60,"SOCKET_TIMEOUT":100}}
04:15:45.085 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS7 = {"StatusTIM":{"UTC":"2021-06-22T09:15:45","Local":"2021-06-22T04:15:45","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"-05:00","Sunrise":"06:27","Sunset":"17:51"}}
04:15:45.165 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS10 = {"StatusSNS":{"Time":"2021-06-22T04:15:45-05:00"}}
04:15:45.242 MQT: mimb758/SonoffRFbridge_C0CCE3/STATUS11 = {"StatusSTS":{"Time":"2021-06-22T04:15:45-05:00","Uptime":"0T23:12:45","UptimeSec":83565,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"Wifi":{"AP":1,"SSId":"LRS-WiFi","BSSId":"84:16:F9:57:BD:21","Channel":1,"Mode":"11n","RSSI":100,"Signal":-33,"LinkCount":3,"Downtime":"0T00:01:24"}}}

```

- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

10:39:51.901 CMD: weblog 4
10:39:51.903 SRC: WebConsole from 10.120.11.21
10:39:51.906 CMD: Grp 0, Cmnd 'WEBLOG', Idx 1, Len 1, Data '4'
10:39:51.910 MQT: edfxm1/SonoffRFbridge_58CAB7/WEBLOG = {"WebLog":4}
10:39:52.889 CFG: Saved to flash at F9, Count 3531, Bytes 4096
10:39:56.685 RUL: No valid JSON ({"RULE1":{"STATE":"ON","ONCE":"OFF","STOPONERROR":"OFF","LENGTH":1618,"FREE":8,"RULES":"ON CLOCK#TIMER=16 DO RESTART 1 ENDON ON SYSTEM#BOOT DO VAR16 EDFXM1/%TOPIC%/ ENDON ON EVENT#S0 DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ST ": "0 "} ENDON ON EVENT#S1 DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ST ": "1 "} ENDON ON EVENT#ST DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ER ": "T "} ENDON ON EVENT#SL DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ER ": "L "} ENDON ON RFRECEIVED#DATA=56D72E DO EVENT S0=56D72X ENDON ON RFRECEIVED#DATA=56D727 DO EVENT S1=56D72X ENDON ON RFRECEIVED#DATA=56D72B D..."}})
10:39:56.690 RUL: No valid JSON ({"RULE1":{"STATE":"ON","ONCE":"OFF","STOPONERROR":"OFF","LENGTH":1618,"FREE":8,"RULES":"ON CLOCK#TIMER=16 DO RESTART 1 ENDON ON SYSTEM#BOOT DO VAR16 EDFXM1/%TOPIC%/ ENDON ON EVENT#S0 DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ST ": "0 "} ENDON ON EVENT#S1 DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ST ": "1 "} ENDON ON EVENT#ST DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ER ": "T "} ENDON ON EVENT#SL DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ER ": "L "} ENDON ON RFRECEIVED#DATA=56D72E DO EVENT S0=56D72X ENDON ON RFRECEIVED#DATA=56D727 DO EVENT S1=56D72X ENDON ON RFRECEIVED#DATA=56D72B D..."}})
10:39:56.695 RUL: No valid JSON ({"RULE1":{"STATE":"ON","ONCE":"OFF","STOPONERROR":"OFF","LENGTH":1618,"FREE":8,"RULES":"ON CLOCK#TIMER=16 DO RESTART 1 ENDON ON SYSTEM#BOOT DO VAR16 EDFXM1/%TOPIC%/ ENDON ON EVENT#S0 DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ST ": "0 "} ENDON ON EVENT#S1 DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ST ": "1 "} ENDON ON EVENT#ST DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ER ": "T "} ENDON ON EVENT#SL DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ER ": "L "} ENDON ON RFRECEIVED#DATA=56D72E DO EVENT S0=56D72X ENDON ON RFRECEIVED#DATA=56D727 DO EVENT S1=56D72X ENDON ON RFRECEIVED#DATA=56D72B D..."}})
10:39:56.703 RUL: No valid JSON ({"RULE1":{"STATE":"ON","ONCE":"OFF","STOPONERROR":"OFF","LENGTH":1618,"FREE":8,"RULES":"ON CLOCK#TIMER=16 DO RESTART 1 ENDON ON SYSTEM#BOOT DO VAR16 EDFXM1/%TOPIC%/ ENDON ON EVENT#S0 DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ST ": "0 "} ENDON ON EVENT#S1 DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ST ": "1 "} ENDON ON EVENT#ST DO PUBLISH2 %VAR16%%VALUE% { "TI ": "%TIMESTAMP% ", "ER ": "T "} ENDON ON EVENT#SL DO PUBLISH2 %VAR16%%VALUE% { "TI ":  [... cut ...]
- The rule creation+uploading process described above completes without errors in both 9.4.0 and 9.5.0. In both cases the reply payload after rule append may sometimes contain surplus information 
-  Under 9.4.0 the rules display and execute as expected
-  Under 9.5.0 the rules do not display (weblog 2) or display with NO VALID JSON  (weblog 4) and fail to execute
-  it seems as if Tasmota's on-demand rule compression and the incoming rule-append commands are running out of sync

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-06-22:
You are not clear about operational details. Are the messages appearing when inputting that long rule set, or later? You did not post the commands you used.

Doing a simple test with adding long rules (including JSON data, random numbers to lower compressibility) to a set gave no such problem, just `RUL: Not enough space` after getting above 1K, and the Rule command acted normally afterwards.

arendst, 2021-06-22:
What you see is the result of a rule trigger from the event consisting of the result from the ``rule1`` command. This result is incomplete due too buffer constraints. It is chopped and extended with three dots.

In a "normal" situation where there is no JSON in the rule commands this is perfectly valid and results in valid JSON. In your case the three dots are inserted inside a JSON message resulting an an invalid JSON response.

As a workaround disable rules when you update/display/execute the ``rule`` command. This makes sure the rule is not used as an event.

As a solution you might want to comment out the below lines in file ``xdrv_10_rules.ino`` starting from lines 2142:
```
//      rule = rule.substring(0, MAX_RULE_SIZE);
//      rule += F("...");
```
Starting with v9.5.0 Tasmota supports responses to at least 4k so the need for the three dots is not really there anymore and it will result in valid JSON as an event to the rules manager.

Pls try the change and let me know if it solved your issue.

joeH68, 2021-06-22:
Thanks @sfromis and @arendst  for the prompt reply.

@sfromis: Below is the console log (weblog 4) from the rule creation process. It shows the rule commands and their replies so I hope it helps in recreating the problem, otherwise please let me know and I will send you the commmands alone.

@arendst: do you mean it is just a display issue and the rules are loaded ok?  I'll do additional tests to confirm the rules are not executing at all. 

=============

15:35:19.475 HTP: Command
15:35:19.481 SRC: WebCommand from 10.120.11.14
15:35:19.484 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 36, Data 'ON clock#Timer=16 DO restart 1 ENDON'
15:35:19.488 RUL: Stored uncompressed, would compress from 36 to 31 (-14%)
15:35:19.492 RSL: RULE = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":36,"Free":475,"Rules":"ON clock#Timer=16 DO restart 1 ENDON"}}
15:35:19.933 CFG: Saved to flash at F4, Count 3675, Bytes 4096
15:35:20.560 HTP: Command
15:35:20.566 SRC: WebCommand from 10.120.11.14
15:35:20.569 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 46, Data '+ON System#Boot DO Var16 edfxm1/%topic%/ ENDON'
15:35:20.574 RUL: Stored uncompressed, would compress from 82 to 68 (-17%)
15:35:20.577 RSL: RULE = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":82,"Free":429,"Rules":"ON clock#Timer=16 DO restart 1 ENDON ON System#Boot DO Var16 edfxm1/%topic%/ ENDON"}}
15:35:20.921 CFG: Saved to flash at FB, Count 3676, Bytes 4096
15:35:21.643 HTP: Command
15:35:21.649 SRC: WebCommand from 10.120.11.14
15:35:21.652 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 75, Data '+ON event#s0 DO publish2 %var16%%value% {"ti":"%timestamp%","st":"0"} ENDON'
15:35:21.660 RUL: Stored uncompressed, would compress from 157 to 133 (-15%)
15:35:21.664 RSL: RULE = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":157,"Free":354,"Rules":"ON clock#Timer=16 DO restart 1 ENDON ON System#Boot DO Var16 edfxm1/%topic%/ ENDON ON event#s0 DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"st\":\"0\"} ENDON"}}
15:35:21.968 CFG: Saved to flash at FA, Count 3677, Bytes 4096
15:35:22.765 HTP: Command
15:35:22.771 SRC: WebCommand from 10.120.11.14
15:35:22.774 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 75, Data '+ON event#s1 DO publish2 %var16%%value% {"ti":"%timestamp%","st":"1"} ENDON'
15:35:22.782 RUL: Stored uncompressed, would compress from 232 to 142 (-39%)
15:35:22.786 RSL: RULE = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":232,"Free":279,"Rules":"ON clock#Timer=16 DO restart 1 ENDON ON System#Boot DO Var16 edfxm1/%topic%/ ENDON ON event#s0 DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"st\":\"0\"} ENDON ON event#s1 DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"st\":\"1\"} ENDON"}}
15:35:22.926 CFG: Saved to flash at F9, Count 3678, Bytes 4096
15:35:24.891 HTP: Command
15:35:24.897 SRC: WebCommand from 10.120.11.14
15:35:24.900 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 75, Data '+ON event#sT DO publish2 %var16%%value% {"ti":"%timestamp%","er":"T"} ENDON'
15:35:24.910 RUL: Stored uncompressed, would compress from 307 to 155 (-50%)
15:35:24.914 RSL: RULE = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":307,"Free":204,"Rules":"ON clock#Timer=16 DO restart 1 ENDON ON System#Boot DO Var16 edfxm1/%topic%/ ENDON ON event#s0 DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"st\":\"0\"} ENDON ON event#s1 DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"st\":\"1\"} ENDON ON event#sT DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"er\":\"T\"} ENDON"}}
15:35:25.956 CFG: Saved to flash at F8, Count 3679, Bytes 4096
15:35:26.018 HTP: Command
15:35:26.025 SRC: WebCommand from 10.120.11.14
15:35:26.028 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 75, Data '+ON event#sL DO publish2 %var16%%value% {"ti":"%timestamp%","er":"L"} ENDON'
15:35:26.037 RUL: Stored uncompressed, would compress from 382 to 163 (-57%)
15:35:26.042 RSL: RULE = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":382,"Free":129,"Rules":"ON clock#Timer=16 DO restart 1 ENDON ON System#Boot DO Var16 edfxm1/%topic%/ ENDON ON event#s0 DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"st\":\"0\"} ENDON ON event#s1 DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"st\":\"1\"} ENDON ON event#sT DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"er\":\"T\"} ENDON ON event#sL DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"er\":\"L\"} ENDON"}}
15:35:26.933 CFG: Saved to flash at F7, Count 3680, Bytes 4096
15:35:27.142 HTP: Command
15:35:27.148 SRC: WebCommand from 10.120.11.14
15:35:27.151 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 51, Data '+ON RfReceived#Data=56D72E DO event s0=56D72x ENDON'
15:35:27.166 RUL: Stored uncompressed, would compress from 433 to 197 (-55%)
15:35:27.170 RSL: RULE = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":433,"Free":78,"Rules":"ON clock#Timer=16 DO restart 1 ENDON ON System#Boot DO Var16 edfxm1/%topic%/ ENDON ON event#s0 DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"st\":\"0\"} ENDON ON event#s1 DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"st\":\"1\"} ENDON ON event#sT DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"er\":\"T\"} ENDON ON event#sL DO publish2 %var16%%value% {\"ti\":\"%timestamp%\",\"er\":\"L\"} ENDON ON RfReceived#Data=56D72E DO event s0=56D72x ENDON"}}
15:35:27.961 CFG: Saved to flash at F6, Count 3681, Bytes 4096
15:35:33.089 WIF: Checking connection...
15:35:33.244 HTP: Command
15:35:33.250 SRC: WebCommand from 10.120.11.14
[... cut ...] e":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer13":{"Enable":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer14":{"Enable":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer15":{"Enable":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Action":0},"Timer16":{"Enable":1,"Mode":0,"Time":"04:55","Window":0,"Days":"1111111","Repeat":1,"Action":3}}
15:36:53.708 RSL: WEBLOG = {"WebLog":2}_

joeH68, 2021-06-22:
`EDIT2: It is indeed caused by the removal of the escape character done on line 515. By commenting this out it solves your issue (but re-introduces another issue). For a quick test try this:`

Thanks!  compiling Tasmota by myself is not an option by now unfortunately. I can wait :D

arendst, 2021-06-22:
In half an hour you'll find a compiled binary in the development department at http://ota.tasmota.com/tasmota/

closed 2021-06-23 by ascillato2

joeH68, 2021-06-23:
I can confirm the issue is solved in 9.5.0.1. The extended Rule display (now without truncation) makes it possible to verify the integrity of the rule as received and stored in Tasmota. The whole rule uploading process is now more robust. Thank you!! 

ps. one small glitch is still there: the http response payload for the Rulex commands AFTER the ruleset begins compressing comes preceded by surplus data from previous Rulex commands. As far as I can tell this has no impact in the process and can be easily circumvented by skiping any data before the expected payload  =>   { "Rule...