arendst/Tasmota #12451: Telegram command tmsend stopped working with no change in Tasmota build or Tasmota settings

ajithvasudevan, 2021-06-23:
### PROBLEM DESCRIPTION
As mentioned in the title, Tasmota's Telegram command `tmsend` stopped working with no change in either the Tasmota build or Tasmota settings

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used : Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 9.3.1.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: PlatformIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
07:39:02.890 CMD: Backlog Template; Module; GPIO 255
07:39:02.893 SRC: WebConsole from 172.20.1.2
07:39:02.896 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
07:39:02.925 SRC: Backlog
07:39:02.928 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
07:39:02.935 RSL: stat/tasmota_74E359/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1856,1888,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:39:03.163 SRC: Backlog
07:39:03.166 CMD: Group 0, Index 1, Command "MODULE", Data ""
07:39:03.171 RSL: stat/tasmota_74E359/RESULT = {"Module":{"18":"Generic"}}
07:39:03.414 SRC: Backlog
07:39:03.417 CMD: Group 0, Index 1, Command "GPIO", Data "255"
07:39:03.425 RSL: stat/tasmota_74E359/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"1856":"SR04 Tri/TX"},"GPIO5":{"1888":"SR04 Ech/RX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
07:39:19.761 WIF: Checking connection...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
07:40:07.680 CMD: Status 0
07:40:07.683 SRC: WebConsole from 172.20.1.2
07:40:07.686 CMD: Group 0, Index 1, Command "STATUS", Data "0"
07:40:07.691 RSL: stat/tasmota_74E359/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_74E359","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
07:40:07.696 RSL: stat/tasmota_74E359/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:13:31","StartupUTC":"2021-06-23T06:26:36","Sleep":50,"CfgHolder":4620,"BootCount":30,"BCResetTime":"2021-06-23T05:41:48","SaveCount":41,"SaveAddress":"FB000"}}
07:40:07.704 RSL: stat/tasmota_74E359/STATUS2 = {"StatusFWR":{"Version":"9.3.1.2(tasmota)","BuildDateTime":"2021-06-23T11:55:37","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"404/699"}}
07:40:07.709 RSL: stat/tasmota_74E359/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Andromeda",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
07:40:07.718 RSL: stat/tasmota_74E359/STATUS4 = {"StatusMEM":{"ProgramSize":647,"Free":356,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","020000CF","050013C0","C000F981","0000C004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,40,45","Sensors":"1,2,3,4,5,6,22,45"}}
07:40:07.727 RSL: stat/tasmota_74E359/STATUS5 = {"StatusNET":{"Hostname":"tasmota_74E359-0857","IPAddress":"172.20.1.222","Gateway":"172.20.1.1","Subnetmask":"255.255.255.0","DNSServer":"172.20.1.1","Mac":"8C:AA:B5:74:E3:59","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:40:07.733 RSL: stat/tasmota_74E359/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_74E359","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:40:07.739 RSL: stat/tasmota_74E359/STATUS7 = {"StatusTIM":{"UTC":"2021-06-23T06:40:07","Local":"2021-06-23T07:40:07","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:47","Sunset":"20:57"}}
07:40:07.745 RSL: stat/tasmota_74E359/STATUS10 = {"StatusSNS":{"Time":"2021-06-23T07:40:07"}}
07:40:07.751 RSL: stat/tasmota_74E359/STATUS11 = {"StatusSTS":{"Time":"2021-06-23T07:40:07","Uptime":"0T00:13:31","UptimeSec":811,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Andromeda","BSSId":"00:11:32:6C:33:E2","Channel":8,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
09:04:13.940 MQT: stat/BELL/RESULT = {"WebLog":4}
09:04:13.025 CFG: Saved to flash at 3F9, Count 186, Bytes 4096
09:04:19.189 WIF: Checking connection...
09:04:25.695 CMD: tmsend test
09:04:25.696 SRC: WebConsole from 172.20.1.2
09:04:25.698 CMD: Group 0, Index 1, Command "TMSEND", Data "test"
09:04:25.700 TGM: sendMessage
09:04:27.098 MQT: stat/BELL/RESULT = {"TmSend":"Failed"}
09:04:40.636 WIF: Checking connection...

```

### TO REPRODUCE
1. set Telegram Bot Token in console with `tmtoken <token>`
2. set Telegram Chat Id in console with `tmchatid <tmchatid>`
3. set `tmstate 1`
4. Send a message with `tmsend test`

### EXPECTED BEHAVIOUR
The test message should be sent to the bot and received in the TG mobile app

### SCREENSHOTS
NA

### ADDITIONAL CONTEXT
This issue started around 20-21 June without any changes in either the Tasmota build nor the settings.

ajithvasudevan, 2021-06-23:
Tried printing the response to the sendMessage Telegram Bot API in the telegram driver, and  this is what I got:

```lua
06:42:18.831 TGM: response = <!DOCTYPE html>
<html class="">
  <head>
    <meta charset="utf-8">
    <title>sendMessage</title>
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <meta property="description" content="">
    <meta property="og:title" content="sendMessage">
    <meta property="og:image" content="">
    <meta property="og:description" content="">
    <link rel="shortcut icon" href="/favicon.ico?4" type="image/x-icon" />

    <link href="/css/bootstrap.min.css?3" rel="stylesheet">
    
    <link href="/css/telegram.css?212" rel="stylesheet" media="screen">
    <style>
    </style>
  </head>
  <body class="preload">
    <div class="dev_page_wrap">
      <div class
06:42:18.841 RSL: stat/tasmota_74E359/RESULT = {"TmSend":"Failed"}
```

I suspect the fingerprint (USE_TELEGRAM_FINGERPRINT) has changed.

So I compared the default fingerprint used in Tasmota: 
```
  #define USE_TELEGRAM_FINGERPRINT "\xB2\x72\x47\xA6\x69\x8C\x3C\x69\xF9\x58\x6C\xF3\x60\x02\xFB\x83\xFA\x8B\x1F\x23" // Telegram api.telegram.org TLS public key fingerprint
```
with the one in the certificate of api.telegram.org:  `F2:AD:29:9C:34:48:DD:8D:F4:CF:52:32:F6:57:33:68:2E:81:C1:90`

I'm not sure this is the right comparison but they don't seem to match.

So I added 
`
#define USE_TELEGRAM_FINGERPRINT "\xF2\xAD\x29\x9C\x34\x48\xDD\x8D\xF4\xCF\x52\x32\xF6\x57\x33\x68\x2E\x81\xC1\x90" 
`
to my user_config_override.h , re-compiled and tested.

However, it still doesn't work.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-06-23:
It seems Telegram is not returning JSON but a whole HTML webpage as you noticed too.

I'll need to dive in recent BOT changes from telegram...

ajithvasudevan, 2021-06-23:
> It seems Telegram is not returning JSON but a whole HTML webpage as you noticed too.
> 
> I'll need to dive in recent BOT changes from telegram...

I checked here: https://core.telegram.org/bots/api-changelog 
Nothing seems to have changed after April 26

Also, that HTML response (I printed the whole thing with incremental offsets, and concatenated!) looks like this:

[image]

What do you make of this?

arendst, 2021-06-23:
Interresting. I have to go for now. Will have a look tomorrow.

ajithvasudevan, 2021-06-24:
It works now, thank you!

And to think I almost got it yesterday ... I just missed the '\r'. I added the "Host:  ..." line and terminated it with only a "\n" instead of "\r\n"   :)

ajithvasudevan, 2021-06-24:
Closing issue as it is fixed.

closed 2021-06-24 by ajithvasudevan