arendst/Tasmota #12461: Single color led controller won't turn off after update

timmipetit, 2021-06-23:
### PROBLEM DESCRIPTION
I recently updated all my devices from 8.1 to 9.5 (using using 8.5.1 and 9.1 as intermediate updates). Since this update my single color led controller won't turn off. When I turn off the device in the tasmota web interface, the leds stay on with low brightness. This was not the case when using the 8.1 release

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [MagicHome Single Color 5-28V LED Controller](https://templates.blakadder.com/magichome_single_color.html)
- [x] Tasmota binary firmware version number used: 9.5.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: not 100% sure, since it's been a while since I flashed, but I think it was done using tasmotizer. Updates were all done OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
14:45:38.408 MQT: stat/ledstripschoorsteenmantel/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:45:38.620 MQT: stat/ledstripschoorsteenmantel/RESULT = {"Module":{"18":"Generic"}}
14:45:38.875 MQT: stat/ledstripschoorsteenmantel/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"416":"PWM1"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`: no rules used
- [x] Provide the output of this command: `Status 0`:
```lua
14:47:05.846 CMD: status 0
14:47:05.854 MQT: stat/ledstripschoorsteenmantel/STATUS = {"Status":{"Module":18,"DeviceName":"LedstripSchoorsteenmantel","FriendlyName":["Ledstrip Schoorsteenmantel"],"Topic":"ledstripschoorsteenmantel","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:47:05.860 MQT: stat/ledstripschoorsteenmantel/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:38","StartupUTC":"2021-06-23T13:36:27","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2021-06-12T14:02:34","SaveCount":555,"SaveAddress":"F8000"}}
14:47:05.866 MQT: stat/ledstripschoorsteenmantel/STATUS2 = {"StatusFWR":{"Version":"9.5.0.1(tasmota)","BuildDateTime":"2021-06-22T18:15:51","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"384/699"}}
14:47:05.871 MQT: stat/ledstripschoorsteenmantel/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["JoseNet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
14:47:05.883 MQT: stat/ledstripschoorsteenmantel/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
14:47:05.890 MQT: stat/ledstripschoorsteenmantel/STATUS5 = {"StatusNET":{"Hostname":"ledstripschoorsteenmantel-5860","IPAddress":"192.168.0.180","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"C4:4F:33:D2:B6:E4","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:47:05.896 MQT: stat/ledstripschoorsteenmantel/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D2B6E4","MqttUser":"tijmen","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:47:05.902 MQT: stat/ledstripschoorsteenmantel/STATUS7 = {"StatusTIM":{"UTC":"2021-06-23T13:47:05","Local":"2021-06-23T14:47:05","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:47","Sunset":"20:57"}}
14:47:05.908 MQT: stat/ledstripschoorsteenmantel/STATUS10 = {"StatusSNS":{"Time":"2021-06-23T14:47:05"}}
14:47:05.914 MQT: stat/ledstripschoorsteenmantel/STATUS11 = {"StatusSTS":{"Time":"2021-06-23T14:47:05","Uptime":"0T00:10:38","UptimeSec":638,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JoseNet","BSSId":"F4:92:BF:AA:B3:B9","Channel":11,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
 14:47:54.481 HTP: Main Menu
14:48:00.498 SRC: WebGui from 192.168.0.232
14:48:00.507 MQT: tele/ledstripschoorsteenmantel/STATE = {"Time":"2021-06-23T14:48:00","Uptime":"0T00:11:33","UptimeSec":693,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JoseNet","BSSId":"F4:92:BF:AA:B3:B9","Channel":11,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}
14:48:00.513 MQT: stat/ledstripschoorsteenmantel/RESULT = {"POWER":"ON"}
14:48:00.517 MQT: stat/ledstripschoorsteenmantel/POWER = ON
14:48:01.059 CFG: Saved to flash at F6, Count 557, Bytes 4096
14:48:08.519 SRC: WebGui from 192.168.0.232
14:48:08.529 MQT: tele/ledstripschoorsteenmantel/STATE = {"Time":"2021-06-23T14:48:08","Uptime":"0T00:11:41","UptimeSec":701,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":61,"MqttCount":1,"POWER":"OFF","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"JoseNet","BSSId":"F4:92:BF:AA:B3:B9","Channel":11,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}
14:48:08.533 MQT: stat/ledstripschoorsteenmantel/RESULT = {"POWER":"OFF"}
14:48:08.536 MQT: stat/ledstripschoorsteenmantel/POWER = OFF
14:48:09.104 CFG: Saved to flash at F5, Count 558, Bytes 4096
14:48:11.265 WIF: Checking connection...
```

### TO REPRODUCE
I'm not sure if this is caused by upgrading, or if a clean 9.5 release would have this issue as well

### EXPECTED BEHAVIOUR
Leds should switch off when status is 'off' in tasmota

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-06-29:
Read the intermediate Releasenotes and check command ``DimmerRange``

timmipetit, 2021-07-07:
`DimmerRange` is set to 0,100; so that doesn't seem to be the issue. Also tried different PWM frequencies, but didn't matter

closed 2021-07-23 by github-actions[bot]