arendst/Tasmota #12479: Only the first <M>,=d ... satnement works with counter based meters

c-MM, 2021-06-25:
### PROBLEM DESCRIPTION
When trying to read two simple energy meters with sensor53 and script, them M,=d ... line only works for the first meter.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): nodemcu
- [x] Tasmota binary firmware version number used: 9.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:51:57.223 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
21:51:57.237 SRC: Backlog
21:51:57.239 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
21:51:57.247 MQT: stat/tasmota_5FAE1A/RESULT = {"NAME":"Z2 counter","GPIO":[1,1,320,1,1,1,1,1,1,1,1,1,320,1],"FLAG":0,"BASE":18}
21:51:57.464 SRC: Backlog
21:51:57.466 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
21:51:57.474 MQT: stat/tasmota_5FAE1A/RESULT = {"Module":{"0":"Z2 counter"}}
21:51:57.715 SRC: Backlog
21:51:57.717 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:53:39.538 CMD: Status 0
21:53:39.540 SRC: WebConsole from 10.23.17.11
21:53:39.543 CMD: Grp 0, Cmnd 'STATUS', Idx 1, Len 1, Data '0'
21:53:39.549 MQT: stat/tasmota_5FAE1A/STATUS = {"Status":{"Module":0,"DeviceName":"Counter test","FriendlyName":["oben"],"Topic":"tasmota_5FAE1A","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:53:39.557 MQT: stat/tasmota_5FAE1A/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:45:39","StartupUTC":"2021-06-25T19:08:00","Sleep":50,"CfgHolder":4617,"BootCount":43,"BCResetTime":"2021-06-24T20:15:57","SaveCount":152,"SaveAddress":"FD000"}}
21:53:39.563 MQT: stat/tasmota_5FAE1A/STATUS2 = {"StatusFWR":{"Version":"9.5.0(Z2node)","BuildDateTime":"2021-06-24T21:11:36","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"331/699"}}
21:53:39.568 MQT: stat/tasmota_5FAE1A/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ESP",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
21:53:39.580 MQT: stat/tasmota_5FAE1A/STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":1428,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87C00497","04088001","00050047","000000C0","C0000219","00004000","00200000","00000020"],"Drivers":"1,2,3,4,8,9,10,12,16,24,28,35,50","Sensors":"1,2,3,5,13,15,53,54"}}
21:53:39.589 MQT: stat/tasmota_5FAE1A/STATUS5 = {"StatusNET":{"Hostname":"tasmota_5FAE1A-3610","IPAddress":"10.23.17.215","Gateway":"10.23.17.1","Subnetmask":"255.255.255.0","DNSServer":"10.23.17.1","Mac":"EC:FA:BC:5F:AE:1A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:53:39.594 MQT: stat/tasmota_5FAE1A/STATUS6 = {"StatusMQT":{"MqttHost":"10.23.17.173","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5FAE1A","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:53:39.601 MQT: stat/tasmota_5FAE1A/STATUS7 = {"StatusTIM":{"UTC":"2021-06-25T19:53:39","Local":"2021-06-25T21:53:39","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"05:48","Sunset":"21:57"}}
21:53:39.608 MQT: stat/tasmota_5FAE1A/STATUS10 = {"StatusSNS":{"Time":"2021-06-25T21:53:39","ENERGY1":{"Total":2351.27,"Power":2217.5},"ENERGY2":{"Total":4244.51,"Power":0.0,"Test":4244.5}}}
21:53:39.615 MQT: stat/tasmota_5FAE1A/STATUS11 = {"StatusSTS":{"Time":"2021-06-25T21:53:39","Uptime":"0T00:45:39","UptimeSec":2739,"Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"ESP","BSSId":"F6:EC:DA:1A:92:C5","Channel":11,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
21:54:11.406 MQT: tele/tasmota_5FAE1A/STATE = {"Time":"2021-06-25T21:54:11","Uptime":"0T00:46:11","UptimeSec":2771,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"ESP","BSSId":"F6:EC:DA:1A:92:C5","Channel":11,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:54:11.415 MQT: tele/tasmota_5FAE1A/SENSOR = {"Time":"2021-06-25T21:54:11","ENERGY1":{"Total":2351.27,"Power":2217.5},"ENERGY2":{"Total":4244.51,"Power":0.0,"Test":4244.5}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I use the following script to read the meters. Only the first line with 1,=d 1 10... works, the second one 2,=d 3 10... doesn't.
I'm feeding both gpios from a pulse generator for testing. So it's not a hardware issue.
```lua
>D

>B
sensor53 r

>M 2
+1,5,c,1,-10,ENERGY1
1,1-0:1.8.0*255(@75,Zaehler,kWh,Total,2
1,=d 1 10 @1,Leistung,W,Power,1
+2,14,c,1,-10,ENERGY2
2,1-0:1.8.0*255(@100,Zaehler,kWh,Total,2
2,=d 3 10 @1,Leistung,W,Power,1
2,=m 3*#1 @1,Test,W,Test,1
#
```
### EXPECTED BEHAVIOUR
I expect the mqtt message from the meter ENERGY2 to have a non zero power value as has the meter ENERGY1

### SCREENSHOTS
```
21:58:11.402 MQT: tele/tasmota_5FAE1A/SENSOR = {"Time":"2021-06-25T21:58:11","ENERGY1":{"Total":2351.39,"Power":2713.4},"ENERGY2":{"Total":4244.60,"Power":0.0,"Test":4244.6}}
21:59:11.397 MQT: tele/tasmota_5FAE1A/SENSOR = {"Time":"2021-06-25T21:59:11","ENERGY1":{"Total":2351.40,"Power":1346.0},"ENERGY2":{"Total":4244.61,"Power":0.0,"Test":4244.6}}
22:00:11.415 MQT: tele/tasmota_5FAE1A/SENSOR = {"Time":"2021-06-25T22:00:11","ENERGY1":{"Total":2351.43,"Power":2212.7},"ENERGY2":{"Total":4244.63,"Power":0.0,"Test":4244.6}}
22:01:11.415 MQT: tele/tasmota_5FAE1A/SENSOR = {"Time":"2021-06-25T22:01:11","ENERGY1":{"Total":2351.47,"Power":2026.3},"ENERGY2":{"Total":4244.66,"Power":0.0,"Test":4244.7}}
```
Both "total" counters increase and the first one shows the correct calculated power, the second is always 0.0.

### ADDITIONAL CONTEXT
**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-07-02:
2,=d 3 10 @1,Leistung,W,Power,1

you refer to line 3, but there are only 2 decode lines. (=d is no decode line)

should be
2,=d 2 10 @1,Leistung,W,Power,1

ascillato2, 2021-07-09:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-07-09 by ascillato2

c-MM, 2021-07-11:
> 2,=d 3 10 @1,Leistung,W,Power,1
> 
> you refer to line 3, but there are only 2 decode lines. (=d is no decode line)
> 
> should be
> 2,=d 2 10 @1,Leistung,W,Power,1

Well it doesn't work with 2,=d 2 10 @1,Leistung,W,Power,1 either.
Actually I tried that before since I wasn't sure how the lines are counted.
I have set up the test node again to verify that.

gemu2015, 2021-07-12:
you are right, sorry for the confusion. all lines except h lines count for index.
its several years ago that i coded this. i am extremely short of time currently so i can't check the actual reason.
but please try forcing the d line with the following additional  script snippet.
(needs #define USE_SML_SCRIPT_CMD)

>S
smld(4)

this forces line 4 (the second d line) to recompute

gemu2015, 2021-07-12:
i think i know what the problem is with counters.
the d line is only computed on a counter impulse.
so you must use smld(x) for both meters to update the d lines every second

please try:

\>S
smld(1)
smld(2)

c-MM, 2021-07-12:
Nope, that doesn't make the second =d line calculated and leads to the first =d line to jump between 0.0 and the calculated value.

gemu2015, 2021-07-17:
if found the bug. please try the source. i used this script for testing.

\>D

\>B
sensor53 r

\>S
smld(1)
smld(2)

\>M 2
+1,5,c,1,-10,ENERGY1
+2,14,c,1,-10,ENERGY2
1,1-0:1.8.0*255(@75,Zaehler,kWh,Total,2
1,=d 1 10 @1,Leistung,W,Power,1
2,1-0:1.8.0*255(@100,Zaehler,kWh,Total,2
2,=d 3 10 @1,Leistung,W,Power,1
2,=m 3*#1 @1,Test,W,Test,1
#

[xsns_53_sml.ino.zip](https://github.com/arendst/Tasmota/files/6834262/xsns_53_sml.ino.zip)

c-MM, 2021-07-17:
Yes, now it works. Thanks a lot.

Just for completeness as github ate a few chars in your above post, this is the script that works for me:
```php
>D

>B
=>sensor53 r

>S
smld(1)
smld(2)

>M 2
+1,5,c,1,-10,ENERGY1
+2,14,c,1,-10,ENERGY2
1,1-0:1.8.0*255(@75,Zaehler,kWh,Total,2
1,=d 1 10 @1,Leistung,W,Power,1
2,1-0:1.8.0*255(@100,Zaehler,kWh,Total,2
2,=d 3 10 @1,Leistung,W,Power,1
2,=m 3*#1 @1,Test,W,Test,1
#
```

Some annotations for documentation in case anyone searches this.

The lines:
```php
>S
smld(1)
smld(2)
```
are possibly not needed with counter based meters if the =d time is lower than the typical time for one round.
If you add them, you will get the power for 10s in the example above and then will get zero power until the next impulse is counted. If you leave them out, you get the calculated power for the last two pulses (if the time between the pulses is larger than the =d parameter) over the time between the pulses.

The graphs in homeassistant would be nicer if tasmota would send "unavailable" or the empty string "" for values, that are not yet calculateable instead of zero.

Anyways, thanks a lot!

PR #12689 (2021-07-20): fix =d mode
## Description:

**Related issue (if applicable):** fixes #12479

fixes SML =d mode for more then 1 entry

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.1.0.7.1
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributo [... cut]