arendst/Tasmota #12487: Configure tamota-lite with TLS via override files

soosp, 2021-06-26:
### PROBLEM DESCRIPTION
If you set FIRMWARE_LITE flag you are not able to enable USE_MQTT_TLS option but only editing tasmota_configurations.h

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic but any other
- [x] Tasmota binary firmware version number used: 9.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: web
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:37:52.801 MQT: stat/tasmota_0EF3D3/RESULT = {"NAME":"Sonoff Basic","GPIO":[32,1,1,1,1,0,0,0,224,320,1,0,0,0],"FLAG":0,"BASE":1}
21:37:52.967 MQT: stat/tasmota_0EF3D3/RESULT = {"Module":{"1":"Sonoff Basic"}}
21:37:53.226 MQT: stat/tasmota_0EF3D3/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:39:15.481 MQT: stat/tasmota_0EF3D3/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_0EF3D3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:39:15.497 MQT: stat/tasmota_0EF3D3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"5O1","GroupTopic":"tasmotas","OtaUrl":"http://www.l10c.hu/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:15:11","StartupUTC":"2021-06-26T19:24:04","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2021-01-22T21:58:12","SaveCount":52,"SaveAddress":"F8000"}}
21:39:15.511 MQT: stat/tasmota_0EF3D3/STATUS2 = {"StatusFWR":{"Version":"9.5.0(lite)","BuildDateTime":"2021-06-26T19:47:12","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"342/699"}}
21:39:15.525 MQT: stat/tasmota_0EF3D3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["L10C",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00206000","00000000"]}}
21:39:15.544 MQT: stat/tasmota_0EF3D3/STATUS4 = {"StatusMEM":{"ProgramSize":530,"Free":472,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F0007C3","04108001","00000006","010001C0","00000000","00004004","00000000","00000020"],"Drivers":"1,2,3,4,9,10,16,20,21,37","Sensors":"3"}}
21:39:15.559 MQT: stat/tasmota_0EF3D3/STATUS5 = {"StatusNET":{"Hostname":"tasmota_0EF3D3-5075","IPAddress":"192.168.44.246","Gateway":"192.168.44.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.44.11","Mac":"BC:DD:C2:0E:F3:D3","Webserver":2,"WifiConfig":4,"WifiPower":4.8}}
21:39:15.572 MQT: stat/tasmota_0EF3D3/STATUS6 = {"StatusMQT":{"MqttHost":"apu3.l10c.hu","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0EF3D3","MqttUser":"sp","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:39:15.586 MQT: stat/tasmota_0EF3D3/STATUS7 = {"StatusTIM":{"UTC":"2021-06-26T19:39:15","Local":"2021-06-26T21:39:15","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"04:52","Sunset":"20:49"}}
21:39:15.599 MQT: stat/tasmota_0EF3D3/STATUS10 = {"StatusSNS":{"Time":"2021-06-26T21:39:15"}}
21:39:15.616 MQT: stat/tasmota_0EF3D3/STATUS11 = {"StatusSTS":{"Time":"2021-06-26T21:39:15","Uptime":"0T00:15:11","UptimeSec":911,"Vcc":3.516,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"L10C","BSSId":"E4:8D:8C:49:3C:FA","Channel":1,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Try to compile with options `-DFIRMWARE_LITE -DUSE_MQTT_TLS`

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
tasmota-lite.bin (aka. tasmota without sensors) with tls support

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
If you comment out `#undef USE_MQTT_TLS` in the tasmota-lite.bin section of tasmota_configurations.h it solves the issue. In this way users don't have to edit tasmota_configurations.h. This solution does not hurt the precompiled binaries because they compiled without TLS support (without `-DUSE_MQTT_TLS`).

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-06-26:
Hi,

The compilation process for the standard Tasmota just uses the my_user_config.h and user_config_override.h. If you set platformio.ini to compile any other firmware type different than the standard, the compilation process will apply several #define and #undef defined at tasmota_configurations.h file.

So, you should use user_config_override.h undefining everything you don't want and defining what you actually need, and compiling the standard version of Tasmota (as explained in the docs and in old issues).

Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-06-26 by ascillato2

soosp, 2021-06-26:
Yes, but  tasmota_configurations.h processed _after_ the my_user_config.h and user_config_override.h (see include/tasmota_options.h), so you can't override any `#undef` in tasmota_configurations.h from these files.

ascillato, 2021-06-26:
Exactly. As designed.

You need to use the procedure explained in the docs. Change the user_config_override.h file and compile standard Tasmota. Not the Lite version.

soosp, 2021-06-26:
I have read the docs and usually I use it this way. Now I decided to use the lite version because I need a firmware with TLS for a Sonoff Basic. If you enable TLS, you won't able to upload the binary over the firmware minimal. With Tasmota version 9.2 it worked, but the version 9.5 with TLS is too big for Sonoff Basic. Alternatively it may be useful if it would be possible to enable or disable feature sets (e.g. `#undef USE_SENSORS`).

ascillato, 2021-06-26:
Sorry, but that way of compiling is not as the one designed.

The documented and supported compilation procedure is:

0- Download the source code
1- Edit ONLY the **_user_config_override.h_** file
2- Hit compile in Platformio

That will compile the standard Tasmota with the `#UNDEFs` and with the `#DEFINEs` of your own **_user_config_override.h_** file

Changing user_config_override.h + use a predefined set of features like Tasmota-Lite is NOT SUPPORTED.

So, you need to just edit your **_user_config_override.h_** file and add at the bottom of the file (just above the last line of the file: `#endif  // _USER_CONFIG_OVERRIDE_H_` ):

```cpp
// ADD HERE YOUR DESIRED FEATURES TO BE ENABLED -  And also the configurations you want
#define USE_MQTT_TLS

// ADD HERE YOUR DESIRED FEATURES TO BE DISABLED - In this example there are less features than the Lite version, so this will be better for your Sonoff Basic.
#undef USE_ARDUINO_OTA                           // Disable support for Arduino OTA
#undef USE_DOMOTICZ                              // Disable Domoticz
#undef USE_HOME_ASSISTANT                        // Disable Home Assistant
#undef USE_KNX                                   // Disable KNX IP Protocol Support
//#undef USE_WEBSERVER                             // Disable Webserver
#undef USE_ENHANCED_GUI_WIFI_SCAN                // Disable wifi scan output with BSSID (+0k5 code)
#undef USE_WEBSEND_RESPONSE                      // Disable command WebSend response message (+1k code)
//#undef USE_EMULATION                             // Disable Wemo or Hue emulation
//#undef USE_EMULATION_HUE                         // Disable Hue Bridge emulation for Alexa (+14k code, +2k mem common)
//#undef USE_EMULATION_WEMO                        // Disable Belkin WeMo emulation for Alexa (+6k code, +2k mem common)
#undef USE_CUSTOM                                // Disable Custom features
#undef USE_DISCOVERY                             // Disable Discovery services for both MQTT and web server
//#undef USE_TIMERS                                // Disable support for up to 16 timers
//#undef USE_TIMERS_WEB                            // Disable support for timer webpage
//#undef USE_SUNRISE                               // Disable support for Sunrise and sunset tools
//#undef USE_RULES                                 // Disable support for rules
#undef USE_SCRIPT                                  // Add support for script (+17k code)
#undef ROTARY_V1                                 // Disable support for MI Desk Lamp
#undef USE_SONOFF_RF                             // Disable support for Sonoff Rf Bridge (+3k2 code)
  #undef USE_RF_FLASH                            // Disable support for flashing the EFM8BB1 chip on the Sonoff RF Bridge. C2CK must be connected to GPIO4, C2D to GPIO5 on the PCB
#undef USE_SONOFF_SC                             // Disable support for Sonoff Sc (+1k1 code)
#undef USE_TUYA_MCU                              // Disable support for Tuya Serial MCU
#undef USE_ARMTRONIX_DIMMERS                     // Disable support for Armtronix Dimmers (+1k4 code)
#undef USE_PS_16_DZ                              // Disable support for PS-16-DZ Dimmer and Sonoff L1 (+2k code)
#undef USE_SONOFF_IFAN                           // Disable support for Sonoff iFan02 and iFan03 (+2k code)
#undef USE_BUZZER                                // Disable support for a buzzer (+0k6 code)
#undef USE_ARILUX_RF                             // Disable support for Arilux RF remote controller
#undef USE_SHUTTER                               // Disable Shutter support for up to 4 shutter with different motortypes (+6k code)
#undef USE_DEEPSLEEP                             // Disable support for deepsleep (+1k code)
#undef USE_EXS_DIMMER                            // Disable support for EX-Store WiFi Dimmer
#undef USE_HOTPLUG                               // Disable support for HotPlug
#undef USE_DEVICE_GROUPS                         // Disable support for device groups (+3k5 code)
#undef USE_PWM_DIMMER                            // Disable support for MJ-SD01/acenx/NTONPOWER PWM dimmers (+4k5 code)
#undef USE_PWM_DIMMER_REMOTE                     // Disbale support for remote switches to PWM Dimmer
#undef USE_KEELOQ                                // Disable support for Jarolift rollers by Keeloq algorithm (+4k5 code)
#undef USE_SONOFF_D1                             // Disable support for Sonoff D1 Dimmer (+0k7 code)
#undef USE_LIGHT_VIRTUAL_CT                      // Disable support for Virtual White Color Temperature (SO106)
#undef USE_LIGHT                                 // Also disable all Dimmer/Light support
#undef USE_WS2812                                // Disable WS2812 Led string using library NeoPixelBus (+5k code, +1k mem, 232 iram) - Disable by //
#undef USE_MY92X1                                // Disable support for MY92X1 RGBCW led controller as used in Sonoff B1, Ailight and Lohas
#undef USE_SM16716                               // Disable support for SM16716 RGB LED controller (+0k7 code)
#undef USE_SM2135                                // Disable support for SM2135 RGBCW led control as used in Action LSC (+0k6 code)
#undef USE_SONOFF_L1                             // Disable support for Sonoff L1 led control
#undef USE_ELECTRIQ_MOODL                        // Disable support for ElectriQ iQ-wifiMOODL RGBW LED controller
[... cut ...]
#undef USE_RF_SENSOR                             // Disable support for RF sensor receiver (434MHz or 868MHz) (+0k8 code)
#undef USE_HRE                                   // Disable support for Badger HR-E Water Meter (+1k4 code)
#undef USE_A4988_STEPPER                         // Disable support for A4988_Stepper
#undef USE_THERMOSTAT                            // Disable support for Thermostat
#undef DEBUG_THEO                                // Disable debug code
#undef USE_DEBUG_DRIVER           
```

ascillato, 2021-06-26:
If you are struggling compiling, it is recommended to use instead Tasmocompiler that is a Graphical Interface to select which features you want and which one you don't want to be added. The compilation with Tasmocompiler is very straight forward. Please check Tasmota documentation for more help on that.
Thanks.

soosp, 2021-06-26:
Thanks for the suggested solution. It was clear for me before. The disadvantage of this method that you have to check the whole feature set if a new version of Tasmota is released. I prefer to define only the diferences of my stuff to a predefined set. It results clear an easy to read custom configs.