arendst/Tasmota #12494: MQTT retain flag ignored when enabling MI32Option6

hoposh2k, 2021-06-27:
### PROBLEM DESCRIPTION
When using MI32Option6=1 to enable single topic per Mi sensor tele/tasmota_ble/ SensorRetain option is to be ignored.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Lolin32 dev board
- [x] Tasmota binary firmware version number used: 9.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
stat/tasmota_2B7404/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
stat/tasmota_2B7404/RESULT = {"Module":{"1":"ESP32-DevKit"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
set SensorRetain=1
sensor messages are retained
MI32Option6=1
sensor messages are sent with topic tele/tasmota_ble/ but WITHOUT retain flag

### EXPECTED BEHAVIOUR
messages sent with topic tele/tasmota_ble/ should obey SensorRetain

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

closed 2021-06-29 by ascillato2

thehijacker, 2021-11-04:
Can we reopen this? For me this is still not working. Using latest 10.0.0.1 version. MI32Option6 = 1 and SensorRetain = On, but tele/tasmota_ble/ messages are still not retained.

arendst, 2021-11-04:
And another try.....

barbudor, 2021-11-04:
This doesn't looks a a good idea to me to have RETAIN on tele/+/BLE messages as the driver will send multiple ones and only the latest one will be retained.
Beside BLE messages are mostly used to detect present of devices in the house (like detecting who is in the house).
By retaining these messages that makes them unusable for such application

Exemple with only 1 device, there is 2 messages.
```
17:13:43.871 MQT: sonnaz/gateway/tele/BLE = {"Time":"2021-11-04T17:13:43+01:00","BLEDevices":{"total":1,"A4C138A8AC99":{"i":0,"r":-81,"a":"misensor5"}}}
17:13:43.908 MQT: sonnaz/gateway/tele/BLE = {"Time":"2021-11-04T17:13:43+01:00","BLE":{"scans":3241,"adverts":77545,"devices":1,"resets":0}}
```
Only the last line will be retained

thehijacker, 2021-11-04:
> This doesn't looks a a good idea to me to have RETAIN on tele/+/BLE messages as the driver will send multiple ones and only the latest one will be retained. Beside BLE messages are mostly used to [...]

You got it all wrong. MI32Option6 only enabled individual MQTT messeges per device in path tele/tasmota_ble/devicealias_or_mac. This is useful when manually defining sensors in HA. When HA is rebooted all the sensor values are unavailable until ESP32 reads them again, which can be for long period. If last message was retained it would solve this reboot issue. An example of MQTT messages:

```
17:43:06.664 MQT: tele/tasmota_ble/Living = {"Time":"2021-11-04T17:43:06","alias":"Living","mac":"a4c1381d1d4b","Temperature":24.8,"Humidity":55.0,"DewPoint":15.1,"Battery":76,"RSSI":-78}
17:43:07.671 MQT: tele/tasmota_ble/Erker = {"Time":"2021-11-04T17:43:07","alias":"Erker","mac":"a4c1385d23e9","Temperature":24.6,"Humidity":57.4,"DewPoint":15.6,"Battery":84,"RSSI":-82}
17:43:08.623 MQT: tele/tasmota_ble/Kabinet = {"Time":"2021-11-04T17:43:08","alias":"Kabinet","mac":"a4c138ea02f7","Temperature":24.2,"Humidity":56.0,"DewPoint":14.9,"Battery":68,"RSSI":-73}
17:43:09.635 MQT: tele/tasmota_ble/Flora3 = {"Time":"2021-11-04T17:43:09","alias":"Flora3","mac":"c47c8d6cbd09","Temperature":24.7,"Illuminance":0,"Moisture":0,"Fertility":0,"RSSI":-87}

```

hoposh2k, 2021-11-04:
Another example is my usage scenario where data doesn't get stored. I view my temperatures from an mqtt dashboard in android. With the retain flag not working properly everytime I open the app on my phone I have to wait for Tasmota to read the sensors to get new data which takes a long time...

ascillato, 2021-11-04:
Please, update to latest version from the development branch and try again. Theo has fixed it.

thehijacker, 2021-11-04:
Updated just now from binary on URL http://ota.tasmota.com/tasmota32/tasmota32-bluetooth.bin

```
Program Version	10.0.0.1(bluetooth)
Build Date & Time	2021-11-04T07:55:10
```
MQTT messages from my Xiaomi BLE Flora sensors are retained, while the ones from my BLE temperature sensors are not.

[image]

[image]

Binaries not yet updated?

hoposh2k, 2021-11-04:
Just updated now to version 10.0.0.1(bluetooth).
Mi32Option6 disabled
`tele/tasmota_2B7404/SENSOR = {"Time":"2021-11-04T20:12:10","ATC06a37d":{"mac":"a4c13806a37d","Temperature":17.3,"Humidity":58.6,"DewPoint":9.1,"Battery":19,"RSSI":-82}} (retained)`
Mi32Option6 enabled
`tele/tasmota_ble/ATCd0a88d = {"Time":"2021-11-04T20:13:48","mac":"a4c138d0a88d","Temperature":20.1,"Humidity":58.7,"DewPoint":11.8,"Battery":22,"RSSI":-77}`
also
`mosquitto_sub -v -t tele/tasmota_ble/ATCd0a88d` 
does not show anything
It seems it's not fixed

hoposh2k, 2021-11-04:
I've just downloaded binary from here 
https://github.com/arendst/Tasmota-firmware/blob/main/firmware/tasmota32/tasmota32-bluetooth.bin
`tele/tasmota_ble/ATCf6ab7f = {"Time":"2021-11-04T20:23:46","mac":"a4c138f6ab7f","Temperature":19.0,"Humidity":64.1,"DewPoint":12.1,"Battery":41,"RSSI":-81} (retained)`
it seems it's working.

Might be something wrong with the builds at [http://ota.tasmota.com/tasmota32/tasmota32-bluetooth.bin](url)

thehijacker, 2021-11-04:
Now updated again. It seems this latest version from 18:19 includes the fix :)

`tasmota32-bluetooth.bin	http://ota.tasmota.com/tasmota32/tasmota32-bluetooth.bin	1270k	10.0.0.1	20211104 18:19`

All messages are now retained. Thank you!

barbudor, 2021-11-04:
> You got it all wrong. MI32Option6 only enables individual MQTT messeges per device in path tele/tasmota_ble/devicealias_or_mac. This is useful when manually defining sensors in HA. When HA is rebooted all the sensor values are unavailable until ESP32 reads them again, which can be for long period. If last message was retained it would solve this reboot issue. An example of MQTT messages:

Ok. dodn't knew this detail.

However this feature is publishing to a hardcoded topic, properly ignoring the FULL TOPIC. 
Bad !

```
18:48:25.822 MQT: tele/tasmota_ble/misensor5 = {"Time":"2021-11-04T18:48:25+01:00","alias":"misensor5","mac":"a4c138a8ac99","Temperature":18.0,"Humidity":58.6,"DewPoint":9.7,"Battery":64,"RSSI":-79}
18:48:42.793 MQT: sonnaz/gateway/tele/STATE = {"Time":"2021-11-04T18:48:42+01:00","Uptime":"0T19:40:16","UptimeSec":70816,"Heap":55,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"mado-wifi","BSSId":"E8:1B:69:02:D4:AA","Channel":11,"Mode":"11n","RSSI":100,"Signal":-23,"LinkCount":1,"Downtime":"0T00:00:10"}}
18:48:42.843 MQT: sonnaz/gateway/tele/SENSOR = {"Time":"2021-11-04T18:48:42+01:00","ESP32":{"Temperature":45.0},"TempUnit":"C"} (retained)
```

thehijacker, 2021-11-04:
That is the point of this option. To replace full topic with all sensor detalis in one topic with each seperatly. You can always disable option MI32Option6.

Now HA reboot will not show values as unavailable.