arendst/Tasmota #12505: Failed to decode %12X in MQTT_USER and FRIENDLY_NAME

tjhiwa, 2021-06-29:
### PROBLEM DESCRIPTION
Trying to compile, at user_config_override.h, set MQTT_USER and FRIENDLY_NAME as "%12X" but it was not decoded after flashing to ESP8266.

MQTT_USER will be literal %12X, while FRIENDLY_NAME setting page will not load completely.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff CH4 Pro R3, ESP01S, NodeMCU
- [x] Tasmota binary firmware version number used: dev 9.5.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled > Yes
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
#undef  MQTT_USER
#define MQTT_USER              "%12X"
#undef  FRIENDLY_NAME
#define FRIENDLY_NAME      "%12X"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
It is expected %12X will translate to MAC address, like how it works for MQTT_CLIENT_ID and MQTT_TOPIC.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-06-29:
As designed. Your expection is wrong. Nowhere in the doc it is described that %12X can be used as general variable everywhere and beeing expanded to MAC address.

tjhiwa, 2021-06-29:
It worked on MQTT_TOPIC and MQTT_CLIENT_ID, it's a surprise that it was designed to not work on MQTT_USER and fails to load page on FRIENDLY_NAME.

Jason2866, 2021-06-29:
As long it is not in the documentation do not expect. There is nothing to be surprised.
You can close this and open a feature request. Describe the reason(s) why it should be implemented, what benefits it would have. Maybe it will be added.

closed 2021-06-29 by ascillato2