arendst/Tasmota #12513: 9.5.0 - MQTT settings not saving correctly

DrTechdom, 2021-06-29:
### PROBLEM DESCRIPTION
After flashing a device via serial and going straight to 9.5.0 and trying to have the device connect via MQTT it seems as if the settings are not saving correctly
-- From HASS MQTT Logs ---
401: Unauthorized1624966031: Socket error on client <unknown>, disconnecting.
1624966036: New connection from 192.168.30.204 on port 1883.
1624966036: Socket error on client <unknown>, disconnecting.
-- From Tasmota logs --
12:48:43.796 MQT: Attempting connection...
12:48:43.812 MQT: Connect failed to 192.168.030.003:1883, rc 5. Retry in 30 sec

After flashing the device with a known good version 9.3.1, saving the MQTT settings and verifying connectivity I can than successfully upgrade to 9.5.0 with no issues

In the logs after dumping debug 2 over serial I found when MQTT is saved via web config the password is saved as **** instead of the actual password. When I run the command "mqttpassword secret" it connects to the mqtt server with no issues

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: 9.5.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
not configured yet, fresh install

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
23:20:30.788 HTP: Main Menu
23:20:32.457 RSL: STATE = {"Time":"2021-06-29T23:20:32","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iot","BSSId":"8A:2A:A8:91:7E:41","Channel":1,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}
23:20:32.608 HTP: Configuration
23:20:33.457 APP: Boot Count 15
23:20:33.593 CFG: Saved to flash at FA, Count 23, Bytes 4096
23:20:34.506 HTP: Configure MQTT
23:20:46.772 WIF: Checking connection...
23:20:56.688 HTP: Configure MQTT
23:20:56.690 SRC: WebGui from 192.168.20.12
23:20:56.691 CMD: Grp 0, Cmnd 'BACKLOG', Idx 0, Len 126, Data ';MqttHost 192.168.30.3;MqttPort 1883;MqttClient test123;MqttUser hass;MqttPassword2 ;Topic test123;FullTopic %prefix%/%topic%/'
23:20:56.745 HTP: Restart
23:20:56.745 SRC: WebGui from 0.0.0.0
23:20:56.747 SRC: Backlog
23:20:56.748 CMD: Grp 0, Cmnd 'MQTTHOST', Idx 1, Len 12, Data '192.168.30.3'
23:20:56.750 CFG: CR 355/699, Busy 0
23:20:56.753 RSL: RESULT = {"MqttHost":"192.168.30.3"}
23:20:56.762 SRC: Backlog
23:20:56.763 CMD: Grp 0, Cmnd 'MQTTPORT', Idx 1, Len 4, Data '1883'
23:20:56.767 RSL: RESULT = {"MqttPort":1883}
23:20:56.771 SRC: Backlog
23:20:56.772 CMD: Grp 0, Cmnd 'MQTTCLIENT', Idx 1, Len 7, Data 'test123'
23:20:56.779 CFG: CR 353/699, Busy 0
23:20:56.782 RSL: RESULT = {"MqttClient":"test123"}
23:20:56.791 SRC: Backlog
23:20:56.792 CMD: Grp 0, Cmnd 'MQTTUSER', Idx 1, Len 4, Data 'hass'
23:20:56.796 CFG: CR 348/699, Busy 0
23:20:56.799 RSL: RESULT = {"MqttUser":"hass"}
23:20:56.804 SRC: Backlog
23:20:56.805 CMD: Grp 0, Cmnd 'MQTTPASSWORD', Idx 2, Len 0, Data ''
23:20:56.812 RSL: RESULT = {"MqttPassword":"****"}
23:20:56.816 SRC: Backlog
23:20:56.817 CMD: Grp 0, Cmnd 'TOPIC', Idx 1, Len 7, Data 'test123'
23:20:56.825 RSL: LWT = 
23:20:56.826 CFG: CR 343/699, Busy 0
23:20:56.829 RSL: RESULT = {"Topic":"test123"}
23:20:56.834 SRC: Backlog
23:20:56.836 CMD: Grp 0, Cmnd 'FULLTOPIC', Idx 1, Len 17, Data '%prefix%/%topic%/'
23:20:56.844 RSL: RESULT = {"FullTopic":"%prefix%/%topic%/"}
23:20:57.035 MQT: Attempting connection...
23:20:57.679 MQT: Connect failed to 192.168.30.3:1883, rc 5. Retry in 10 sec
23:20:57.871 CFG: Saved to flash at F9, Count 24, Bytes 4096
23:20:58.780 APP: Restarting

```

### TO REPRODUCE
Flash new device with 9.5.0
setup mqtt server info
watch for it to break
to fix it use mqttpassword at command line

### EXPECTED BEHAVIOUR
mqtt will not connect do to bad auth

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-06-30:
Do you use latest HA mqtt broker? If yes, delete the configured mqtt user/pwd and reconfigure. It is a know issue from HA broker.

DrTechdom, 2021-06-30:
I may be (6.0.1)... But that isn't the issue.. I didnt mask out the password string in the logs... its literally sending the masked out password (*) in the html form and saving that if you configure MQTT via http. The temp password i used was 1234 that should be in there username hass.

23:20:56.792 CMD: Grp 0, Cmnd 'MQTTUSER', Idx 1, Len 4, Data 'hass'
23:20:56.799 RSL: RESULT = {"MqttUser":"hass"}
23:20:56.805 CMD: Grp 0, Cmnd 'MQTTPASSWORD', Idx 2, Len 0, Data '' <-----
23:20:56.812 RSL: RESULT = {"MqttPassword":"****"} <-----

ascillato, 2021-06-30:
Tasmota won't give you the password. That is why is masked. It is an intended behaviour to show you that instead of the real password.

So, your issue is another thing like Jason said.

NobleKangaroo, 2021-07-01:
Also having this issue. Working on 9.4.0 but not 9.5.0. Using latest Home Assistant Mosquitto Broker addon (core_mosquitto). It's a bit of a complicated issue, as I am able to log in with MQTT Explorer but not Tasmota using the same credentials. It seems that there's something being handled oddly with numeric-only passwords, as changing the password to include an x after the end (e.g. 1234x) resolves the issue, however putting it back to a numeric string `'1234'` the issue re-appears. Again this doesn't affect MQTT Explorer, and isn't an issue with 9.4.0, so I'm really confused as to what is exactly going on here.

ascillato, 2021-07-01:
Ah. Ok. The issue is that Tasmota accept passwords with a minimum of 5 characters. 4 characters or less are considered too easy to crack, hence they are not accepted.

Just use a longer password and your issue is solved.

NobleKangaroo, 2021-07-02:
> Ah. Ok. The issue is that Tasmota only accepts passwords with a minimum of 5 characters. 4 characters or less are considered too easy to crack, hence they are not accepted.
> 
> Just use a longer password and your issue is solved.

That makes sense. I wonder why it only applies to new flashes, while ignoring OTA upgrades? That just to make sure you don't break anyone? Also, doesn't appear to be mentioned in [the doco](configure-mqtt). A quick mention may be good, or perhaps an error message if the user tries to enter less than 5 characters. Thanks for the quick reply!

ascillato, 2021-07-02:
It should be in the docs. If not, please help us and add it. You can edit any part of the docs by clicking on the pencil and then that will be reviewed and merged. Thanks.

DrTechdom, 2021-07-02:
A JS message would nice greatly appreciated if added to the web interface to catch the save and than prompt the user about the minimum requirements of the MQTT password.

Jason2866, 2021-07-03:
Resources are very low with the ESP82XX. Tasmota has nearly nowhere checks for valid inputs.
By doing this a lot of features would not be possible. I have updated the documentation.
Thx for pointing out that a  important information was missing there.

NobleKangaroo, 2021-07-03:
One suggestion that takes next to nothing is minlength on the input. Let the browser do the work with the minlength suggestion.

<input type="text" id="mqtt_password" minlength="5">

See: https://developer.mozilla.org/en-US/docs/Web/HTML/Attributes/minlength

NobleKangaroo, 2021-07-03:
One suggestion that takes next to nothing is [minlength](https://developer.mozilla.org/en-US/docs/Web/HTML/Attributes/minlength) on the input. Let the browser do the work with the minlength suggestion.

```html
<input type="password" id="mqtt_password" minlength="5">
```

ascillato, 2021-07-03:
Thanks, I will add that.

closed 2021-07-04 by arendst