arendst/Tasmota #12514: GitPod-Tasmota-Serial to TCP Bridge, does not work

adharyanto, 2021-06-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Need to activate the "Serial to TCP Bridge" on Tasmota, by compiling the latest Tasmota v9.5.0.1, online compiling using GitPod, but setting the TCPBaudRate does not work.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [v] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [v] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [v] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [v] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [v] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [v] Device used (e.g., Sonoff Basic): WeMos D1 Mini and NodeMCU v3_____
- [v] Tasmota binary firmware version number used: 9.5.0.1_____
  - [ ] Pre-compiled
  - [v] Self-compiled ==> online using GitPod
- [v] Flashing tools used: ESP8266 Download Tool v3.6.6_____
- [v] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

00:00:00.001 HDW: ESP8266EX
00:00:00.047 CFG: Loaded from flash at F9, Count 11
00:00:00.052 QPC: Count 1
00:00:00.103 SNS: Hardware Serial
00:00:00.106 Project tasmota Tasmota Version 9.5.0.1(tasmota)-2_7_4_9(2021-06-30T02:15:06)
00:00:00.541 WIF: Connecting to AP1 omahklodran Channel 8 BSSId 0C:37:47:93:63:DF in mode 11n as tasmota_805E40-7744...
00:00:01.752 WIF: Connected
00:00:02.003 HTP: Web server active on tasmota_805E40-7744 with IP address 192.168.1.26
03:38:29.030 MQT: Attempting connection...
03:38:29.110 MQT: Connected
03:38:29.112 MQT: tele/tasmota_805E40/LWT = Online (retained)
03:38:29.113 MQT: cmnd/tasmota_805E40/POWER = 
03:38:29.118 MQT: tele/tasmota_805E40/INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"9.5.0.1(tasmota)","FallbackTopic":"cmnd/DVES_805E40_fb/","GroupTopic":"cmnd/tasmotas/"}}
03:38:29.120 MQT: tele/tasmota_805E40/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_805E40-7744","IPAddress":"192.168.1.26"}}
03:38:29.131 MQT: tele/tasmota_805E40/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4024049b","00000000","00000000"],"EXCVADDR":"00000040","DEPC":"00000000","CallChain":["40282f3d","4027f683","4027f683","4021dd04","4027f670","4027f668","402240ab","40216cb1","40228c8d","40101b73","40245fd1","4024f900","401018eb","4021e700","40101b73","4023dd4c","40228e80","40228d00","4021e700","40229f29","4022a02e","4021e700","4021ed77","40231cec","4021e700","4024c8f0","4024c892","4024c8f0","401000e1","4021e738","40248929"]}}}
03:38:29.133 MQT: stat/tasmota_805E40/RESULT = {"POWER":"OFF"}
03:38:29.134 MQT: stat/tasmota_805E40/POWER = OFF
03:38:32.485 QPC: Reset
03:38:33.441 MQT: tele/tasmota_805E40/STATE = {"Time":"2021-06-30T03:38:33","Uptime":"0T00:00:09","UptimeSec":9,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":8,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}
03:38:41.075 CMD: TCPBaudRate
03:38:41.081 MQT: stat/tasmota_805E40/RESULT = {"TCPBaudrate":0}
03:38:51.314 CMD: TCPStart 8888
03:38:51.319 MQT: stat/tasmota_805E40/RESULT = {"Command":"Error"}

```
- [v] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

Not using any Rules yet

```
- [v] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

03:50:53.474 CMD: Status 0
03:50:53.480 MQT: stat/tasmota_805E40/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_805E40","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
03:50:53.485 MQT: stat/tasmota_805E40/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:12:29","StartupUTC":"2021-06-30T02:38:24","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-06-30T03:32:52","SaveCount":12,"SaveAddress":"F8000"}}
03:50:53.488 MQT: stat/tasmota_805E40/STATUS2 = {"StatusFWR":{"Version":"9.5.0.1(tasmota)","BuildDateTime":"2021-06-30T02:15:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"361/699"}}
03:50:53.492 MQT: stat/tasmota_805E40/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["omahklodran",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
03:50:53.504 MQT: stat/tasmota_805E40/STATUS4 = {"StatusMEM":{"ProgramSize":608,"Free":392,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16301C","FlashFrequency":40,"FlashMode":0,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00024004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,41,45","Sensors":"1,2,3,4,5,6"}}
03:50:53.510 MQT: stat/tasmota_805E40/STATUS5 = {"StatusNET":{"Hostname":"tasmota_805E40-7744","IPAddress":"192.168.1.26","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"E8:68:E7:80:5E:40","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
03:50:53.514 MQT: stat/tasmota_805E40/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_805E40","MqttUser":"aspi","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
03:50:53.519 MQT: stat/tasmota_805E40/STATUS7 = {"StatusTIM":{"UTC":"2021-06-30T02:50:53","Local":"2021-06-30T03:50:53","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:50","Sunset":"20:57"}}
03:50:53.522 MQT: stat/tasmota_805E40/STATUS10 = {"StatusSNS":{"Time":"2021-06-30T03:50:53"}}
03:50:53.527 MQT: stat/tasmota_805E40/STATUS11 = {"StatusSTS":{"Time":"2021-06-30T03:50:53","Uptime":"0T00:12:29","UptimeSec":749,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":8,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}}
03:50:53.584 MQT: stat/tasmota_805E40/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["4024049b","00000000","00000000"],"EXCVADDR":"00000040","DEPC":"00000000","CallChain":["40282f3d","4027f683","4027f683","4021dd04","4027f670","4027f668","402240ab","40216cb1","40228c8d","40101b73","40245fd1","4024f900","401018eb","4021e700","40101b73","4023dd4c","40228e80","40228d00","4021e700","40229f29","4022a02e","4021e700","4021ed77","40231cec","4021e700","4024c8f0","4024c892","4024c8f0","401000e1","4021e738","40248929"]}}

[... 50 lines cut ...]
- set the IP Address, SSID, SSID password, MQTT broker host, MQTT password
- enable (delete the "//"), #define USE_TCP_BRIDGE
- compile the tasmota by running command "Platformio run -e tasmota"
- download the result (firmware.bin) inside folder .pio/build/tasmota/firmware.bin
- flash the firmware.bin to ESP8266 modules (Wemos D1 Mini/NodeMCU)
- run/open tasmota on pc

Step-3: Follow this document https://tasmota.github.io/docs/Serial-to-TCP-Bridge/#configuration
- assign GPIO01 = Serial Tx, GPIO03 = Serial Rx
- setting TCPBaudRate 9600 = fails, tasmota does not response
- check TCPBaudRate status = {TCPBaudrate":0}
- TCPStart 8888 = will not work, because TCPBaudRate has failed

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Setting TCPBaudRate 9600 should gives response from tasmota that the TCPBaudrate is set to 9600. Only after this we can start the TCPStart 8888, and later on we can create the Rules.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-06-30:
As expected ;-)  You have to use `TCP Rx` and `TCP Tx` for TCP Bridge

adharyanto, 2021-06-30:
Ah...... YES..... after changing them to TCP Tx and TCP Rx, and now it WORKS.....
Thank you...

closed 2021-06-30 by adharyanto