arendst/Tasmota #12526: sonoff basic R2 flash 9.5 credentials error but they aren't. no issue on 9.1 version

megamarco833, 2021-07-01:
### PROBLEM DESCRIPTION
with device sonoff R2 basic when i flash from zero a new tasmota firmware (9.5) after reboot, when i enter in configuration webpage, and i insert my credentials i receive this error: connect failed to NETGEAR please check your credentials
of course my credentials are correct.
before flash i use: esptool.py --port COM3 erase_flash

if i flash 9.1 version i do not see any issue
i used tasmota.bin file in both case

i experimented to flash also a nodemcu with 9.5 and in this case it works.

i try to flash again sonoff basic R2 with 9.1 then i upgrade to 9.5 and it works.
the issue is only if you stat from zero with 9.5 version tasmota.bin on sonoff R2 basic.

I try to flash from zero tasmota-lite.bin and this time it succeed.
with version 9.5 is it supposed to flash ONLY with lite version the sonoff basic R2 ??

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: espflash.exe and ESP-Flasher-Windows-x64.exe
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  no rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
explained in problem issue

### EXPECTED BEHAVIOUR
receive back the internal IP

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-07-01:
Sonoff Basic R2 is a badly designed device. The power supply is to weak to run Tasmota. You will encounter the strangest errors and issues with...
You may have more luck with Tasmota-lite since it has less features and needs less current.
**It is NOT a Tasmota issue.**

lmagyar, 2021-07-02:
**This seems to be a Tasmota issue, maybe triggered only by some HW config.**

I've investigated further:
- v9.4.0 and v9.5.0 has this issue, if you erase & upload a v9.4.0 or v9.5.0 firmware, you can't make WiFi to connect to your network.
- v9.1.0, v9.2.0, v9.3.0 and v9.3.1 does not have this issue, if you erase & upload an old firmware, you can connect to the WiFi! **Then upgrade to v9.5.0 and it works, without any issue.**

As I see, v9.4.0 and v9.5.0 has a simplified WiFi configuration menu at 192.168.4.1. **Maybe** this WiFi config/setup change triggers the issue on some HW. My **guess** is that v9.3.1 and older saves some data in the config that an upgrade can use, but these information are not present when you save the config directly with v9.4.0 and newer.

I noticed this error on my fresh new Wemos NodeMcu v3 boards (https://www.aliexpress.com/item/1005001889625136.html). I've tried pre-compiled and self-compiled firmware also. I've tried multiple power supplies. It seems to me to be some WiFi config data issue.

If you need any further information, please help me to investigate (I'm new to Tasmota and ESP8266, though I'm not a newbie in IT).

Jason2866, 2021-07-03:
@lmagyar  Mixing to different issues: Sonoff R2 is clearly a hardware issue. Thats the reason we introduced build Tasmota-lite and Tasmota sleep function was done. Andre Thomas and I did a lot of investigations here. There is NO doubt about. The issue is 100% reproducable. I have a sonoff basic r2. With extern power supply and chip coolers on the ESP8266 and the flash chip it does work. The thermal PCB design is the next mess beside the weak power supply. Best solution is replacing this device.
Second one you mentioned is the new connection method with trying to connect to the wifi network and showing the IP address the device got from DHCP Server. With some browsers and mobile phones this is not working. See here https://github.com/arendst/Tasmota/discussions/12512#discussioncomment-958409

tinamore, 2021-07-04:
I have same problem on sonoff basic R2 with version 9.5.0, before not have issue
--------
Connect failed to ...
Please, check your credentials
--------

Jason2866, 2021-07-04:
And? Example: My car run 1000km before. Now it cant drive anymore.

tinamore, 2021-07-04:
I don't know if the error is hardware or firmware.
But it took me an afternoon to connect to wifi for Sonoff. Maybe v9.5 doesn't work well for sonoff basic, but works well for other devices.
I'll go back to the older version.

3cky, 2021-07-04:
I did upgrade my Sonoff Basic R1 firmware to Tasmota 9.5.0 with full flash erasing, and it can't connect to any of my WiFi networks:

```
00:03:00.738 WIF: Connecting to AP test as tasmota_1C01BE-0446 ...
00:03:00.790 WIF: Connecting to AP test as tasmota_1C01BE-0446 ...
00:03:09.045 WIF: Connect failed with AP timeout
```

So I did downgrade to Tasmota 9.3.1 as suggested by @lmagyar, and it worked:

```
00:00:00.050 Project tasmota Tasmota Version 9.3.1(tasmota)-2_7_4_9(2021-03-09T16:13:55)
00:00:00.537 WIF: Connecting to AP1 test in mode 11n as tasmota_1C01BE-0446...
00:00:06.754 QPC: Reset
00:00:07.014 WIF: Connected
00:00:07.264 HTP: Web server active on tasmota_1C01BE-0446 with IP address 192.168.1.55
```

After that, I flashed back 9.5.0, and now it works too:

```
00:00:00.285 Project tasmota Tasmota Version 9.5.0(tasmota)-2_7_4_9(2021-06-17T08:26:59)
00:00:00.774 WIF: Connecting to AP1 test Channel 11 BSSId 00:18:F3:29:14:82 in mode 11n as tasmota_1C01BE-0446...
00:00:01.801 WIF: Connected
00:00:02.003 HTP: Web server active on tasmota_1C01BE-0446 with IP address 192.168.1.55
```

ascillato2, 2021-07-04:
Hi,

For the people with Sonoff Basics, it is well known that there are a batch of those with a problem in the power regulator that makes it very unstable when using the CPU too much. That is why Tasmota-Lite.bin version was introduced. A version with less features to use even less energy for those types of devices. Most of sonoff basics works fine with standard Tasmota and even with Tasmota with TLS enabled (that uses a lot of CPU). The Sonoff Basics that can't work with standard Tasmota, they can use tasmota-lite.

For the ones having a different hardware and having initial Wi-Fi connection issues only with latest Tasmota (9.5+) and in some routers, please refer to https://github.com/arendst/Tasmota/discussions/12512 where we are addressing it.

Thanks.

closed 2021-07-04 by ascillato2