arendst/Tasmota #12527: Node-RED, Serial to TCP Bridge, Tasmota vs ESP-Link

adharyanto, 2021-07-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Issue on Tasmota fragmented received message payload, via Serial to TCP Bridge.
This fragmentation problem does not happen with ESP-Link.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [v] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [v] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [v] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [v] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [v] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [v] Device used (e.g., Sonoff Basic): WeMos D1 Mini, WeMos NodeMCU_____
- [v] Tasmota binary firmware version number used: 9.5.0.1 (Serial to TCP Bridge enabled)_____
  - [ ] Pre-compiled
  - [v] Self-compiled
- [v] Flashing tools used: GitPod_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
Instead, I attached the Node-RED flow below:

[{"id":"6b3d606.248eca","type":"tab","label":"TCPBridge","disabled":false,"info":""},{"id":"bfac9e5c.fb6f3","type":"debug","z":"6b3d606.248eca","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":850,"y":120,"wires":[]},{"id":"1231d5f8.a480ea","type":"tcp in","z":"6b3d606.248eca","name":"WemosNMCUv31_tasmota","server":"client","host":"192.168.1.27","port":"8888","datamode":"stream","datatype":"buffer","newline":"","topic":"","base64":false,"x":630,"y":120,"wires":[["bfac9e5c.fb6f3"]]},{"id":"d5045369.fed81","type":"tcp out","z":"6b3d606.248eca","host":"192.168.1.27","port":"8888","beserver":"client","base64":false,"end":false,"name":"WemosNMCUv31_tasmota","x":360,"y":120,"wires":[]},{"id":"9dd80dc9.b7d41","type":"inject","z":"6b3d606.248eca","name":"buffer-jawab","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"[165,1,144,8,1,16,0,0,116,22,2,251,214]","payloadType":"bin","x":130,"y":180,"wires":[["d5045369.fed81"]]},{"id":"7716acba.ddb734","type":"inject","z":"6b3d606.248eca","name":"buffer-tanya","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"[165,64,144,8,0,0,0,0,0,0,0,0,125]","payloadType":"bin","x":130,"y":120,"wires":[["d5045369.fed81"]]},{"id":"54d20509.01271c","type":"comment","z":"6b3d606.248eca","name":"NodeMCU with Tasmota","info":"","x":350,"y":60,"wires":[]},{"id":"7a28c685.5d86e8","type":"inject","z":"6b3d606.248eca","name":"buffer-jawab","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"[165,1,144,8,1,16,0,0,116,22,2,251,214]","payloadType":"bin","x":130,"y":420,"wires":[["70195459.a7efac"]]},{"id":"26bdb81d.11e788","type":"inject","z":"6b3d606.248eca","name":"buffer-tanya","props":[{"p":"payload"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"[165,64,144,8,0,0,0,0,0,0,0,0,125]","payloadType":"bin","x":130,"y":360,"wires":[["70195459.a7efac"]]},{"id":"70195459.a7efac","type":"tcp out","z":"6b3d606.248eca","host":"192.168.1.26","port":"2323","beserver":"client","base64":false,"end":false,"name":"WemosD1Mini_esp-link","x":350,"y":360,"wires":[]},{"id":"eef17970.bc8b78","type":"debug","z":"6b3d606.248eca","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":850,"y":360,"wires":[]},{"id":"81b1ee3f.3166c","type":"tcp in","z":"6b3d606.248eca","name":"WemosD1Mini_esp-link","server":"client","host":"192.168.1.26","port":"2323","datamode":"stream","datatype":"buffer","newline":"","topic":"","base64":false,"x":620,"y":360,"wires":[["eef17970.bc8b78"]]},{"id":"3e871ca4.50f1b4","type":"comment","z":"6b3d606.248eca","name":"WeMosD1Mini with esp-link","info":"","x":360,"y":300,"wires":[]}]

```
- [v] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

12:06:52.556 CMD: backlog rule1
12:06:52.635 MQT: WemosNMCUv31/stat/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":37,"Free":474,"Rules":"ON System#Boot do TCPStart 8888 endon"}}

```
- [v] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

12:07:37.190 CMD: status 0
12:07:37.197 MQT: WemosNMCUv31/stat/STATUS = {"Status":{"Module":18,"DeviceName":"Wemos NodeMCU v3_1st","FriendlyName":["Wemos NodeMCU v3_1st"],"Topic":"WemosNMCUv31","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:07:37.205 MQT: WemosNMCUv31/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:02:23","StartupUTC":"2021-07-01T10:05:14","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-06-30T03:20:33","SaveCount":57,"SaveAddress":"FB000"}}
12:07:37.211 MQT: WemosNMCUv31/stat/STATUS2 = {"StatusFWR":{"Version":"9.5.0.1(tasmota)","BuildDateTime":"2021-06-30T02:15:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"393/699"}}
12:07:37.217 MQT: WemosNMCUv31/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["omahklodran_EXT","omahklodran"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
12:07:37.229 MQT: WemosNMCUv31/stat/STATUS4 = {"StatusMEM":{"ProgramSize":608,"Free":392,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16301C","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00024004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,41,45","Sensors":"1,2,3,4,5,6"}}
12:07:37.236 MQT: WemosNMCUv31/stat/STATUS5 = {"StatusNET":{"Hostname":"WemosNMCUv31-7744","IPAddress":"192.168.1.27","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"E8:68:E7:80:5E:40","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:07:37.241 MQT: WemosNMCUv31/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_805E40","MqttUser":"aspi","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:07:37.248 MQT: WemosNMCUv31/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-07-01T11:07:37","Local":"2021-07-01T12:07:37","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:56"}}
12:07:37.254 MQT: WemosNMCUv31/stat/STATUS10 = {"StatusSNS":{"Time":"2021-07-01T12:07:37"}}
12:07:37.261 MQT: WemosNMCUv31/stat/STATUS11 = {"StatusSTS":{"Time":"2021-07-01T12:07:37","Uptime":"0T01:02:23","UptimeSec":3743,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"omahklodran","BSSId":"0C:37:47:93:63:DF","Channel":8,"Mode":"11n","RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [v] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

12:08:11.459 CMD: weblog 4
12:08:11.465 MQT: WemosNMCUv31/stat/RESULT = {"WebLog":4}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

I have a Device which has UART interface as the communication port, and I need to connect it wirelessly, so created a Serial to TCP Bridge, and so I can access it via WiFi network.
I want control & monitor this device and integrated into my existing Node-RED, which already has many other devices integrated.
I use esp8266 as the Serial to TCP Bridge and of course the firmware inside, but currently only ESP-Link firmware that works, while Tasmota does not work.
Why? Below is the problem;
[... cut ...]
Why Tasmota cannot received the intact one msg.payload as originally was sent, same as ESP-Link?
Actually I want to use Tasmota as the Serial to TCP Bridge, because it can control the GPIOs as well, while ESP-Link cannot.
But this fragmented msg.payload problem has forced me back to use ESP-Link again.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-07-01:
Please provide the missing `Backlog Template; Module; GPIO 255`
Are you using hardware or software serial? Which Baudrate do you use?

adharyanto, 2021-07-01:
Hi,

 

I have added the Backlog Template, Module, GPIO255.

I am using hardware serial, means I connect (using jumper cable) the Tx (GPIO01) of one ESP8266 module (with ESP-Link firmware) to RX (GPIO03) of the other ESP8266 module (with Tasmota firmware).

And vice versa, the Rx (GPIO03) of one module (with ESP-Link firmware) to the Tx (GPIO01) of the other module (with Tasmota firmware).

Baud rates are set to 9600 for both modules. For the Tasmota, GPIO01 = TCP Tx, GPIO03 = TCP Rx.

 

I am using WeMos D1 Mini module and NodeMCU v3 hardware, one of them can be with the Tasmota or the ESP-Link firmware.

In the actual application I will only need 1 (one) module to be connected (Tx-to-Rx & Rx-to-Tx) to my DEVICE that will be monitored/controlled.

 

Below I attached snapshot of this two development boards used in this testing.

-        Black jumper cable = ground to ground

-        Yellow jumper cable = Rx to Tx

-        White jumper cable = Tx to Rx

 

 

Regards,

Antonius

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

adharyanto, 2021-07-01:
Further observation from the Tasmota console:

17:37:29.050 HTP: Console
17:37:33.015 TCP: to MCU/1: A540900800000000000000007D
17:37:36.235 TCP: from MCU: A5
17:37:36.238 TCP: from MCU: 409008
17:37:36.241 TCP: from MCU: 000000
17:37:36.244 TCP: from MCU: 000000
17:37:36.247 TCP: from MCU: 00007D

The above is the same message of buffer [165,64,144,8,0,0,0,0,0,0,0,0,125], it is just in HEX mode; A5 40 90 08 00 00 00 00 00 00 00 00 7D.
As we can see here, to MCU/1 there is one message with 13 bytes.
But from MCU, there are several messages, in this case five messages, with one message with 1 bytes and four messages with 3 bytes. The message from MCU suppose to be the same as to MCU/1, one message with 13 bytes, not fragmented. This fragmentation happens in random, combination of several messages with several combination of bytes.

Jason2866, 2021-07-01:
Did a quick test, not such a complex setup. Just sending via USB serial from PC and and receiving with a terminal programm which can connect to a tcp port. There is no fragmentation.
Your setup probably introduces hickups which ends in the fragmentation.

adharyanto, 2021-07-01:
I am not sure if I made any "hickups"... but this was just a normal/simple compiling Tasmota online using GitPod, nothing very special, just enable the #define USE_TCP_BRIDGE.

In real application, I use the WeMos D1 Mini, connected the Tx/Rx to my Device Rx/Tx (a BMS = battery management system), since I need to monitor and modify the parameters remotely. It is working fine using ESP-Link as the TCP Bridge, but when I tried using Tasmota it fails. Tried to investigate, then I found that message received by Tasmota is always fragmented.

You will need to check using Node-RED, since this message will be processed in the Node-RED.
Please try to use any ESP8266 module (WeMos, NodeMCU, etc.) flash with Tasmota, send serial signal to Rx (GPIO03), and then try to grab/read the signal from Node-RED, and you will see it will be fragmented. If you flash it using ESP-Link, it will not be fragmented.

The real scenario is like this:
- I send a question to the BMS, via TCP Bridge, send from Tx of my WeMos D1 Mini to Rx of the BMS
- The question is [165,64,144,8,0,0,0,0,0,0,0,0,125]
- The BMS should reply, by sending the message from Tx of the BMS to Rx of the WeMos D1 Mini
- The answer should be something like [165,1,144,8,1,16,0,0,116,22,2,251,214]
- I will process this answer further in the Node-RED, by parsing the bytes and get the value to be displayed
- But if the answer is fragmented, then I can not process...

Everything is fine with ESP-Link, but why not with Tasmota (fragmented)?

Jason2866, 2021-07-01:
Sorry, i dont have the time to do this complex test setup.

adharyanto, 2021-07-02:
> Did a quick test, not such a complex setup. Just sending via USB serial from PC and and receiving with a terminal programm which can connect to a tcp port. There is no fragmentation.
> Your setup probably introduces hickups which ends in the fragmentation.

Which esp8266 module/development board do you use? (WeMos D1 or NodeMCU)?
Need to inject the serial signal into the Rx (GPIO03) of this esp8266 module, then monitor it on any terminal program, I tried to use PuTTY, and the result is also fragmented.

joba-1, 2021-07-02:
Hi,

I don't see Tasmota behaviour as a problem:
Serial is character oriented and ip is packet oriented. 
That means serial can stop sending after one char or send as much characters at a time as it wants.
When would you send the character(s) over IP as a packet (of usually ~ 1-1500bytes)? How long would you wait for more data? Best approach depends on the usecase:
1) best latency and least memory usage if you read serial as much as is currently available and send it immediately
2) better bandwith if you wait for some time if more serial chars appear before sending (cost: more latency, memory for buffer).
3) Depend on protocol that you run over serial (e.g. wait for fixed number of chars, send length then data or send until line feed)

1) seems to be what Tasmota does and since serial is slower than esp can send IP it will be fragmented.
2) is probably what esp-link does (or it is just slower in sending IP data)
3) is not feasible for a general firmware that should work with all application protocols (you would need to specify protocols in settings and implement them)

Even for 2) there is no guarantee that the timeout to accumulate data will always be long enough or the message is not longer than the buffer can accomodate and you still get fragments.

So my recommendation is, to always assemble the fragments until your application protocol has all it needs to process it.

adharyanto, 2021-07-02:
Hi,
Thank for the comments.
The problem I am facing now is that I am processing the incoming serial data in the Node-RED, and the current Node-RED will need to read the message as a single buffer with 13 bytes long, and then need to parse/extract byte number-5 up to byte number-12 as the values of the measurements to be displayed or modified. So, the message must be a 13-bytes long buffer.

I am now trying to find a way and modify my Node-RED flow, try to join the fragmented incoming messages, join them into a single message with 13-bytes long, and so it can be processed further in the current existing Node-RED flow. The other problem is that the fragmented message is coming in random way, sometimes the message coming in 2 or 3 or 4 or 5 messages, even in 13 messages with only 1-byte per message. And there are many different messages with different info coming as a stream, so I also need to detect the header of each message.

I am just wondering why the ESP-Link can send the message in single 13-bytes buffer, and why Tasmota is fragmented into several messages.

adharyanto, 2021-07-03:
Fyi, I have tried to join the fragmented incoming buffer messages inside the Node-RED, using "join" node. The fragmented messages coming from Tasmota, now become a single message with 13-bytes long, and can be processed further in the Node-RED flows. But this only works if the message is sent/received manually, slowly sent one by one, or a stream of messages with slow rate.

But when it is connected to the original signal coming from the Device (the BMS), the rate of this incoming stream signal is too high for the Node-RED to process. The Node-RED fails/cannot process very fast incoming signal stream, resulting some wrong displayed data. I believe this failure happens especially when the message is fragmented into many messages, that takes longer time to be processed by the Node-RED.

For the mean time I still have to rely on the ESP-Link to monitor and control remotely this BMS device.

Jason2866, 2021-07-03:
As @joba-1 wrote TCP Bridge in Tasmota has no buffer. It does exactly what he described as "case 1" The effect you encounter is expected by this design. Introducing a buffer would cost to many of the rare resources. The price beeing a universal firmware with many features included.

s-hadinger, 2021-07-03:
Hi, the current design of TCP Bridge is more suited for high speed serial (115200 bauds). It sends to TCP whatever is in the incoming buffer at every tick (default every 50ms but can be changed with `Sleep`)

Grouping into packets is always arbitrary in Serial since there is no notion of a packet. It's just a continuous flow. 

Also there is only one buffer for sending and receiving.

Let me see if I can easily add a "wait for x ms before sending packet" feature...

adharyanto, 2021-07-05:
Hi,
Fyi, I tried to add 5ms delay on the "xdrv_41_tcp_bridge" below.
This helps for my application, I receive intact 13-bytes message (not fragmented), and no need any additional join function in the Node-RED. But this maybe not the best solution, (not flexible).
I think that "wait for x ms" feature would be much better.

[image]

nagyrobi, 2021-07-07:
I had similar issues with a Sanxing SX631 smart meter P1. It's sending bursts of data at 115200 baud every 10 secsonds. TCP to Serial in Tasmota transmits heavily fragmented data to the terminal.

adharyanto, 2021-07-07:
> I had similar issues with a [Sanxing SX631](https://tasmota.github.io/docs/P1-Smart-Meter/) smart meter P1. It's sending bursts of data at 115200 baud every 10 secsonds. TCP to Serial in Tasmota transmits heavily fragmented data to the terminal.
> 
> Increasing the buffer did not give 100% results, the transmission was unusable with Home Assistant's built-in P1 Smart Meter integration.
> 
> Tasmota's own [Smart Meter interface](https://tasmota.github.io/docs/Smart-Meter-Interface/) worked, after some buffering tweaks, but raw Serial to TCP never worked.

Hi nagyrobi,
What about using ESP-Link as the Serial-TCP Bridge, does it work for your application?

nagyrobi, 2021-07-07:
I never tried that, and I ended up deploying SMI linked above. @gemu2015 offered great support in debugging it and making it work with Sanxing.

closed 2021-07-23 by github-actions[bot]