arendst/Tasmota #12528: Moving average (M:vname) for Scripting gives inadequate results

nikvostro, 2021-07-01:
### PROBLEM DESCRIPTION
Moving average (M:vname) for Scripting gives inadequate results. This actual at least for 2 independand builds (env): TTGO and EPS32CAM. Tested on master 9.5.0 and development on 01 Jul 2021 branches.

### REQUESTED INFORMATION

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP32CAM and Lilygo TTGO (T-Display)
- [ ] Tasmota binary firmware version number used: 
  - [ ] Pre-compiled
  - [ ] Self-compiled master 9.5.0 and development on 01 Jul 2021
- [ ] Flashing tools used: VS Code/Platformio
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
16:45:14.440 RSL: RESULT = {"NAME":"ESP32-ttgo-display","GPIO":[32,1,1,1,992,6592,1,1,1,1,1,1,6624,1,896,864,0,1,1,1024,0,1,1,288,0,0,0,0,1,1,1,33,1,0,0,1],"FLAG":0,"BASE":1}
16:45:14.690 RSL: RESULT = {"Module":{"0":"ESP32-ttgo-display"}}
16:45:14.997 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"992":"Backlight"},"GPIO5":{"6592":"ST7789 CS"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"6624":"ST7789 DC"},"GPIO17":{"0":"None"},"GPIO18":{"896":"SSPI SCLK"},"GPIO19":{"864":"SSPI MOSI"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"1024":"Display Rst"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"288":"Led1"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"33":"Button2"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [n/a] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- [ ] Provide the output of this command: `Status 0`:
```lua
16:45:49.697 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"TasmotaTTGO","FriendlyName":["TasmotaTTGO"],"Topic":"tasmota_0BC6B8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
16:45:49.732 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:08:10","StartupUTC":"2021-07-01T13:37:39","Sleep":100,"CfgHolder":4617,"BootCount":31,"BCResetTime":"2021-06-28T17:18:34","SaveCount":473}}
16:45:49.761 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.1(display)","BuildDateTime":"2021-07-01T16:30:45","Core":"1_0_7_1","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6-V3","CR":"398/699"}}
16:45:49.779 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["domuk",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","1405C8000100060000005A00000000000000","00800080","00006008","00000000"]}}
16:45:49.806 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1297,"Free":1856,"Heap":192,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":16384,"FlashSize":16384,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","07AA858E","000C1FA1","000000C3","00000000","00000080","00000020","403F8000","00000008"],"Drivers":"1,2,4,5,8,9,10,13,50,52","Sensors":"1,2,5,6,87"}}
16:45:49.836 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_0BC6B8-1720","IPAddress":"192.168.11.122","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"3C:61:05:0B:C6:B8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:45:49.858 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-01T13:45:49","Local":"2021-07-01T16:45:49","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+03:00","Sunrise":"03:49","Sunset":"21:15"}}
16:45:49.877 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-01T16:45:49","ESP32":{"Temperature":42.2},"TempUnit":"C"}}
16:45:49.888 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-01T16:45:49","Uptime":"0T00:08:10","UptimeSec":490,"Heap":191,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":10,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"domuk","BSSId":"F0:79:59:36:57:40","Channel":3,"Mode":"11n","RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
Nothin appears in log for that case

### TO REPRODUCE
1. Input and enable following script and look at ave variable in Console:
```
>D
rand=0
M:ave=0

>S
rand=rnd(100)
ave=rand
print %rand% - %ave%
```

2. Specify the length of filter M:ave=0 to value 4, 6, 12 and other. Save.
`M:ave=0 4` or `M:ave=0 6` or `M:ave=0 1`
3. Look now at ave value - it will become inadequate.
4. You can also try rebooting device with restart 1 or manually -> values for moving average will remain inadequate

### EXPECTED BEHAVIOUR
Moving average should be calculated properly

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT
I'm not a proffessional coder but it looks like an error with some pointer variables that are referencing to the moving average data array.

user_config_override.h:
```
#ifdef USE_RULES
#undef USE_RULES
#endif

#ifndef USE_SCRIPT
#define USE_SCRIPT
#endif

#define USE_SPI
#define USE_DISPLAY
#define USE_DISPLAY_ST7789
#define SHOW_SPLASH
```

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-07-01:
i checked your script with an older version and it worked. i then tried with most recent dev version and it fails like yours.
however the bug is not in moving average but the print function somehow fails.
if you spilt the print into 2 lines it works.
print %rand%
print movav = %ave%

since nothing was changed in the scripter printing function it must be a side effect of other changes in Tasmota.
 i am currently extremely busy with other things and have no time to check what causes this bug in the printing function.
will check this if i have  more time again.

nikvostro, 2021-07-01:
Hello gemu2015!

I can't agree with you regarding supposed issue with printing function, because:

Nothing changed for me if I splitting print into two rows:
[image]
[image]

Declared length 12:
[image]
[image]

ESP32CAM with added HTML rows like in examle (docs/Scripting - ESP32 Webcam support):
[image]
[image]

gemu2015, 2021-07-02:
ok, recently memory allocation was changed from calloc to malloc. this caused an issue with moving average vars not set to zero before start. this should be fixed now. please test by replacing scripter source file.

[xdrv_10_scripter.ino.zip](https://github.com/arendst/Tasmota/files/6752698/xdrv_10_scripter.ino.zip)

nikvostro, 2021-07-02:
(I've replaced LOG_LEVEL with DEBUG to hide extra info intersecting with my "print")
Now looks much better. I've tried to change legth -> result seems to adequeate now in all cases.
I consider issue as resolved now :) Thanks a lot!

[image]
[image]

nikvostro, 2021-07-02:
By the way - abs() function is under USE_ANGLE_FUNC def as well as geometric functions. I think it's more logical to move it out OR modify the Docs by adding notice in brackets:
[image]

gemu2015, 2021-07-02:
this was actually a mistake. will move abs() to standard functions.

PR #12541 (2021-07-02): some fixes
## Description:

**Related issue (if applicable):** fixes #12528

fixes moving average 
abs(x) now always compiled
smlv[x] to check if sml imput valid

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.1.0.7
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/b [... cut]
closed 2021-07-02 by arendst