arendst/Tasmota #12532: Powerset, VoltageSet and Currentset commands fail with "Error" on Module 54 (TuyaMCU)

wb-ha, 2021-07-02:
### PROBLEM DESCRIPTION
Powerset, VoltageSet and Currentset commands fail with "Error" on Module 54 (TuyaMCU)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Brilliant Smart Elite Wifi Light Switch (Tuya)
- [ x] Tasmota binary firmware version number used: 9.5.0.2
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: Tuyaconvert
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
03:22:52.366 CMD: Backlog Template; Module; GPIO 255
03:22:52.441 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
03:22:52.647 RSL: RESULT = {"Module":{"54":"Tuya MCU"}}
03:22:52.869 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
03:23:42.324 CMD: status 0
03:23:42.330 RSL: STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
03:23:42.335 RSL: STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:52","StartupUTC":"2021-07-02T02:18:50","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-07-02T03:18:57","SaveCount":30,"SaveAddress":"FB000"}}
03:23:42.339 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.2(tasmota)","BuildDateTime":"2021-07-01T13:41:45","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"349/699"}}
03:23:42.343 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT40",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
03:23:42.352 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":0,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
03:23:42.357 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-5305","IPAddress":"10.1.5.129","Gateway":"10.1.5.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.5.1","Mac":"EC:FA:BC:44:F4:B9","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
03:23:42.361 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_44F4B9","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
03:23:42.366 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-02T02:23:42","Local":"2021-07-02T03:23:42","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:56"}}
03:23:42.369 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
03:23:42.375 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-02T03:23:42","ENERGY":{"TotalStartTime":"2021-07-02T03:09:26","Total":0.003,"Yesterday":0.000,"Today":0.003,"Power":55,"ApparentPower":55,"ReactivePower":0,"Factor":1.00,"Voltage":232,"Current":0.238}}}
03:23:42.381 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-02T03:23:42","Uptime":"0T00:04:52","UptimeSec":292,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"IOT40","BSSId":"F6:92:BF:21:B4:F9","Channel":6,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
03:20:43.392 CMD: powerset 100
03:20:43.394 SRC: WebConsole from 10.1.1.75
03:20:43.396 CMD: Grp 0, Cmnd 'POWERSET', Idx 1, Len 3, Data '100'
03:20:43.400 RSL: RESULT = {"Command":"Error"}
03:20:43.851 CFG: Saved to flash at F5, Count 28, Bytes 4096
03:20:47.681 CMD: voltageset 240
03:20:47.683 SRC: WebConsole from 10.1.1.75
03:20:47.685 CMD: Grp 0, Cmnd 'VOLTAGESET', Idx 1, Len 3, Data '240'
03:20:47.689 RSL: RESULT = {"Command":"Error"}
03:20:52.490 CMD: currentset420
03:20:52.492 SRC: WebConsole from 10.1.1.75
03:20:52.494 CMD: Grp 0, Cmnd 'CURRENTSET', Idx 420, Len 0, Data ''
03:20:52.498 RSL: RESULT = {"Command":"Error"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash Tasmota development binary
From console
Module 54
Backlog TuyaMCU 31,103;TuyaMCU 32, 102;TuyaMCU 33, 104; TuyaMCU 11,1;TuyaMCU 12,2
- run commands above

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
These commands should calibrate the power sensor output displayed

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
As you can see from screenshot the sensor is working ok but should read 240V , 100W and around 0.42 A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I have around 25 of these devices operating within the house which were flashed approx 2 years ago and had Tasmota versions upgraded several times (now 9.2.0).  The commands worked fine when the devices were originally setup but now I have gone in and tested them as well and they produce the same error (however they are calibrated and continue to work properly - however I could not change the calibration if I wanted to).  I have tried versions 7.2.0, 8.5.1 and 9.2.0 as well as the latest dev version and all produce the same error).
I have tried 'reset 5' and reconfigured settings but this also doesn't work.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-07-08:
As far as I know the Tuya Energy reporting never used local Tasmota calibration; I see no reference in the Tuya driver to any Tasmota calibration variable or code.

It just reports what it receives from the Tuya hardware.

wb-ha, 2021-07-08:
Thanks for looking, that is weird as I have nearly 30 of these working with calibrated data with Tasmota using TuyaMCU - I have spare units so can test options but if you say there is no way a Voltageset command could work then I am pretty lost on where to go with this one - as I can't duplicate how I did it originally.

closed 2021-07-23 by github-actions[bot]