arendst/Tasmota #12547: Power blink mode on system#boot ignored

jonandel, 2021-07-03:
### PROBLEM DESCRIPTION
System#boot rule sets Powermode to 3 (blink) - but is being ignored.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: 9.4.0-9.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:14:32.438 CMD: Backlog Template; Module; GPIO 255
18:14:32.440 SRC: WebConsole from 192.168.2.64
18:14:32.443 CMD: Grp 0, Cmnd 'BACKLOG', Idx 1, Len 26, Data 'Template; Module; GPIO 255'
18:14:32.498 SRC: Backlog
18:14:32.501 CMD: Grp 0, Cmnd 'TEMPLATE', Idx 1, Len 0, Data ''
18:14:32.508 MQT: stat/External_Siren/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:14:32.716 SRC: Backlog
18:14:32.719 CMD: Grp 0, Cmnd 'MODULE', Idx 1, Len 0, Data ''
18:14:32.723 MQT: stat/External_Siren/RESULT = {"Module":{"18":"Generic"}}
18:14:32.966 SRC: Backlog
18:14:32.969 CMD: Grp 0, Cmnd 'GPIO', Idx 1, Len 3, Data '255'
18:14:32.979 MQT: stat/External_Siren/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"226":"Relay3"},"GPIO3":{"0":"None"},"GPIO4":{"257":"Relay_i2"},"GPIO5":{"256":"Relay_i1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
18:16:04.190 CMD: Grp 0, Cmnd 'RULE', Idx 1, Len 0, Data ''
18:16:04.196 MQT: stat/External_Siren/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":32,"Free":479,"Rules":"ON System#Boot DO Power3 3 ENDON"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:16:34.690 MQT: stat/External_Siren/STATUS = {"Status":{"Module":18,"DeviceName":"Ext_Siren","FriendlyName":["Ext_Siren","Ext_Siren_aux","Ext_Siren_led"],"Topic":"External_Siren","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:16:34.702 MQT: stat/External_Siren/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:31:14","StartupUTC":"2021-07-03T16:45:20","Sleep":50,"CfgHolder":4617,"BootCount":214,"BCResetTime":"2021-06-08T19:36:17","SaveCount":422,"SaveAddress":"F8000"}}
18:16:34.710 MQT: stat/External_Siren/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"371/699"}}
18:16:34.720 MQT: stat/External_Siren/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NetGear",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
18:16:34.736 MQT: stat/External_Siren/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:16:34.748 MQT: stat/External_Siren/STATUS5 = {"StatusNET":{"Hostname":"External_Siren-2108","IPAddress":"192.168.2.183","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"C4:5B:BE:4D:C8:3C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:16:34.758 MQT: stat/External_Siren/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.210","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4DC83C","MqttUser":"mqtt_u","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:16:34.768 MQT: stat/External_Siren/STATUS7 = {"StatusTIM":{"UTC":"2021-07-03T17:16:34","Local":"2021-07-03T18:16:34","StartDST":"2021-03-28T01:00:00","EndDST":"2021-10-31T02:00:00","Timezone":99,"Sunrise":"04:46","Sunset":"21:21"}}
18:16:34.776 MQT: stat/External_Siren/STATUS10 = {"StatusSNS":{"Time":"2021-07-03T18:16:34"}}
18:16:34.788 MQT: stat/External_Siren/STATUS11 = {"StatusSTS":{"Time":"2021-07-03T18:16:34","Uptime":"0T00:31:14","UptimeSec":1874,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"NetGear","BSSId":"DC:39:6F:31:AD:A1","Channel":6,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Install Tasmota
setup three relays.
D4 is relay3
execute rule : ON System#Boot DO Power3 3 ENDON
Then restart.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Power3 should be set to mode 3 - blink, when System#boot rule is executed - but is off/ignored.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-07-03:
Do not cross-post. See my answer on Discord.

ascillato2, 2021-07-03:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-07-03 by ascillato2

jonandel, 2021-07-04:
Not duplicated, and not resolved.
While console won't show 'Blink mode' on subsequent calls, it remains the boot rule isn't setting the state to blink.

sfromis, 2021-07-04:
Blink is not a state, it is an ongoing background action. You just left the conversation on Discord. Did you try with my test rule?

jonandel, 2021-07-04:
Yes, but it's missing the point as commented in the (now closed) issue.Setting blink - whether a state or otherwise - doesn't work when triggered via system#boot event.Your rule simply informed me of what I stated. Blink isn't triggered, the state on that relay remains static (off), unless activated via a command  (console or mqtt).

sfromis, 2021-07-04:
Well, when your steps to reproduce does not reproduce the issue, debugging gets nowhere. When following those steps, and including the test rule `on power3#state do var1 %value% endon`, my test device starts blinking after restarting, including the test rule chattering about the changes.

jonandel, 2021-07-05:
Aha, so it does work for you !, I wonder what's different ?Do you have mqtt on and configured ?Do you have three relays ?Is your relay3 on D4 on a Nodemcu configured as a generic tasmota device as mine is ?Are you using 9.4, or 9.5 or latest dev as of Saturday? (I tried all three same effect for me).You can see in the log I posted, it appears to execute the command on boot, but no blink on the d4 pin.I notice a Reset in the log after the system boot,  is that relevant ? Do you see that too ?Knowing the blink power 'state' is never shown beyond when the command is issued was definitely news, so thanks for that ... and perhaps that should be in the docs.  But I'm still scratching my head as to why mine doesnt work.I will test different scenarios over next few days...got any other ideas ?

sfromis, 2021-07-05:
Using 9.5 including fully configured MQTT, 3 relay pins, 3rd on gpio2. Not a NodeMCU, but which board the chip is on makes no difference to how Tasmota operates on the pins. Not clear what "did notice a reset" means, however "QPC: Reset" is just a normal part of the boot sequence. And my test rule clearly shows how the pin is being toggled, even with no physical relay connected.

```
10:07:50.235 RUL: SYSTEM#BOOT performs "Power3 3"
10:07:50.240 MQT: xxxxxxxx/stat/POWER = {"POWER3":"Blink ON"}
10:07:50.486 RUL: POWER3#STATE performs "var1 1"
10:07:50.493 MQT: xxxxxxxx/stat/VAR = {"Var1":"1"}
10:07:51.487 RUL: POWER3#STATE performs "var1 0"
10:07:51.493 MQT: xxxxxxxx/stat/VAR = {"Var1":"0"}
....
```

jonandel, 2021-07-05:
Thanks for your persistence on this.

I dont need a rule, as I have an LED attached...and it aint flashin' ! :-)

 

Give me a few days to try a few different perms and combs (as mentioned in previous reply), to at least get the blink running via a rule on a pin - I can also try triggering via a timer etc - but at the moment, unless I use the console for issuing 'POWER3 3', the boot rule doesnt work for me.

Only other info I can think f for now is my blinkcount is set to zero (as I want it to blink 'forever').

Any other ideas on what to try or insight greatly appreciated....

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

sfromis, 2021-07-05:
You could try fixing it by following your "steps to reproduce".... Generally getting back to config square one can be useful when things get "surprising".