arendst/Tasmota #12554: Command Ruletimer0 0 -> Error ?

cool-lox, 2021-07-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Problem with command RuleTimer0 0, which should stop all ruletimer simulatiously according to documentation
But I get only Error. Single usage like Ruletimer7 0 are working

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):  YTF IR Bridge Module
- [ ] Tasmota binary firmware version number used: 9.0.0.2(ir)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

09:14:04 CMD: ruletimer7 60
09:14:04 MQT: stat/IRbox/RESULT = {"T1":0,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":60,"T8":2}
09:14:08 CMD: ruletimer0 0
09:14:08 MQT: stat/IRbox/RESULT = {"Command":"Error"}
09:14:12 CMD: ruletimer 0
09:14:12 MQT: stat/IRbox/RESULT = {"T1":0,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":52,"T8":0}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-07-04:
Please use latest v9.5. firmware. You use a very old version.

sfromis, 2021-07-04:
Verified to work with current Tasmota, 9.5. Current docs does not 100% apply to old versions.

cool-lox, 2021-07-04:
Ohh sorry sometimes is soo easy thanks, I`m using now 9.3.1. (ir) and it works - thanks for fast help

closed 2021-07-04 by cool-lox