arendst/Tasmota #12555: ESP32: Berry: tasmota.publish("strength", '{}') crashes tasmota

msperl, 2021-07-04:
### PROBLEM DESCRIPTION

On the berry console running
```
tasmota.publish("strength", '{}')
```
crashes tasmota (ping lost) - sometimes it reboots and comes back up, other times it requires a power off/on.

while:
```
tasmota.publish("strength/a", '{}')
```
works just fine.

I understand that this is not "ideal" from the use case perspective - it really should be structured - but I came accross it while prototyping a simple driver for TF Mini Plus LIDAR on a device without mqtt configured (so I actually thought it was related to MQTT missing, which is not the case)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Node MCU
- [X] Tasmota binary firmware version number used: 9.5.0.2(tasmota) - 2021-07-03T15:59:15
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:18:12.922 RSL: RESULT = {"NAME":"TF-MINI-PLUS","GPIO":[1,1792,0,1824,1,1,1,1,1,1,1,1,0,0,0,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
08:18:13.136 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
08:18:13.401 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:19:26.819 CMD: Status 0
08:19:26.826 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_XXXXXX","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:19:26.832 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:08:42","StartupUTC":"2021-07-04T07:10:44","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-07-02T14:51:21","SaveCount":17}}
08:19:26.836 RSL: STATUS2 = {"StatusFWR":{"Version":"9.5.0.2(tasmota)","BuildDateTime":"2021-07-03T15:59:15","Core":"1_0_7_1","SDK":"v3.3.5-138-g714f27f56","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"386/699"}}
08:19:26.841 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00808009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
08:19:26.858 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1240,"Free":1984,"Heap":190,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","000840D2","20200000","0000002D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,38,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,87"}}
08:19:26.863 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_XXXXXX-XXXX","IPAddress":"X.X.X.X","Gateway":"X.X.X.X","Subnetmask":"255.255.255.0","DNSServer":"X.X.X.X","Mac":"24:62:AB:XX:XX:XX","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
08:19:26.867 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FC23B8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:19:26.873 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-07-04T07:19:26","Local":"2021-07-04T08:19:26","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:53","Sunset":"20:55"}}
08:19:26.881 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-07-04T08:19:26","ESP32":{"Temperature":53.3},"TempUnit":"C"}}
08:19:26.889 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-07-04T08:19:26","Uptime":"0T00:08:42","UptimeSec":522,"Heap":189,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"XXXXXXXX","BSSId":"64:70:02:XX:XX:XX","Channel":1,"Mode":"11n","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
No output as network drops and no output can be seen
```

### TO REPRODUCE
On the berry console running
```
tasmota.publish("strength", '{}')
```

### EXPECTED BEHAVIOUR
On the berry console I should see the following output:
```
tasmota.publish("strength", '{}')
{}
```

Alternatively an error/exception would also be ok ...
Actually I guess it would be better, as it would also give a warning about bad behavior to the developer...

### SCREENSHOTS
none

### ADDITIONAL CONTEXT
I understand that the topic name is on the root level, but it came up during my first steps with Berry, so as I was not concerned about mqtt at this very moment (i.e not configured), I just used something quick to see that it works before finishing it (and it took me quite some time to realize that the problem was not in the berry script - or not completely...)

sfromis, 2021-07-04:
AFAICT, the issue is simply that it does not make sense to use `tasmota.publish` without MQTT enabled. If MQTT is enabled, but the broker is not available, the command works as expected.

sfromis, 2021-07-04:
With MQTT enabled/connected, I do not get the crash.

msperl, 2021-07-04:
Yes, I understand that is in the end the way to go, but for simple testing  it should not crash tasmota either when no mqtt is configured.

also it only crashes when using a topic without “/“ - I.e “ab” not “a/b”

So there is somewhere an active bug that results in that crash that requires a power off/on to resolve…

Maybe it would be sufficient to have a check to say: mqtt not enabled and stop processing the message itself - but that does not look at the root cause.

s-hadinger, 2021-07-04:
Does it crash if you send the equivalent publish command is Tasmota console?

msperl, 2021-07-04:
Will give it a try when I am right next to the device again…

msperl, 2021-07-05:
well I have tried now with
```
publish strength "{}" 
```
and in this case tasmota just rebooted.

The console shows the following info for the reboot:
```
19:10:53.180 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU"}}
```

Ping looks like this:
```
64 bytes from 10.10.10.207: icmp_seq=74 ttl=255 time=65.443 ms
64 bytes from 10.10.10.207: icmp_seq=75 ttl=255 time=87.307 ms
64 bytes from 10.10.10.207: icmp_seq=76 ttl=255 time=116.990 ms
64 bytes from 10.10.10.207: icmp_seq=77 ttl=255 time=131.354 ms
64 bytes from 10.10.10.207: icmp_seq=78 ttl=255 time=161.037 ms
64 bytes from 10.10.10.207: icmp_seq=79 ttl=255 time=177.575 ms
64 bytes from 10.10.10.207: icmp_seq=80 ttl=255 time=209.418 ms
64 bytes from 10.10.10.207: icmp_seq=81 ttl=255 time=21.235 ms
Request timeout for icmp_seq 82
Request timeout for icmp_seq 83
Request timeout for icmp_seq 84
Request timeout for icmp_seq 85
Request timeout for icmp_seq 86
Request timeout for icmp_seq 87
Request timeout for icmp_seq 88
Request timeout for icmp_seq 89
64 bytes from 10.10.10.207: icmp_seq=90 ttl=255 time=16.692 ms
Request timeout for icmp_seq 91
Request timeout for icmp_seq 92
Request timeout for icmp_seq 93
Request timeout for icmp_seq 94
Request timeout for icmp_seq 95
Request timeout for icmp_seq 96
64 bytes from 10.10.10.207: icmp_seq=97 ttl=255 time=170.584 ms
64 bytes from 10.10.10.207: icmp_seq=98 ttl=255 time=186.109 ms
Request timeout for icmp_seq 99
Request timeout for icmp_seq 100
Request timeout for icmp_seq 101
Request timeout for icmp_seq 102
Request timeout for icmp_seq 103
Request timeout for icmp_seq 104
64 bytes from 10.10.10.207: icmp_seq=105 ttl=255 time=142.198 ms
64 bytes from 10.10.10.207: icmp_seq=106 ttl=255 time=9.696 ms
64 bytes from 10.10.10.207: icmp_seq=107 ttl=255 time=183.123 ms
64 bytes from 10.10.10.207: icmp_seq=108 ttl=255 time=205.177 ms
64 bytes from 10.10.10.207: icmp_seq=109 ttl=255 time=25.394 ms
64 bytes from 10.10.10.207: icmp_seq=110 ttl=255 time=56.991 ms
64 bytes from 10.10.10.207: icmp_seq=111 ttl=255 time=71.512 ms
64 bytes from 10.10.10.207: icmp_seq=112 ttl=255 time=92.057 ms
64 bytes from 10.10.10.207: icmp_seq=113 ttl=255 time=115.300 ms
```

Hope this gives some more insight.

s-hadinger, 2021-07-05:
Thanks. So the bug seems to be deeper in the publish command

arendst, 2021-07-06:
Verified if no MQTTHost is defined and MQTT is enabled.

Investigating...

ascillato2, 2021-07-06:
Fixed by Theo. Thanks.

closed 2021-07-06 by ascillato2