arendst/Tasmota #12560: rst looping when flashing ESP32_CAM with tasmota32_webcam.bin

feutl, 2021-07-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am flashing using esp32-cam AI-Thinker with ESP-Flasher 64 for Windows. Flash runs fine, no errors, after resetting or "power on/off) the devide ends up in a boot loop.
I tested Tasmota-32 9.4 and 9.5 Webcam version with the USB Programmer as Power Source using 3,3V and 5v. I tested a 5V / 1A Power Source and also used a proffesional power supply with up to 5A, but with 3,3V not more than 0.24A were consumed during the boot loop.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 9.5 and 9.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESP Flasher 64 Windows 10
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output of ESP-Flasher after reboot here:

[20:42:01]
[20:42:01]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:42:01]configsip: 0, SPIWP:0xee
[20:42:01]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:42:01]mode:QIO, clock div:1
[20:42:01]load:0x3fff0018,len:4
[20:42:01]load:0x928b2121,len:-1725455839
[20:42:01]1162 mmu set 00010000, pos 00010000
[20:42:01]1162 mmu set 00020000, pos 00020000
[20:42:01]1162 mmu set 00030000, pos 00030000
[20:42:01]1162 mmu set 00040000, pos 00040000
[20:42:01]1162 mmu set 00050000, pos 00050000
[20:42:01]1162 mmu set 00060000, pos 00060000
[20:42:01]1162 mmu set 00070000, pos 00070000
[20:42:01]1162 mmu set 00080000, pos 00080000
[20:42:01]1162 mmu set 00090000, pos 00090000
[20:42:01]1162 mmu set 000a0000, pos 000a0000
[20:42:01]1162 mmu set 000b0000, pos 000b0000
[20:42:01]1162 mmu set 000c0000, pos 000c0000
[20:42:01]1162 mmu set 000d0000, pos 000d0000
[20:42:01]1162 mmu set 000e0000, pos 000e0000
[20:42:01]1162 mmu set 000f0000, pos 000f0000
[20:42:01]1162 mmu set 00100000, posets Jun  8 2016 00:22:57
[20:42:01]

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Initial flashing an ESP32-Cam AI-Thinker.
I had ESPHome on it previously, which was working.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
At least booting the device to end up with an WIFI to connect to.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
There were tickets opened already, having similar issued, but they were closed without a proper solution to the problem, so I start this endeavour from scratch :) perhaps the combination of ESPHome as the image and flashing Tasmota-32 with ESP-Flasher could be the problem.

https://github.com/arendst/Tasmota/issues/12293
https://github.com/arendst/Tasmota/issues/12291

**(Please, remember to close the issue when the problem has been addressed)**

martin072, 2021-07-08:
Not sure if this helps, I had a similar issue. flashed it initially with the default tasmota_esp32.bin, and from there upgraded to the webcam version, that seemed to work for me.

feutl, 2021-07-08:
@martin072 this was a great hint.
Did install the `tasmota_esp32.bin` and used the webui to install the `webcam.bin` version and voila - it works.

Thanks

NoskaOff, 2021-07-08:
@martin072 First time I'm trying to install Tasmota and your solution is the only one that worked so far. Thanks a lot

closed 2021-07-10 by arendst

Jason2866, 2021-07-10:
Thx, for your input. It was the the tip where to search!

PR #12613 (2021-07-10): Use mode `dout` for all ESP32 builds
since Platformio has a bug generating/flashing the correct flash modes.
Issue https://github.com/platformio/platform-espressif32/issues/584

Maybe fixes #12560

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.1.0.7
  - [x] I accept the [CLA](h [... cut]