arendst/Tasmota #12564: MQTT topic can no longer be changed

dansiebert, 2021-07-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Since Tasmota 9.5.0 the MQTT topic can no longer be changed. Regardless of what I enter under Configuration/MQTT/ Topic, the value is always tasmota_% 06X after restart. I can simulate the problem on several Sonoff, Gosund, Nous switch sockets.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Basic, Nous A1, Gosund SP112
- [X] Tasmota binary firmware version number used: 9.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA Firmware Upgrade
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-07-05:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

dansiebert, 2021-07-05:
Provide the output of command: Backlog Template; Module; GPIO 255:
12:26:59.624 CMD: Backlog Template; Module; GPIO 255
12:26:59.650 MQT: stat/tasmota_907EDB/RESULT = {"NAME":"NOUS A1","GPIO":[320,0,576,0,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
12:26:59.876 MQT: stat/tasmota_907EDB/RESULT = {"Module":{"0":"NOUS A1"}}
12:27:00.132 MQT: stat/tasmota_907EDB/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

If using rules, provide the output of this command: Backlog Rule1; Rule2; Rule3:
12:27:47.198 CMD: Backlog Rule1; Rule2; Rule3
12:27:47.219 MQT: stat/tasmota_907EDB/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
12:27:47.449 MQT: stat/tasmota_907EDB/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
12:27:47.699 MQT: stat/tasmota_907EDB/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

Provide the output of this command: Status 0:
12:28:44.176 CMD: Status 0
12:28:44.184 MQT: stat/tasmota_907EDB/STATUS = {"Status":{"Module":0,"DeviceName":"gosund9","FriendlyName":["gosund9"],"Topic":"tasmota_907EDB","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":1,"StateRetain":1}}
12:28:44.190 MQT: stat/tasmota_907EDB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T04:27:17","StartupUTC":"2021-07-05T07:01:27","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2021-06-24T19:52:58","SaveCount":83,"SaveAddress":"FB000"}}
12:28:44.196 MQT: stat/tasmota_907EDB/STATUS2 = {"StatusFWR":{"Version":"9.5.0(tasmota)","BuildDateTime":"2021-06-17T08:26:35","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"363/699"}}
12:28:44.200 MQT: stat/tasmota_907EDB/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":2,"SysLog":2,"LogHost":"192.168.1.3","LogPort":514,"SSId":["Wohnstube","Kaminzimmer"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","000000C8","00006000","00000180"]}}
12:28:44.212 MQT: stat/tasmota_907EDB/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":396,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
12:28:44.219 MQT: stat/tasmota_907EDB/STATUS5 = {"StatusNET":{"Hostname":"gosund9","IPAddress":"192.168.1.98","Gateway":"192.168.1.1","Subnetmask":"255.255.254.0","DNSServer":"8.8.8.8","Mac":"70:03:9F:90:7E:DB","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
12:28:44.225 MQT: stat/tasmota_907EDB/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.3","MqttPort":1885,"MqttClientMask":"gosund9","MqttClient":"gosund9","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:28:44.232 MQT: stat/tasmota_907EDB/STATUS7 = {"StatusTIM":{"UTC":"2021-07-05T11:28:44","Local":"2021-07-05T12:28:44","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"04:54","Sunset":"20:55"}}
12:28:44.235 MQT: stat/tasmota_907EDB/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:28:44.242 MQT: stat/tasmota_907EDB/STATUS10 = {"StatusSNS":{"Time":"2021-07-05T12:28:44","ENERGY":{"TotalStartTime":"2021-06-24T19:52:58","Total":0.247,"Yesterday":0.073,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
12:28:44.249 MQT: stat/tasmota_907EDB/STATUS11 = {"StatusSTS":{"Time":"2021-07-05T12:28:44","Uptime":"0T04:27:17","UptimeSec":16037,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Wohnstube","BSSId":"3C:A6:2F:2F:50:05","Channel":11,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}}

Set weblog to 4 and then, when you experience your issue, provide the output of the Console log:
12:30:20.318 HTP: Main Menu
12:30:22.921 HTP: Console
12:30:34.753 WIF: Checking connection...
12:30:36.312 CMD: Topic gosund9
12:30:36.314 SRC: WebConsole from 192.168.1.3
12:30:36.316 CMD: Grp 0, Cmnd 'TOPIC', Idx 1, Len 7, Data 'gosund9'
12:30:36.322 MQT: stat/tasmota_907EDB/RESULT = {"Topic":"tasmota_%06X"}

You can see that if I want to set the MQTT-Topic to gosund9, tasmota_%06X is still retained as MQTT-Topic Name. The topic is not changed.
The same thing happens when I make the setting via the web frontend.

ascillato, 2021-07-05:
Hi,

Tasmota rejects your topic due to it is the same as MQTT client name. It must be different as explained in the docs.

closed 2021-07-05 by ascillato2

dansiebert, 2021-07-05:
OK thank you. I can't find that in the docs. But it can also be that my English is too bad. Can you possibly post the relevant excerpt here for me? In previous versions it was possible to set the MQTT client and MQTT topic identically.